nodemcu/nodemcu-firmware #2420: Module Doesn't wake up from sleep

EssamEid614, 2018-07-07:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

when sleeping the module using the node.sleep() function with rising and falling edges added as the source of interrupt to wake the module the module should wakeup and print a wifi resume message

### Actual behavior

module doesnt wake up and goes into complete sleeping mode , i can measure the current and yes it has significantly dropped however the module doesnt wake up unless restarted although the documentation states that this should only happen with the deep sleep functionality

### Test code

cfg={}
 cfg.wake_pin=6
 cfg.resume_cb=function() print("WiFi resume") end
cfg.int_type =node.INT_BOTH
 node.sleep(cfg)

pin 6 is attached to a reed switch i can read the change of voltage on the pin however the module doesnt wake up.
i have also tried the wifi.sta.sleeptype(wifi.LIGHT_SLEEP) which does also put the module to sleep after some time of wifi inactivity, however the same issue persists the wifi does not wake up and completely doesnt respond to even commands from the esplorer
### NodeMCU version
My NodeMCU version is 2.2.0.0 built with Docker provided by frightanic.com
Powered by Lua 5.1.4 on SDK 2.2.1
i modified the initialization as suggested by the documentation to modify PMSLEEP_ENABLE in app/include/user_config.h . however the documentation didnt specify to modify TIMER_SUSPEND_ENABLE but the comment right beside it was Required by node.sleep() which led me to uncomment that as well 

### Hardware
nothing special here some LEDS and a reed switch connected to pin 6 

8<------------------------ END BUG REPORT -------------------------------------

EssamEid614, 2018-07-07:
also i would like to know if there is any possible reliable way of putting the module to sleep then waking up using an external interrupt?

EssamEid614, 2018-07-07:
@TerryE of course i tested the gpio routines ..i mean this was basically the first thing that came to mind ... i thought it was too trivial to even write down. i still don't see why a bug in the firmware should be opened on esp8266.com not on the github repo which contains the project ..

TerryE, 2018-07-07:
Good to know that you tested against gpio; this isn't a trivial point but a piece of relevant data.  Yes this is the correct place to raise bugs in the firmware, but not bugs in the application.

EssamEid614, 2018-07-07:
surprisingly i tried with D4 yesterday which didnt work so i thought all the GPIOs would do the same 

testing with D1 however is somewhat reliable when using levels but not rising and falling edges. both of the later dont work reliably at all.

however what i dont understand is, you said that this is probably due to the fact that IO12 or D6 is used in external SPI interface, but i am not even using the SPI Module in my build.
below is the list of installed modules in my build provided when i start my node board:-
modules: adc,bit,dht,encoder,file,gpio,http,i2c,mqtt,net,node,ow,pcm,pwm,sjson,tls,tmr,uart,websocket,wifi

but i dont understand the second part of your comment the FATFS in my build, i am using the file module if thats what you mean. 

is there any workaround to make pin 6 usable like that with pin 1 ?

TerryE, 2018-07-07:
You need to be careful with switches.  Your pin will float unless you pull up or down as required; typically you connect the switch to ground and use a pull up on the ESP side so that it floats high on NC and the closed switch pulls it down.  You can either use and external pullup or set the pins mode to `gpio.PULLUP`.  There's no point in trying to trigger off floating pins.

The fact that this works sometimes when using pin D1 tells me that this isn't firmware but electronics or application, which is why this discussion doesn't really belong here.  The problem is that the majority of "firmware bugs" reported by new developers are actually design bugs in their electronics or application.  We are focussed of enhancements and fixing genuine bugs here.  Other forums are for application / hardware design issues as we make very clear [in our online documentation](https://nodemcu.readthedocs.io/en/dev/en/support/).

closed 2018-07-07 by TerryE

EssamEid614, 2018-07-07:
as a matter of fact on Pin6 its set on float and i am pulling it low and high via external pull up so the fact remains that D6 actually should be the one working not D1 since D1 actually floats to NC at all yet still works as  expected , i think you got me wrong when i said somewhat reliable by which i meant in the case of using LEVELS it works perfectly fine and this DOES NOT work if using pin 6.

TerryE, 2018-07-07:
Nowhere in the firmware or the SDK does it differentiate between D1 and D6, so this isn't going to be fixed by any change to the firmware.  I give my time _freely_ to develop the firmware, not to help debug someone's hardware.  My choice, and end of this discussion.

EssamEid614, 2018-07-07:
i am not completely sure why are you that aggressive to someone who simply is asking questions .. i know my setup is correct in terms of circuit design its a simple 3 wire switch (NO connected to GND , NC connected to vcc output connected to D6)  and the code isn't honestly that large to debug .. its 6 lines of code 

but thank you for dealing with the issue in such professional manner like closing the issue without even understanding what i am on about, it does show alot about your character and how understanding/smart you can be :) .

TerryE, 2018-07-07:
Sorry I am confused.  I am not being aggressive: I hadn't made any ad hominem or discourteous comments.  I simply pointed out that a failure that varies depending on which pin (D1 vs D6) is used can't be related to the NodeMCU firmware _as this code is pin agnostic_; your problem is therefore nothing to do with the scope of this issue list.  As we explicitly document, such issues will be closed.

I also reopened it at your request and then closed it again when our subsequent dialogue confirmed my suspicions.  I encourage you to download the source code and have browse and search for the error yourself.  If you propose a PR then I will be pleased to review it.