nodemcu/nodemcu-firmware #2425: Apple devices macOS and iOS unable to connect to softAP

tnavda, 2018-07-11:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
The devices should be able to join the AP, open, wpa auth, or wpa2 auth
### Actual behavior
I am unable to join softAP with an iPhone or MBA with latest versions on both devices.  I saw the same issue with Aurdino-esp8266 core as well.  I've tried building a basic firmware with esp8266 2.2.0 and 2.2.1 sdk and have the same issue.  The only AP I have been able to join is the default AP created by esp8266 when it boots up with a sample from SDK.

### Test code

```Lua
print("Ready to start soft ap")
  
local str=wifi.ap.getmac();
local ssidTemp=string.format("%s%s%s",string.sub(str,10,11),string.sub(str,13,14),string.sub(str,16,17));

cfg={}
cfg.ssid="ESP8266_"..ssidTemp;
cfg.pwd="12345678"
cfg.auth=wifi.WPA_PSK
wifi.ap.config(cfg)

cfg={}
cfg.ip="192.168.1.1";
cfg.netmask="255.255.255.0";
cfg.gateway="192.168.1.1";
wifi.ap.setip(cfg);
wifi.setmode(wifi.SOFTAP)

str=nil;
ssidTemp=nil;
collectgarbage();

print("Soft AP started")
print("Heep:(bytes)"..node.heap());
print("MAC:"..wifi.ap.getmac().."\r\nIP:"..wifi.ap.getip());
```
### NodeMCU version

NodeMCU custom build by frightanic.com
.branch: master
.commit: 8181c3be7aed9f0a0ceb73ac8137c1a519e8a8e9
.SSL: true
.modules: bit,crypto,file,gpio,http,mdns,net,node,pwm,spi,sqlite3,tmr,uart,wifi,wifi_monitor,wps,tls
 build created on 2018-07-06 13:48
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
### Hardware
This is a nodemcu v1.0 device (40mhz crystal)

8<------------------------ END BUG REPORT -------------------------------------

tnavda, 2018-07-11:
Also tried with 1.5 SDK,
NodeMCU custom build by frightanic.com
.branch: 1.5.4.1-final
.commit: f03a8e45261fb5ab260e316173baacae9a248a62
.SSL: false
.modules: file,gpio,net,node,tmr,uart,wifi
 build created on 2018-07-11 02:30
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

macOS wifi.log output

```
Tue Jul 10 22:46:03.173 Info: <airportd[201]> <en0> Failed to associate to Wi-Fi network ESP8266_70f561, returned error code -3905, retrying
Tue Jul 10 22:46:03.376 Info: <airportd[201]> SUSPEND AWDL for interface en0, timeout=10.0s, reason=Assoc, token=297
Tue Jul 10 22:46:03.376 Info: <airportd[201]> RESUME AWDL for interface en0, reason=Assoc token=296
Tue Jul 10 22:46:03.383 <kernel> parseWPAIE: groupCipherType = 3 pairwiseCipherType = 5 authSel = 2
Tue Jul 10 22:46:03.383 <kernel> initWithInterfaceAndIE: _myMacAddress 94:94:26:08:c8:e0
Tue Jul 10 22:46:03.383 <kernel> setPMK: PMK SET!
Tue Jul 10 22:46:06.081 <kernel> Unexpected payload found for message 9, dataLen 0
Tue Jul 10 22:46:06.087 <kernel> en0: Terminating supplicant.
Tue Jul 10 22:46:06.087 <kernel> RSNSupplicant: Releasing authenticator for 00:00:00:00:00:00
Tue Jul 10 22:46:06.087 Info: <airportd[201]> <en0> Failed to associate to Wi-Fi network ESP8266_70f561, returned error code -3905, retrying
Tue Jul 10 22:46:06.291 Info: <airportd[201]> Failed to associate to Wi-Fi network ESP8266_70f561 on interface en0, returned error code -3905
```

tnavda, 2018-07-11:
After doing some packet capture, it looks like esp8266 isn't responding to the 802.11 Authentication packet from Apple devices.  The only difference between the Windows and Apple auth requests is the OUI data, which means the frame check sequence doesn't immediately follow the 802.11 wireless management frame and maybe this is why the esp8266 isn't responding.

ildar, 2018-07-11:
Did you check with the stock firmware?

tnavda, 2018-07-11:
Yeah,
I wanted to say at one point the default SDK firmware flash as:

ESP8266_AT_Bin_V1.6.2:
```
esptool.py --port /dev/tty.SLAB_USBtoUART --baud 115200  write_flash -fm dio -fs 4MB 0x00000 boot_v1.7.bin 0x01000 user1.2048.new.5.bin 0x3fc000 esp_init_data_default_v08_40mhz.bin 0xfe000 blank.bin 0x3fe000 blank.bin
```
was working when WPA/2 wasn't configured, but even know I can't connect to it with iOS device.  Looking at the 802.11 packets again, it looks like the same issue.

FrankX0, 2018-07-11:
I can connect to both iPhone (iOS 11.4.1) and iPad (9.3.5) with a similar build and your code:
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 5d1dae7f1e0a52557858e457f53afbe32b63f2fa
	SSL: false
	modules: enduser_setup,file,gpio,net,node,tmr,uart,wifi,wifi_monitor,wps
 build created on 2018-07-04 17:48
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
```

marcelstoer, 2018-07-11:
> I saw the same issue with Aurdino-esp8266 core as well

> Tried both ESP8266_AT_Bin_V1.6.2 and ESP8266_NONOS_SDK-2.2.0 [...] but even now I can't connect to it with iOS device. 

Definitely not a firmware issue -> closing

tnavda, 2018-07-11:
@marcelstoer 

I just flashed with ESP8266_AT_Bin_V1.6.2 and enabled via the AT commands and it worked without issue. :/

closed 2018-07-16 by marcelstoer