nodemcu/nodemcu-firmware #2426: rtctime.dsleep does NOT keep rtctime after wakeup

JamesOlvertone, 2018-07-11:
### Expected behavior
rtctime after wakeup from deep sleep should not be 0.
node.bootreason() should be  "3, software watchdog reset"

### Actual behavior
rtctime is 0sec 0us after wake up
node.bootreason()  is: " 6, external reset" 

### Test code
```Lua
sec, us = rtctime.get()
print("sec:"..sec.." usec:" .. us)
if (sec == 0) then
 rtctime.set(1);  -- to start running the rtc
end

tmr.alarm(0,10*1000, tmr.ALARM_SINGLE, function()
 -- we wait a few seconds so we can upload new versions and other programs

 sec, us = rtctime.get()        -- show time again before we go to sleep
 print("sec:"..sec.." usec:" .. us)
 print("\n\n\n");
 rtctime.dsleep(5*1000)
  --node.dsleep(5*1000);  -- this vesion is documented that it does NOT keep time
 end)

```
### NodeMCU version
powered by Lua 5.1.4 on SDK 2.1.0(116b762)

### Hardware
testet on:
nodeMCU board (ESP-12 "AI-Thinker"),
Wemos   D1 old revisin not the R2 (ESP-12, "Hysiri")

### Workarounds
Idea: Store the time before going to deep sleep in rtcmem
after wakeup read the time from rtcmem and add the slept time on it.

TerryE, 2018-07-12:
@jmattsson, one for you to comment on.

jmattsson, 2018-07-13:
Confirmed same (working) behaviour on my second NodeMCU devkit 0.9. I can't test on the 1.0 board I have as it's not wired for deep sleep.

@pjsg would you mind checking this out too?

JamesOlvertone, 2018-07-13:
I flashed the newestr firmware 2.2.1 now it works.

closed 2018-07-13 by TerryE