nodemcu/nodemcu-firmware #2428: file operations take up to 30 seconds

eku, 2018-07-15:
I used to use luatool.py to upload my lua files over usart connetion. It worked for the past years. Recently I upgraded my modules with HEAD of dev branch.

### Expected behavior
File operations only take fractions of seconds up to a maximum of a few seconds.

### Actual behavior
file.open takes up to 30 seconds. This breaks upload by luatool.py.

### Test code
Just upload some files with [luatool.py]luatool.py(github.com:4refr0nt/luatool) . It will stop uploading file with ```Exception('No proper answer from MCU')```.

```
> =file.fsinfo()
7279    24096   31375
```
The file I'm trying to upload is 2203 bytes. ```file.open("foo")```  takes about 30 seconds till prompt returns. I'm pretty sure that I had more free space for spiffs before upgrading to HEAD. Unfortunately I don't remember the version I've had been using before.

### NodeMCU version
Latest HEAD of the dev branch.

### Hardware
Plain ESP-03 with 512kB of flash.

TerryE, 2018-07-15:
Hi.  We can't answer this type of general issue meaningfully.  The Lua core ha grown slightly as we've added more features and functionality though some of these are configurable out in your `user_config.h`.  A more serious constraint is that the Espressif SDK size on flash has grown steadily over the years as new features and extra checking have been added.  Espressif don't even support their latest AT firmware versions on the 512Kb parts.

As a result the 512 Kb builds are always marginal unless you do hard customisation, and to be honest, the smallest parts that I test on these days are 1Mb.   I can't give more help without more information and even then I may not be able to help.

eku, 2018-07-16:
@TerryE  well,  I know of the fact that the NodeMCU code has increased and spiffs size decreased. This issue is about execution times of file operations. What's the root cause?

I'm using plenty of small lua files. Would it be possible to reduce the cut of the file system by reducing the base sector size?

TerryE, 2018-07-16:
> I'm using plenty of small lua files. Would it be possible to reduce the cut of the file system by reducing the base sector size?

You have to remember that SPIFFS over flash is not the same as a conventional FS such as FAT32 over a block device.  The flash on an ESP module is bare and doesn't have a wear-leveling controller or the like in front of it.  The flash is organised into 4Kb pages.  You can read pages or part pages.  You can overwrite part-page flash content so long as you are only clearing (some) bits in the content.  If you need to set a single bit then your only option is to get the contents of the full page, erase it and write back the updated content.  SPIFFS is optimised to work withing these constraints to minimize the number of erases and to avoid "hot" pages which will collapse the life of the flash.  There are a number of SPIFFS configuration options which accelerate SPIFFS performance, but these require more RAM to be allocated.

I used to break my Lua code into lots of small component and Just-in-Time load them, but now with LFS I've moved over to a similar style to my host-based programming with modules typically 300-500 lines, as there is no need to split stuff up just to be able to run in RAM.

Out of interest is there any reasons why you are clinging to 512Kb parts and still  doing active development on them?  I can understand that you might have some older devices running Lua 1.5.4 apps, but if you want the current SDK / Lua feature set, then the Wemos parts (for example) are ~$3 in batches of 10+ and these have a lot more functionality than the old ESP-03s

eku, 2018-07-16:
> Out of interest is there any reasons why you are clinging to 512Kb parts and still doing active development on them? 

I run these 512kb parts in production for a long time with no plans to replace them. I would just like to have taken over the bug fixes. Went back to SDK 2.1.

@TerryE thanks for your support.

closed 2018-07-16 by eku