nodemcu/nodemcu-firmware #2439: None of the DS18B20-Libs work, raw OW works

JamesOlvertone, 2018-07-25:
### Expected behavior
Finding a DS18B20 on the specific pin and get its data with the DS18B20-Firmware-Lib or the Moduleversion, both do not work for me. No sensor found.

### Actual behavior
DS18B20 Firmware support: no sensor found
DS18B20 Module Version: no sensor found

The irony here is: The scan example from the ow-lib which searches for a tempsensor finds the DS.

Whats wrong with this libraries?

### Test code
Used the examples from the documentation of the module-version and the firmwareversion.

Tested with different Pullups on breadboard, different Sensors, no long cable on the Sensor-Pin, 3.3V powered.
The OW-Example works with all kind of values, even without a pullup.
Used pin: D3
DS18B20-Mode: non-parasite

I noticed a conflict if you use the module version and compiled in version parallel, same have name.
Do a reboot after each test or rename the module version. After handling this there is still no improvement, the ds18b20 libs dont work here.

### NodeMCU version
modules: adc,bit,bme280,bmp085,dht,ds18b20,file,gpio,http,i2c,mdns,net,node,ow,rtcfifo,rtcmem,rtctime,sntp,sqlite3,tmr,uart,wifi
powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)

### Hardware
Wemos D1 Mini, nodeMCU Board

cwrseck, 2018-07-30:
The code I use for checking DS18B20s is at:
https://pastebin.com/uiyamTMq
It needs to be compiled with luac.cross (I think), but it runs on the
Adafruit Huzzah and the Wemos D1 Pro.
(This is on an integer version of the 20180608 release.)
Will

vsky279, 2018-08-03:
Hi, I think we don't have enough information to help you.

Maybe you can try to debug the DS18B20 Lua Module Version to find where the issue is. You know to add some `print()` commands to see how where the issue is. If you were able to get it working with `ow` module it should not be too difficult to find out where the issue in the module version is.

Once you solve this very likely you will get the answer to the C version (or firmware) of the module too.

TerryE, 2018-09-17:
The OP hasn't posted back and probably found his problem, so I am closing this.

closed 2018-09-17 by TerryE