nodemcu/nodemcu-firmware #2441: Lua driver module support for Eink/Epaper displays

dtran123, 2018-07-27:
### Missing feature

We need more support for Eink/EPaper displays available on the market today such as:
- Good Display
   http://www.good-display.com/
- Waveshare
   https://www.waveshare.com

Some developpers have created drivers for them for the ESP8266 here:
https://github.com/ZinggJM/GxEPD

It would be nice to port this to Lua. Possibly include it in U8glib ?

### Justification

Eink/Epaper displays are very useful for IoT applications due to their minimum use of battery resources. The ESP8266 can wake up, update the display, and go back to sleep.

### Workarounds
No workaround for now other than use other types of displays that consume a LOT more energy.

FrankX0, 2018-07-28:
These type of displays are supported in u8g2lib.
Ther is a pull request #2184 to replace u8glib with u8g2lib.
I have an epaper display working using this pull request.

TerryE, 2018-07-28:
Given @FrankX0 Frank's valid point I think that we've already implemented @dtran123's request.  However, I'll leave this issue for a few more days for further comment before closing it.

dtran123, 2018-07-30:
Thanks. that's good to know. As long as the specific eink displays are listed as supported displays in the documentation under u8g2lib, I'm fine with closing it.

closed 2018-08-06 by marcelstoer