nodemcu/nodemcu-firmware #2443: trouble with enabling LUA_FLASH_STORE

mikep01, 2018-07-30:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Proper startup calling init.lua

### Actual behavior
after a reset I see my own messages at 9600bps then the CPU appears to hang.  Changing to 115000 and another reset I see:

2018.07.29 21:03:23.948 lua: init.lua:1: unexpected symbol near '¼'
2018.07.29 21:03:23.891  powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
2018.07.29 21:03:23.871 2.2.0.0 build created on 2018-07-29 04:28
2018.07.29 21:03:23.867 	SSL: false
2018.07.29 21:03:23.865 	commit: c6f6c54bc82e5868a45ebc44f73bec9674837c65
2018.07.29 21:03:23.862 	branch: dev
2018.07.29 21:03:23.859 NodeMCU  built with Docker provided by frightanic.com
2018.07.29 21:03:23.856 
2018.07.29 21:03:23.848 Flash sig not correct: 0 vs fafaaf52
2018.07.29 21:03:23.748 s$ ==> this is probably my messages from 9600bps...

### Test code
Works fine when built with LUA_FLASH_STORE not defined, calling init.lua properly.
When built with it defined I see the above messages.

### NodeMCU version
NodeMCU 2.2 but I've back ported the tmr functions (as per an earlier 

### Hardware
NodeMCU v0.9, nothing special.
8<------------------------ END BUG REPORT -------------------------------------

mikep01, 2018-07-30:
Enabling (defining) LUA_INIT_STRING seems to help, now things run to the point of showing the interactive prompt at my defined 9600bps.  The docs gave me the impression this was optional?  Running resume("init") fails:

2018.07.29 21:15:48.701 	stdin:1: in main chunk
2018.07.29 21:15:48.700 	[C]: in function 'require'
2018.07.29 21:15:48.680 	[C]: ?
2018.07.29 21:15:48.646 stack traceback:
2018.07.29 21:15:48.613 	init.lua:1: unexpected symbol near 'é'
2018.07.29 21:15:48.580 error loading module 'init' from file 'init.lua':
2018.07.29 21:15:48.299 > require("init")
2018.07.29 21:15:48.278 require("init")

Perhaps this failure is because I have not yet provided an luac.out and run node.flashload()?  Again, I was under the impression this was optional, but perhaps not.  I'll try now.

BTW, notice that in my logs the oldest messages are at the bottom.

TerryE, 2018-07-30:
@mikep01 This might be related to #2430.  Juggling three balls ATM, but I'll stash my WIP and take a look at these both tomorrow.  

What modules are you including?

//Terry

mikep01, 2018-07-30:
Mostly standard plus some of my own modules for a custom build.

When it starts I get those above messages at 115000bps, then my custom code changes to 9600 and the command problem is fine - I can execute things like =node.heap(), require(), etc.

So I think the startup is ok (and it should be since it works on a build without LFS enabled).

I'm not sure I've got my luac.out working properly, I'm using virtualbox 'cause I'm a windows user.  Luac.cross runs and gives me output, but I'm not sure if it survives being moved back to the windows fs and then uploaded.  When I try to run node.flashinit("luac.out") the ESP doesn't reboot, it just comes back to the prompt (and I can still run =node.heap()).

So what I've done is load up my file system with no init.lau - mine I've renamed to "myinit.lua".
At the prompt I then run "node.flashindex("_init") and the prompt comes back after a second or two.

Now if I run "myinit.lau" (remember my log has oldest at the bottom, I get weird things as if my other lua files like MppCommon.lua have been corrupted.  But I formatted the file system and uploaded them fresh - they should be ok:

2018.07.29 23:52:57.170 MppStartup.lua:169: attempt to call global 'SetPin' (a nil value)
2018.07.29 23:52:57.117 MppStartup processing for MppESP8266 6.0.3...
2018.07.29 23:52:56.870 	MppCommon.lua:1: unexpected symbol near '@'
2018.07.29 23:52:56.823 error loading module 'MppCommon' from file 'MppCommon.lua':
2018.07.29 23:52:56.527 > MppServer MPP 4.1.3 started on port 8898
2018.07.29 23:52:51.829 > 
2018.07.29 23:52:51.809 startup pause...
2018.07.29 23:52:51.500 []
2018.07.29 23:52:51.388 MppInit 6.0.1 is starting at 42979491...
2018.07.29 23:52:51.374 
2018.07.29 23:52:51.096 > require("myinit")

mikep01, 2018-07-30:
BTW, this, from user_config.h, looks a bit odd to me - are the 's in the right place?
#define LUA_INIT_STRING "pcall(function() node.flashindex'_init'() end)"

TerryE, 2018-07-30:
> When I try to run node.flashinit("luac.out") the ESP doesn't reboot.

The current implementation returns without rebooting if there is a problem with the LFS image.  This is confusing.  Not good.  I am doing a rework which gzips the image and it is exploded on the fly to write to flash.  This is done in two passes: the first is a "dress rehearsal" which doesn't write to flash, but checks the file header, the gzip format length and CRC and caches the relocation table.  Any errors here are returned with a string error, since the flash hasn't been updated yet,  The second pass then writes to flash and reboots, so the only scenario which could lead to flash corruption would be power-fail during the write;  though this is detected at reboot.

The LFS does have a Unix DTS of the compile embedded in it.  I need to update the cross compile to emit an information message giving stats on the image (length, compressed length, number of modules, DTS) so you can reconcile LFS versions.

> I'm using virtualbox ...

Install guest additions and read up on using shared folders.  The easiest thing to do is to share your `local` and `bin` directories back onto your dev WinX FS so that you can work with them in Explorer and your favourite WinX tools.
 
> BTW, this, from user_config.h, looks a bit odd to me  ...

Dropping the \() around a singleton string constant or array parameter  is standard Lua syntactic sugar.

HHHartmann, 2018-07-30:
Hi, locks like you first got to get the luac.out right.
It is important to have it built 
- position independent 
- matching your build type (int or float)
  the luac.cross has to be built as the firmware image. I fyou use the docker image it builds both and only leaves the last build. See my comments on #2413 

As the image is not flashed it would logical that _init and telnet cannot be found.

mikep01, 2018-07-30:
There is something odd about the FS with this version, and I *think* it only happens with LFS enabled but I can't be 100% sure.  Loaded up a non-LFS enabled firmware and things ran fine.  But with the earlier device, and now with a second device, after loading up the LFS version some odd FS behavior started happening.  It took 3 or 4 formats to get it sorted and reloads to get it working (w/o any LFS stuff running).  Sorry, I know that's not much help, I have no idea how to recreate it.

mikep01, 2018-07-30:
I'm using the integer build and I took the luac.cross from the result, is that the right way to grab it?  I do use the docker image.  To build luac.out I have to move everything to virtualbox.

There aren't any messages from luac.cross so what happens there is a bit magic.

HHHartmann, 2018-07-30:
Should be OK. Do you start the docker build with `-e "INT_ONLY=1"` ?
Try starting docker with `bash` as last param. It will go into interactive mode.

Unfortunately luac.cross does not reveal its int/float state in -v (version) either.

HHHartmann, 2018-07-30:
INTEGER_ONLY is ok. I didn't bother to lock up the exact string, Sorry.

> To run it I used the instructions: ./lua.cross -f *.lua

Looks good if you used luac.cross  ;-)
 
> I'm not sure what I would do in bash/interactive mode for docker? (I may have misled here, I run docker locally on my system to make my binary, not in the cloud).

You could call luac.cross there directly and not have to copy it.
Works for me, also having a local docker with Marcels image.

mikep01, 2018-07-30:
Trying to run on my docker console gives me this:
bash: luac.cross: cannot execute binary file: Exec format error
Because I'm on windows 7?  It runs ok on my virtualbox ubuntu.

TerryE, 2018-07-30:
@mikep01, the luac.cross make assumes a POSIX environment. So it's either a case of running on VBox or installing Cygwin and running it within Cygwin on Win7.  In the case of VBox ubuntu, if you use a shared folder then there is nothing stopping you installing openssh for windows and then you can kick of any recompile from a cmd32 script or an icon.

mikep01, 2018-07-30:
Thanks Terry, that's what I thought.  I am good on the .lua build side, once LFS works I'll change my process for the .lua file builds.

So what seems to be stopping me is getting luac.out to work with LFS, hopefully some messages will help.  I can send you my luac.out if you need it.

HHHartmann, 2018-07-31:
I meant the docker command you use to compile the firmware. Appending a command will execute it instead of the configured startup command.
Use something like
``docker run --rm -it  -v C:/{...}/nodemcu-firmware marcelstoer/nodemcu-build bash``

@TerryE I could prepare a PR to build luac.cross.integer and luac.cross.float also adapting the tools Makefile.

mikep01, 2018-07-31:
Thanks - what I don't understand is what I should do with it since luac.cross can't run in my docker instance.

TerryE, 2018-07-31:
@mikep01 Mike, I don't know if you are at all familiar with C and make, but the firmware source is primarily designed to be build with the Xtensa toolchain to generate a binary which runs on the ESP.  However the `lua` sources  and `lua/luac_cross` directory make also use a standard (POSIX) toolchain (the host gcc etc) to build a standard ELF binary luac.cross which an run on the host environment.  This could be Linux, Win10/WSL or WinX/Cygwin.  So you can run luac.cross on any of these.  Docker is just an application virtualisation technology that runs over Linux,so luac.cross _can_ be run in your docker container, but as Gregor says you can use the default (make) script; you need another script, which someone needs to develop.

TerryE, 2018-07-31:
I did the first and second implementations of luac.cross and various other hairy extension to the firmwares capability such as LFS itself, I've only got so much bandwidth.  What would be great is if others in the community could develop the wrapper stuff, so I can focus on the complexities of the lua core. 

So yes, Gregor it would be great if you work on the PR and if you need any input just drop me an email or we can talk on [gitter](https://gitter.im/nodemcu/nodemcu-firmware)

mikep01, 2018-07-31:
Ok, I understand now, thanks, it's not necessary for me.  Most of my development will be in Lua and will be rebuilding/reloading the luac.out file.  I don't need it combined with my nodemcu build so virtualbox is fine once I can get LFS working.

closed 2018-08-15 by TerryE