nodemcu/nodemcu-firmware #2444: build error missing time.o for 2.2 (no error 2.1)

stevemac00, 2018-07-31:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Build
### Actual behavior
I have been making builds with 2.1 for the past year without any issues. I created a new disk for 2.2 but when I attempt to build from 2.2 I get this error:

ar d /Volumes/casesenshfs-220/nodemcu-firmware/sdk/esp_iot_sdk_v2.2.0-f8f27ce/lib/libmain.a time.o
ar: time.o: not found in archive

Here is what I use to build:
git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
cd esp-open-sdk
make STANDALONE=y
export PATH=...
cd ..
git clone https://github.com/nodemcu/nodemcu-firmware.git
cd nodemcu-firmware
make clean
make

### Test code
### NodeMCU version
2.2 esp_iot_sdk_v2.2.0

### Hardware
N/A

8<------------------------ END BUG REPORT -------------------------------------

TerryE, 2018-07-31:
@stevemac00 I am doing these `sdk_v2.2.0-f8f27ce` builds regularly and my build doesn't attempt to access `time.o`

I can't find anything in the makefile to attempt to delete `time.o` from `libmain.a`.  Can you post the make log as a gist and link it here.  Thanks.

stevemac00, 2018-07-31:
I may have aborted at one time (I put this away for awhile and can't remember).
I did rm -rf /Volumes/casesenshfs-220/nodemcu-firmware/sdk/ but still got the same error.
The gist is https://gist.github.com/stevemac00/059ec9109f9865132ddcda70925bf8ee

Since it's complaining about time.o, my only clue is I'm defining SNTP but not RTCTIME although that was fine in 2.1.

devsaurus, 2018-07-31:
@stevemac00 building from `master` also works for me.
What is the output when you list the archive manually once the build failed?
For me it's
```
$ ar t sdk/esp_iot_sdk_v2.2.0-f8f27ce/lib/libmain.a
app_main.o
eagle_lib.o
eagle_lwip_if.o
ets_timer.o
mem_manager.o
spi_overlap.o
time.o
user_interface.o
debug-vector.o
double-vector.o
kernel-vector.o
nmi-vector.o
user-vector.o
vector.o
spi_flash.o
```

Deleting `time.o` will succeed of course.

stevemac00, 2018-07-31:
$ ar t sdk/esp_iot_sdk_v2.2.0-f8f27ce/lib/libmain.a
 ###      
    /
    //
    app_main.o/
    eagle_lib.o/
    eagle_lwip_if.o/
    ets_timer.o/
    mem_manager.o/
    spi_overlap.o/
    time.o/
    /0
    debug-vector.o/
    double-vector.o/
    kernel-vector.o/
    nmi-vector.o/
    user-vector.o/
    vector.o/
    spi_flash.o/

Now I'm even more confused. What is that /0 after time.o ?

stevemac00, 2018-08-01:
Thanks for staying on this as I need the help. Something is wrong with what I've used on the 220 but I list the simple instructions I followed on my original post at the top.

I'm on MacOS 10.13.6 and using /usr/bin/ar included but it doesn't support the 'V' option for version.
I have my 2.1 and 2.2 builds on separate images so I went back to 2.1 and did a make clean then make which worked fine although it lists the same '/0'.

_I'm wondering why the date is older on -220 than it is on -210?_

**nodemcu-firmware-210$ ar tv sdk/esp_iot_sdk_v2.1.0/lib/libmain.a** 
---------       0/0          6728 May  5 03:09 2017 /
---------       0/0            18 Dec 31 18:00 1969 //
rw-rw-r--     502/502       34708 May  5 03:09 2017 app_main.o/
rw-rw-r--     502/502        9820 May  5 03:09 2017 eagle_lib.o/
rw-rw-r--     502/502        6936 May  5 03:09 2017 eagle_lwip_if.o/
rw-rw-r--     502/502       10740 May  5 03:09 2017 ets_timer.o/
rw-rw-r--     502/502        8024 May  5 03:09 2017 mem_manager.o/
rw-rw-r--     502/502        3152 May  5 03:09 2017 spi_overlap.o/
rw-rw-r--     502/502        7144 May  5 03:09 2017 time.o/
rw-rw-r--     502/502      104516 May  5 03:09 2017 /0
rw-rw-r--     502/502        1288 May  5 03:09 2017 debug-vector.o/
rw-rw-r--     502/502        1288 May  5 03:09 2017 double-vector.o/
rw-rw-r--     502/502        1288 May  5 03:09 2017 kernel-vector.o/
rw-rw-r--     502/502        2140 May  5 03:09 2017 nmi-vector.o/
rw-rw-r--     502/502        1816 May  5 03:09 2017 user-vector.o/
rw-rw-r--     502/502        1564 May  5 03:09 2017 vector.o/
rw-rw-r--     502/502       18388 May  5 03:09 2017 spi_flash.o/

**nodemcu-firmware-220$ ar tv sdk/esp_iot_sdk_v2.2.0-f8f27ce/lib/libmain.a** 
---------       0/0          6890 Feb 21 23:36 2018 /
---------       0/0            18 Dec 31 18:00 1969 //
rw-rw-r--     502/502       34076 Feb 21 23:36 2018 app_main.o/
rw-rw-r--     502/502        9820 Feb 21 23:36 2018 eagle_lib.o/
rw-rw-r--     502/502        7640 Feb 21 23:36 2018 eagle_lwip_if.o/
rw-rw-r--     502/502       10740 Feb 21 23:36 2018 ets_timer.o/
rw-rw-r--     502/502        8024 Feb 21 23:36 2018 mem_manager.o/
rw-rw-r--     502/502        3152 Feb 21 23:36 2018 spi_overlap.o/
rw-rw-r--     502/502        6932 Feb 21 23:36 2018 time.o/
rw-rw-r--     502/502      109444 Feb 21 23:36 2018 /0
rw-rw-r--     502/502        1288 Feb 21 23:36 2018 debug-vector.o/
rw-rw-r--     502/502        1288 Feb 21 23:36 2018 double-vector.o/
rw-rw-r--     502/502        1288 Feb 21 23:36 2018 kernel-vector.o/
rw-rw-r--     502/502        2140 Feb 21 23:36 2018 nmi-vector.o/
rw-rw-r--     502/502        1816 Feb 21 23:36 2018 user-vector.o/
rw-rw-r--     502/502        1564 Feb 21 23:36 2018 vector.o/
rw-rw-r--     502/502       18664 Feb 21 23:36 2018 spi_flash.o/

stevemac00, 2018-08-01:
**Found a solution!**
Turns out the 2.1 build has:
`esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/bin/ar`
where as the 2.2 build does not but has:
`esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar`

I simply cp xtensa-lx106-elf-ar to ar so it was used in my $PATH before /usr/bin/ar and it worked. Maybe the make file could be updated to explicitly use xtensa-lx106-elf-ar?

devsaurus, 2018-08-01:
Sure, that should do the trick :+1: Also removes dependencies to the host system installation.
Thanks a lot for digging this up!

PR #2446 (2018-08-01): use ar from esp toolchain instead of host
Fixes #2444.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

The steps for preparing the binary libraries use `ar` from the host's toolchain. This appears to fail sometimes as reported in the related issue.
Since we're operating on archives for the target esp system, I'd say it's safer to use `ar` from the target  [... cut]
closed 2018-08-02 by marcelstoer