nodemcu/nodemcu-firmware #2457: LFS Quick Start Documentation for Docker Builds

jmd13391, 2018-08-12:
8<------------------------ DOCUMENTATION REQUEST ------------------------------------
### Missing Documentation
LFS Quick Start (Step by Step) Documentation for application developers using the Docker NodeMCU platform (https://hub.docker.com/r/marcelstoer/nodemcu-build/).

### Justification
Currently, the setup and use of LFS is not intuitive to newbie application developers wanting to leverage the ESP8266 NodeMCU platform. Recently, LFS Quick Start documentation for application developers using the Cloud Build Service has been introduced in the NodeMCU Dev branch (https://github.com/nodemcu/nodemcu-firmware/blob/dev/docs/en/lfs.md). While this is an excellent start, it does not translate smoothly over to application developers using the Docker NodeMCU Build approach (specifically, the section titled "Build the LFS image").

### Workarounds
n/a

8<------------------------ END DOCUMENTATION REQUEST --------------------------------

jmd13391, 2018-08-13:
RE: #2443 - specifically: https://github.com/nodemcu/nodemcu-firmware/issues/2443#issuecomment-409189592

Running that through my handy @TerryE -to- Entry Level Application Developer translator results in:

"The current implementation of NodeMCU Docker Build (https://github.com/marcelstoer/docker-nodemcu-build) by @marcelstoer does not support LFS"

If my translator is working then I really think it that statement should be included in the docs (until which time someone develops a working NodeMCU-Docker make script).

If my translator is broken then my original request for a NodeMCU-Docker Quick Start stands as-is.

_Now off to pop a few Ibuprofen..._

jmd13391, 2018-08-13:
@HHHartmann, Spot-On.

Guys (a.k.a. NodeMCU firmware developer gods), IMO the goal of all the countless hours of effort you have put in to make NodeMCU what it is today (phoenix from the 0.95 days) should be to make using the ESP platform via the lua/Node.js-like programming model as intuitive and easy to use / easy to ramp on to the ESP gettin' 'er done highway as possible. In order to reach that goal, you have to look at the documentation with eyes of a newbie application developer. Example: Close your eyes and visualize for a moment that you have absolutely no idea what a "make" file is or that I have no idea how to code in any of the C programming language variants.... words like "memalloc" are alien to you... (yes, I know, painful... keep your eyes closed). Flush everything from your brain except lua, Node.js and the BASICS of common working environments like Windows, Linux, Docker. Keep the documented modules of NodeMCU in there too. Now, open your eyes and describe the step by step process to get a xxxxx.lua application running using the LFS approach. Keep in mind that your audience at this point is absolutely not interested in understanding the mechanics behind each step (that comes later when they level-up)... they are only interested in the steps themselves.

Get the docs to that point and soon there will be a lot more coders out there building ESP applications using LFS NodeMCU then there are using the Arduino IDE. I look forward to the day when I Google search for an instructable or code example and get more NodeMCU hits than Arduino IDE hits. ;-)

-Joe

TerryE, 2018-08-13:
I don't know Joe's developer background, so let me suggest a "typical" developer who is wanting to use NodeMCU:
-  No experience of OSes other than WinXX.  
-  So largely comfortable with the Windows GUI environment and 
-  Generally uncomfortable with a command prompt environment.  Maybe some minimal `cmd` working or possibly  but no experience of `bash` or its MS superset of subset, `powershell`.
-  No experience of \*nix OSs or any POSIX compliant OS.
-  Also used to the WinX Ecosystems where you just install and start using, and fall back to online help if you need it, rather than reading the documentation first.
-  Unfamiliar with C.
-  Wants to hack on IoT devices to do Home Automation and related projects.

This type of developer might want to pick a Lua environment over an Arduino C-based one.  For this developer, using docker represents a real learning curve as everything is so foreign.  Shit, I've used 4 mainframe OSes, ditto mini OSes, even more \*nix variants, and everything from CP/M to all of the WinX variants and I still haven't bothered to get to grips with Docker, as for me standing up a base Linux VM is a 15 min job.  

Going back to the `cmd` vs `powershell` and docker either way, if I we doing this I would want a BAT file or a function to wrap the `docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware` bit so I can just do one of:
```cmd32
docrun help
docrun build
docrun lfs
```
that is the default directory should be pathed in. (and it shouldn't be `marcelstoer` anything) and a basic help should explain the commands (scripts) available and how you can tailor them.

-----
What I just don't understand is why developers want to use docker to do luac.cross.  
-  When I am developing the Lua core or a new library, I build the firmware a lot, but for production use, once in a blue moon.   
-  When I am developing a Lua app, I am rebuilding the LFS all of the time sometimes once every few minutes, because I find it easier and faster just to recompile and reflash LFS because this takes seconds and is faster than using Explorer to update SPIFFS, etc.

Surely docker is overkill in this second case.  I don't understand why developers just don't  use Cygwin as it takes less than 5 mins to download the Cywin setup, tick a few boxes, download the latest NodeMCU source, tweak `user_config.h` and run the `app/lua/luac_cross` make and now you have a binary that you can run directory on your Windows machine.  

And for those developers than just want to stop at the "tick a few boxes" stage, here is the [luac.cross.cygwin](http://files.ellisons.org.uk/esp8266/luac.cross.cygwin) image to run in Cygwin, but remember to move it to a folder on your cygwin path and rename it to `luac` and do a `chmod +x` on it.  (Its MD% is \15e5010626d435ea48aaf1e64879d686).  The only problem that I had was to stop Avast treating it as suspicious.

HHHartmann, 2018-08-13:
maybe it's just a matter of documentation and easyness why people start using docker and not cygwin or a linux VM.
For docker there are just a couple of commands needed and provided to get a build.

As for setting up gygwin even "tick a few boxes" is easy if you know which ones. Otherwise you are just lost.
Someone not knowing anything about linux/posix even does not  know what "do a chmod +x on it" means.

I could have used cygwin or any linux but it was just easy to install docker and issue two commands to have a build done.

I agree that to go the full way, there should be batches to even wrap the somewhat long docker commands.
But then again who should modify the path to include the batches. Do we also need some power shell scripts to automate this?

TerryE, 2018-08-13:
> Do we also need some power shell scripts to automate this?

IMO, if you've mastered power shell, then you are a long way into being able to use native WSH or Cygwin,  If we are going minimalist then we are really talking about CMD prompt batch scripts, but yes having one person take on this learning curve for the community of WinX developers would be great.

jmd13391, 2018-08-13:
@HHHartmann Spot-On,,, again,

I will also add that there is a greater chance of someone new to NodeMCU & the ESP platform already having Docker installed... and very high probability of those people complaining about having to install yet another supporting environment (Cygwin) just to get work done.

I will speculate that the Cloud Build Service is the most popular method of getting from A to Z with the ESP & NodeMCU simply because it happens to be the path of least resistance. Unfortunately, there are several modules and features of NodeMCU that are not exposed on the Cloud Build Service. For application developers such as myself that want/need to leverage some of those "not enabled by default" functions and have no desire to dive into the bowels of the full NodeMCU toolchain, The Docker Build method has been the compromise... and the docs on how to build using Docker has been adequate... until LFS came along.

If using LFS (and I really, really need to use it) means that I have to stop application development, install and get comfortable with yet another build procedure (Cygwin), so be it. I've been staring at the LFS carrot dangling a few inches from my mouth for months now... (waiting for the paint to dry, trying to decode Terry's LFS whitepaper, and trying to find the build "path of least resistance". Time to accept that at least for the moment, I'll have to go the Cygwin route.

I'll leave you with this final comment: As someone who has been in this industry as long as @TerryE has (if not a bit longer), I can tell you without hesitation that being able to access LFS and the non-default features of NodeMCU via @marcelstoer's Cloud Build or Docker paths will be a major boost to the NodeMCU user base. LFS is a work of art... Make it super easy for the world to use and great things will follow.

-Joe

marcelstoer, 2018-08-13:
Guys, can we please stop discussing the respective merits of the various ways to build NodeMCU - and focus on the issue(s)? I don't mind if other people like technology X, Y, Z better than Docker. It's big in the "cloud" and microservices architecture world and lots of developers have it already installed for that reason (may be their day jobs). Likewise, I'm not the least bit offended if people don't want to use my NodeMCU image; I don't take it personal.
Back in the days I simply created it because it felt easy to do and offered an intermediary abstraction between the cloud builder and a Linux+toolchain environment. However, so far it's been pulled >9k times which means a few thousand individuals (I'm guessing 1-2k) find it useful. I just accept it as a fact and are humbled by that relative success.

That being said it's of course not so good that it couldn't become better. I'm happy to assess ideas filed as issues over at https://github.com/marcelstoer/docker-nodemcu-build.

TerryE, 2018-08-13:
Marcel, we're just having fun :smile: 

Gregor (@HHHartmann ) will sort out the scripts and he knows that he can use me if he does need any expert help.

I'll put the images on my ellisons,org.uk website for those that do want to use bare linux or cygwin and can't be bothered to do the builds, and I will do a simple webservice where you can post a ZIP of your source and get back an LFS image by return.

HHHartmann, 2018-08-14:
@jmd13391 there are also a few steps to get an LFS image built with docker so your waiting is not too long.
In recent dev there is a local/lua dictionary.
Put your lua files in there and issue the following commands:
Assuming that you have a float build or an integer build configured in user_config.h .
Your build command ``` docker run .... :/opt/nodemcu-firmware``` but appended by ``` bash``` (there must be a space before ```bash```)
You will get a prompt like

> root@56d64166f5e4:/opt/nodemcu-firmware#

enter the following three commands:
```
cd tools
build
exit
```
Ignore the error at the end of the build command.
This generates the file ```local/fsflash.img``` the name is a bug I fixed in #2459 
Once the bug is merged it will be in ```local/fs/LFS.img```

jmd13391, 2018-08-15:
I feel the point of this "issue" has been adequately expressed. Feedback has been provided to my satisfaction. The seed has been planted... I will close this issue. Thank you all.

-Joe

closed 2018-08-15 by jmd13391

TerryE, 2018-08-15:
@jmd13391 Joe, you are an active Lua developer who is interested enough to give constructive feedback.  IMO, I would welcome you tracking other issues and giving feedback based on your own experience as a developer.  

Thanks again :smile: Terry

HHHartmann, 2018-09-07:
@jmd13391 my instructions above have been wrong.
It must be ```make``` instead of ```build```. Sorry for that. Corrected it above

You are also aware of Terrys online service and the new version of docker(not sure if it is live already)?

jmd13391, 2018-09-07:
@HHHartmann Thanks! I will give that a whirl...

I am aware of Terry's online luac.cross online service and plan to incorporate it into the quick-start doc "flow".

However, I'm not sure what you mean by "new version of docker" (other than the obvious "a docker update is available for download...")

-Joe

marcelstoer, 2018-09-07:
I guess he means the LFS-image building feature in my NodeMCU Docker image that landed last night thanks to Gregor's hard work.

poorandunlucky, 2019-03-07:
I don't want to create a new issue, but there's really 0 documentation on the internet regarding the LFS...  I think a lot of work probably went into it, and compared to other features, it might be largely underused because, as the original author wrote, it's a bit "unintuitive" to the new user...

The documentation page is great, and it was a great read, and I'm really happy to be part of the NodeMCU family, but I still don't really know what the LFC can do for me in real life, and it's a bit unclear to me at this point why the option to create a LFS is available on the nodemcu-builder.com website if I can only use the LFS by compiling my firmware using the Linux toolchain, or the Docker image...

nwf, 2019-03-07:
@poorandunlucky LFS is not a generic "double your RAM because you turned it on" kind of thing.  The LFS holds compiled lua files, which will be specific to your project, much as the SPIFFS holds lua files.

All that said, https://github.com/cmukgb/ctfws-timer-iot/blob/master/mklfs.sh might be of interest: this is how that project builds its LFS image.  https://github.com/cmukgb/ctfws-timer-iot/blob/master/mkspiffs.sh lands the LFS image in the SPIFFS image that gets flashed down, and https://github.com/cmukgb/ctfws-timer-iot/blob/master/init2.lua knows to update LFS whenever it has a `luac.out` file in SPIFFS.  So, having been flashed, the system will land the LFS image and reboot.  If you want to turn this flow into documentation, please open a PR. :)

TerryE, 2019-03-07:
>  I still don't really know what the LFS can do for me in real life

If you use it you go from 44 Kb RAM for Lua code and data to upto 256KB for Lua code and read-only data and 44Kb for application data.  
  
> it's a bit unclear to me at this point why the option to create a LFS is available on the nodemcu-builder.com website if I can only use the LFS by compiling my firmware using the Linux toolchain, or the Docker image...

Read the whitepaper in the docs if you want to understand the details.  The firmware builds a binary to run on the ESP that is LFS-enabled.  You still need to compile Lua code _on the dev_ machine into  an LFS image.  At the moment you still need to do this compile of the compiler yourself.  On Windows, you've got a choice of four toolchains to do this: WSL, Cygwin, MinGW and native MSVS.  Your pick.