nodemcu/nodemcu-firmware #2466: had some idea with IR Community(1838

godchenmeng, 2018-08-23:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and **DELETE THE REST**.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2018-08-23:
See #1010 -> closing

closed 2018-08-23 by marcelstoer

FrankX0, 2018-08-23:
Like @marcelstoer commented, this is not the place for these kind of questions.
But maybe this helps (no personal experience though): https://github.com/mykh/ESP8266-IR-Remote-Receiver