nodemcu/nodemcu-firmware #2477: Unable to Update gpio.pulse "count" Parameter

jmd13391, 2018-09-05:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Update gpio.pulse.build() `count=` parameter with gpio.pulse:update.

### Actual behavior
gpio.pulse:update does not alter `count=` parameter.

### Test code

```Lua
gpio.mode(1, gpio.OUTPUT)
gpio.mode(2, gpio.OUTPUT)
    
pulser = gpio.pulse.build( {
  { [1] = gpio.HIGH, [2] = gpio.LOW, delay=250000 },
  { [1] = gpio.LOW, [2] = gpio.HIGH, delay=250000, loop=1, count=20, min=240000, max=260000 }
})
    
pulser:start(function() print ('done') end)

pulser:update(1, { count=99999 })

```

This bug reports is associated with [Stack Overflow 52176021](https://stackoverflow.com/questions/52176021/is-it-possible-to-set-a-continuous-loop-with-gpio-pulse)

@pjsg, Heads-Up :-)

### NodeMCU version
DEV Branch as of 20180904 / SDK 2.2.1(6ab97e9)

### Hardware
Amica ESP8266 Dev Board with ESP12-E

8<------------------------ END BUG REPORT -------------------------------------

pjsg, 2018-09-05:
Hmm. As I recall, it was never supposed to update the count parameter -- it isn't clear to me what exactly it would mean. However, I do see the need for an infinite pulse chain (and, in fact, that is how I use it myself). I don't recall how I did it -- I'll take a look this weekend and update the ticket.

jmd13391, 2018-09-05:
Hi @pjsg, 

> _I do see the need for an infinite pulse chain (and, in fact, that is how I use it myself)._

I have opened #2478 as a new feature request.

> _As I recall, it was never supposed to update the count parameter_

Consider modifying gpio.pulse.build and gpio.pulse:update to accept a `count=` value of -1 indicating an infinite loop.

BTW/FYI: Initially setting `count=0` with gpio.pulse.build results in a panic.

> _I don't recall how I did it_

Consider refining the gpio.pulse.build and gpio.pulse:update descriptions and code examples to include generating an infinite pulse chain.

> I'll take a look this weekend and update the ticket.

Anything you can do to expedite a resolution to this issue would be greatly appreciated. This requirement (infinite pulse train) is on my development critical path. I will have to look at other MCU's if I am unable to get the ESP8266 gpio.pulse module to meet this requirement.

-Joe

pjsg, 2018-09-05:
My ESP8266 systems are all at home, and I'm on the road at the moment so I can't get to it until this weekend.

pjsg, 2018-09-08:
In your case, just setting count to the maximum value (-1 -- the value is treated as an unsigned 32 bit integer) will let it run for 68 years. If you need longer, then you can use a nested loop as below

```
pulser = gpio.pulse.build( {
  { [1] = gpio.HIGH, [2] = gpio.LOW, delay=250000 },
  { [1] = gpio.LOW, [2] = gpio.HIGH, delay=250000, loop=1, count=20, min=240000, max=260000 },
  { loop=1, count=20 }
})
```

In this example above, it should run for 20 * 20 cycles before stopping.

jmd13391, 2018-09-08:
Outstanding... Thanks!

Any chance you can update the gpio.pulse docs with this info?

jmd13391, 2018-09-08:
@pjsg , Just so I'm sure I got this right...

```
gpio.mode(1, gpio.OUTPUT)
gpio.mode(2, gpio.OUTPUT)

pulser = gpio.pulse.build( {
  { [1] = gpio.HIGH, [2] = gpio.LOW, delay=500 },
  { [1] = gpio.LOW, [2] = gpio.HIGH, delay=500, loop=1, count=-1, min=400, max=600 },
  { loop=1, count=-1 }, { loop=1, count=-1 }, { loop=1, count=-1 }, { loop=1, count=-1 }
})

pulser:start(function() print ('pulse done') end)

print("just waiting around...")

Would generate a pair of 1Khz. square waves for how long?

```

pjsg, 2018-09-08:
It would be 0.001 seconds * 2^32 * 2^32 * 2^32 * 2^32 * 2^32  -- this is *much* longer than the lifetime of the universe so far. Each loop has its own counter and once they finish, then the counter is reset for the next iteration. i.e. this is equivalent to:
```
for (uint32_t i1 = 1; i1 > 0; i1++) {
  for (uint32_t i2 = 1; i2 > 0; i2++) {
    for (uint32_t i3 = 1; i3 > 0; i3++) {
      for (uint32_t i4 = 1; i4 > 0; i4++) {
        for (uint32_t i5 = 1; i5 > 0; i5++) {
           emit one cycle()
        }
      }
    }
  }
}
```
Just a single extra `{ loop = 1, count=-1 }` as step 3 would be over 500 million years.

closed 2018-09-08 by jmd13391

pjsg, 2018-09-08:
I'll update the docs -- thanks.