nodemcu/nodemcu-firmware #2479: Why is there no tmr.wdclr in dev-esp32?

syntroniks, 2018-09-06:
### Expected behavior
`=tmr.wdclr()`
### Actual behavior
`=tmr.wdclr() stdin:1: attempt to call field 'wdclr' (a nil value)`
### Test code

### NodeMCU version
dev-esp32 45be28cf60901513ed8ec72bb66bd8db36240d7a

### Hardware
ESP32 CORE / DevkitC V4

Some examples doing http requests feed the watchdog periodically as well as file download snippets from ESPlorer. Is this feature missing? Has it changed so much in ESP32 that a new methodology is required?

For long-running functions, is it expected that the watchdog will be fed at appropriate times in C code instead?

jmd13391, 2018-09-06:
From the docs...

> 
> **_In general, if you ever need to use this function, you are doing it wrong._**
> 
> The event-driven model of NodeMCU means that there is no need to be sitting in hard loops waiting for things to occur. Rather, simply use the callbacks to get notified when somethings happens. With this approach, there should never be a need to manually feed the system watchdog.

Web (http) functions are no exception. Take "baby bites" so you don't deplete the WDT. Bad things start to happen when you extend the WDT beyond 15 ms. Ex:

> The WiFi subsystem is maintained by background tasks that must run periodically. Any function or task that takes longer than 15ms (milliseconds) may cause the WiFi subsystem to crash. To avoid these potential crashes, it is advised that the WiFi subsystem be suspended with wifi.suspend() prior to the execution of any tasks or functions that exceed this 15ms guideline.

syntroniks, 2018-09-06:
Thanks for the quick reply. I hadn't seen that snippet on wifi resource requirements.

In this case, I'm concerned that the default config is:
```  │ │                   [*] Interrupt watchdog                                                                    │ │
  │ │                   (300) Interrupt watchdog timeout (ms)                                                     │ │
  │ │                   [*]   Also watch CPU1 tick interrupt                                                      │ │
  │ │                   [*] Initialize Task Watchdog Timer on startup                                             │ │
  │ │                   [*]   Invoke panic handler on Task Watchdog timeout                                       │ │
  │ │                   (8)   Task Watchdog timeout period (seconds)                                              │ │
  │ │                   [*]   Watch CPU0 Idle Task                                                                │ │
  │ │                   [*]   Watch CPU1 Idle Task```
With the possible exception of the task watchdog being increased by me. Is this something that should be addressed?

closed 2018-09-07 by syntroniks

maciejmiklas, 2019-03-15:
I have the same problem, watchdog gets triggered, but increasing "Task Watchdog timeout period" did not change a thing. 

> Any function or task that takes longer than 15ms (milliseconds) may cause the WiFi subsystem to crash
GTTP GET takes more then 15 ms, but I cannot suspend WiFi

What do we have to do in order to get running HTTP GET request? In my case it works fine with precompiled firmware, it just does not when I compile it from sources....