nodemcu/nodemcu-firmware #2481: Add more/better references for LFS infrastructure

HHHartmann, 2018-09-07:
### Missing feature
Better and more links to new LFS image build infrastructure (online service and Docker)
At least add/rework
- Quick start guide does not mention online service and docker.
- the "Compiling Lua on your PC for Uploading" in upload.md mentions LFS at the very end but even I over-read it the first time reading. The section should be restructured and clearly show LFS as favorized way to do it. Also mention the different ways to get an lfs image.
- In "Building the firmware" the LFS section should be more of a teaser and in short explain what to do, As it is in the other sections too.

### Justification
Start using LFS would be easier 

### Workarounds
Find the bits by digging deeper.

jmd13391, 2018-09-07:
Hi @HHHartmann, Along these lines, please expand on your https://github.com/nodemcu/nodemcu-firmware/issues/2457#issuecomment-412867070. I am getting ready to make a small contribution to the NodeMCU Quick Start docs and have been unable to get the process you call out in that comment to work. Specifically, there is no "build" script in the tools folder. The closest script I could find is "pr-build.sh" and along with it being inappropriate for a entry level build, it fails with errors changing to \opt\app in addition to having an undefined TRAVIS_BUILD_DIR variable.

Your feedback on this would go a long way to me sorting-out the Docker build process under Windows.

-Joe

marcelstoer, 2018-09-08:
Gregor, I started working on something in my fork two nights ago: https://msr-nodemcu.readthedocs.io/en/feat-doc-quickstart/en/getting-started/
Source: https://github.com/marcelstoer/nodemcu-firmware/blob/feat/doc-quickstart/docs/en/getting-started.md
Feedback and contributions welcome.

marcelstoer, 2018-09-13:
@TerryE @jmd13391 @HHHartmann 80% of what I've had in mind is now available in my new "Getting Started" at https://msr-nodemcu.readthedocs.io/en/feat-doc-quickstart/en/getting-started/. The "Quick start" in the LFS whitepaper was moved over.

It has it shortcomings but I think it's still a major improvement. WDYT?

marcelstoer, 2018-09-14:
> It would be easier to do them against a PR.

That was my plan if you guys agree that this should land in upstream. So, it's fine if you hold back with the detailed comments until it's up for PR. If I get more encouraging feedback in the next let's say 12h I'll create a PR which ideally might then land just before the master snap.

jmd13391, 2018-09-14:
@marcelstoer Let 'er rip! We can fine tune we go along. Nice work!

TerryE, 2018-09-14:
@marcelstoer \@ 20,000ft this is a very good document and well placed in the hierarchy.  I have reviewed it and will add the review comments once you raise the PR.  

(Incidentally my preferred way to do this is to view the page without a page style and cut and past the rich text into Writer, then turn on mark-up so I can do the comments and suggested corrections using in-place WYSIWYG redlining, before then transcribing the comments back into the MD file.  If anyone has a better methods of marking up markup files, then I would be grateful for any suggestions.)

>  I'll create a PR which ideally might then land just before the master snap.

My main concern is that 12 hrs will turn into 24 + review cycles and merge will add another 3-4 days delay before we push to master.  I am taking a few weeks break next Friday, so I doubt that the next batch off LFS enhancements which are stalled and an unmerged PR won't make this cut so won't get released into dev for maybe another month.  My preference would be to make the dev->master cut now as it is well overdue.  Since this is a documentation-only change and really only impacts the  readio site view, surely we could easily push it to master after the dev->master cut, and this would allow me to release the functional improvements in parallel.

marcelstoer, 2018-09-15:
PR ready at #2487

PR #2487 (2018-09-17): Add Getting Started page
Fixes #2481, #2431

As discussed earlier in several other issues. Ideally this should land rather quickly and then immediately be pulled over to master to fill the (LFS) documentation gaps.
closed 2018-09-17 by TerryE