nodemcu/nodemcu-firmware #2482: http module does not always correctly decode chunked response

heythisisnate, 2018-09-10:
### Expected behavior
When decoding a HTTP response using the http module, the module should respect the _case insensitive_ `Transfer-Encoding: chunked` header in the response to determine if the response will be chunked.

[RFC 7230](https://tools.ietf.org/html/rfc7230#section-3.2) states that header fields are case insensitive.

### Actual behavior
Chunked responses are decoded correctly if the response includes the header `Transfer-Encoding: chunked`, but does _not_ decode the response correctly if the header is in all lowercase characters: `transfer-encoding: chunked`

### Test code
This is difficult to test in isolation because a server is required. In this case, I've encountered this issue with a server I do not control that for whatever reason returns a _raw_ response like this (notice the all lowercase transfer-encoding header):
```
% curl -i --raw -H "Accept: application/json" http://example.com/api/state

HTTP/1.1 200 OK
Content-Type: application/json;charset=utf-8
Date: Sun, 09 Sep 2018 23:19:22 GMT
transfer-encoding: chunked
Connection: keep-alive

13
{"pin":8,"state":0}
0
```

The numbers `13` and `0` here are the delimiters of the the first (and only) chunk. When this request is made using the HTTP module, the chunked response is not decoded correctly and these control characters erroneously end up in the response:

```Lua
http.get('http://example.com/api/state', "Accept: application/json\r\n", function(code, response)
  print(response)
end)
```
the above prints:
```
13
{"pin":8,"state":0}
0
```
### NodeMCU version
master

### Hardware
NodeMCU with ESP8266

### Proximate cause
After digging into the source, this appears to be a straightforward fix. I believe the problem lies here:
https://github.com/nodemcu/nodemcu-firmware/blob/8181c3be7aed9f0a0ceb73ac8137c1a519e8a8e9/app/http/httpclient.c#L417

Not being familiar or experienced with C programming, what's the best way to make this comparison case insensitive?

heythisisnate, 2018-09-10:
This fixed my problem: #2483 
It really should be a totally case-insensitive comparison though. Sorry I don't know how to do that efficiently in C.

marcelstoer, 2018-09-10:
> the http module doesn't really implement http. It needs a rewrite.

The "upstream" source (no longer connected, though) at https://github.com/Caerbannog/esphttpclient doesn't seem to be well maintained either.

PR #2483 (2018-09-12): Decode chunked responses case insensitve headers
Fixes #2482 

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

HTTP Module: Correctly decode chunked responses if `Transfer-Encoding` header is all lowercase.
closed 2018-09-12 by marcelstoer