nodemcu/nodemcu-firmware #2488: ds18b20.c slower than a lua version

eyaleb, 2018-09-17:
8<------------------------ FEATURE REQUEST ------------------------------------
I use a lua module to read a ds18b20. Loading the module takes longer than the actual reading so I tried to use the ds18b20.c module. I found it to be slower to read (yet it saves the lua module load time). Reading using lua takes 20ms/sensor, using the c module takes 45ms/sensor, loading the lua module takes about 40ms (just once).

I suspect that (much of) this may be due to the fact that it converts the ow address between formats when it should really accept/return a raw format. Users can then convert to any other format if they need it. Furthermore, much of the returned values I do not use, only the temperature is required.

Another issue (just for me?) is that the "callback after a convert delay" does not match my usage. I read all the sensors, then start a convert, then go to sleep for a minue or more. This is not supported by this module, yet it is trivial to make it do so.

### NodeMCU version
Latest dev branch.

### Hardware
Any raw esp8266 (esp-07, esp-12 etc.) module.

### Missing feature
1) separate the convert action from the reading.
2) use raw ow address format throughout.
3) return less information [maybe we need read_temp()]

### Justification
1) It makes it more flexible.
2,3) more efficient, important when battery powered.

### Workarounds
This is just a rough hack for my testing.
I adjusted ds18b20.c such that `ds18b20_lua_read()` just calls `ds18b20_lua_readoutdone()` (remove the convert and timer part) and added a small function do start a convert.
`ds18b20_read_prev` should really be an argument to `ds18b20_lua_read()`.

8<------------------------ END FEATURE REQUEST --------------------------------

```c
--- app/modules/ds18b20.c       2018-09-16 17:27:31.982076936 +1000
+++ ../local/dev/ds18b20-20180916-el.c  2018-09-17 11:29:05.518465181 +1000
@@ -62,6 +62,8 @@
 int ds18b20_table_ref;
 static int ds18b20_table_offset;
 
+static int ds18b20_read_prev = 1;                      // read first, then convert
+
 static int ds18b20_lua_readoutdone(void);
 
 // Setup onewire bus for DS18B20 temperature sensors
@@ -168,6 +170,11 @@
                ds18b20_device_search = 1;
        }
        
+       if (ds18b20_read_prev) {
+               ds18b20_lua_readoutdone();
+               return 0;
+       }
+
        os_timer_disarm(&ds18b20_timer);
        
        // perform a temperature conversion for all sensors and set timer
@@ -246,6 +253,8 @@
 static int ds18b20_lua_readoutdone(void) {
        
        lua_State *L = lua_getstate();
+
+       if (!ds18b20_read_prev)
        os_timer_disarm(&ds18b20_timer);
        
        ds18b20_device_index = 1;
@@ -292,10 +301,18 @@
        ds18b20_timer_ref = LUA_NOREF;
 }
 
+static int ds18b20_lua_convert(lua_State *L) {
+       // perform a temperature conversion for all sensors
+       onewire_reset(ds18b20_bus_pin);
+       onewire_write(ds18b20_bus_pin, DS18B20_ROM_SKIP, 0);
+       onewire_write(ds18b20_bus_pin, DS18B20_FUNC_CONVERT, 1);
+}
+
 static const LUA_REG_TYPE ds18b20_map[] = {
        {       LSTRKEY( "read" ),                              LFUNCVAL(ds18b20_lua_read)              },
        {       LSTRKEY( "setting" ),                   LFUNCVAL(ds18b20_lua_setting)   },
        {       LSTRKEY( "setup" ),                             LFUNCVAL(ds18b20_lua_setup)             },
+       {       LSTRKEY( "convert" ),   LFUNCVAL(ds18b20_lua_convert)   },
        {       LNILKEY, LNILVAL                                                                                                }
 };

```

eyaleb, 2018-09-17:
I did a bit more reading of related issues (in this repo) and I get the impression that this module is an orphan, and others that asked for features got no satisfaction. Due to the slowness I do not use it now so I am closing this issue.

closed 2018-09-17 by eyaleb

TerryE, 2018-09-17:
@eyaleb, Thanks for closing this.  IMO, the C library is moribund.  

I have my own DS18B20 module written in Lua and which calls uses the OW library directly.  The main latency constraints are caused by the bit diddling of the OW protocol itself.   My module caches the OW addresses and on a minute interval timer issues a convert(all) and arms a second one-shot timer a second later to loop round the known thermometers reading back the temperatures and then once collected sends the dataset back to my HA system.  

Using LFS, the module is always loaded, so there isn't any other delay.  Why bother using C when there is absolutely no advantages in doing so?