nodemcu/nodemcu-firmware #2489: Local docs

cwrseck, 2018-09-17:
### Expected behavior
Some sort of indexed local documentation.

### Actual behavior
No indexed local documenation.

When trying to build the NodeMCU-2.2.1 docs I found that there's no Makefile
target or build system.  I work a lot off-line, so a local set of docs is
essential; and Markdown, in any of its incarnations, isn't especially helpful.

Sphinx is the obvious choice, but needs .rst or .txt files as input, and
apparently can't handle .md. Pandoc can convert .md to .rst, and building
a set of Sphinx documents from there is straightforward (barring a few minor
formatting details) but it all seems a bit much for a new user.

One solution might be for Pandoc to collect the .rst files into a single PDF;
unfortunately the conversion goes via Latex, which is fine if it works, but
a pig to debug if it doesn't.

Once I get a Sphinx build system working I'll see if it's worthwhile; does
anyone have any better ideas?

(This is intended only for actual master releases - obviously on-line docs
are the only approach to the dev branch.)

Will

marcelstoer, 2018-09-17:
It's all there already. Just install MkDocs (e.g. via `pip`) and run `mkdocs serve`: https://github.com/nodemcu/nodemcu-firmware/blob/master/CONTRIBUTING.md#writing-documentation

closed 2018-09-17 by marcelstoer

cwrseck, 2018-09-18:
Perhaps document the use of MkDocs somewhere?

Will