nodemcu/nodemcu-firmware #2495: App conversion to LFS HOWTO

joysfera, 2018-09-21:
### Missing a LFS conversion HOWTO
I have read through the https://nodemcu.readthedocs.io/en/dev/en/lfs/ but I am not sure how to convert an existing NodeMCU application to LFS one. It would be great if you could add a small paragraph to the LFS docs explaining what to do with current typical NodeMCU application that consists of a main.lua file that calls dozens of small files using `dofile("")` that in turn call other smaller lua files using another `dofile("")`'s.

I assume that the plain code in main.lua is to be converted to a function to be callable from LFS, so perhaps by wrapping the whole main.lua into a `function main()` would work. Then `LFS.main()` called from init.lua instead of typical `dofile("main.lua")` would do the job.

But how to convert the typical structure of hundreds nested `dofile("other.lua")` calls to LFS?

Thanks!

joysfera, 2018-09-21:
Hmmm, so if I have a file named "a.lua" that contains the following two lines:
```lua
function who() return "world" end
print("hello", who())
```
and I'm calling it from "main.lua" using `dofile("a.lc")` currently, then compiling both main.lua and a.lua into LFS plus replacing that `dofile` with `LFS.a()` would work properly, i.e. it would print "hello world"? I doubt it as it would have to define the function automagically. The LFS documentation says just this about using the `LFS`:

> LFS.someFunc(params) does exactly what you would expect it to do: this will call someFunc with the specified parameters

I feel like all the "submodules" need to be wrapped in functions first to be callable from LFS, like this:
```lua
function a()
  function who() return "world" end
  print("hello", who())
end
```

nwf, 2018-09-21:
@joysfera I think you have the right idea already, but in case you want to see some more worked examples, you might look at https://github.com/AcmeTensorToys/esp-lua_core/commit/373be38a103b307f9e7fb1d59506368c305e2b01

TerryE, 2018-09-22:
How about something like the following in _init:
```lua
do
  local ROdofile, index = dofile, node.flashindex
  function _G.dofile(fn)
    local f
    if fn:sub(-3) == '.lc' then 
      f = index(fn:sub(1,-4)) 
    elseif fn:sub(-4) == '.lua' then 
      f = index(fn:sub(1,-5)) 
    end
    if type(f) == 'function' then 
      return f() 
    else 
      ROdofile(fn)
    end
  end
end
```
Note that `ROdofile` and `index` are upvalues to new global `dofile()` .  These are fast and storage efficient.  NB, this is just a brain dump in a taverna and still needs debugging.

HHHartmann, 2018-09-22:
Great idea @TerryE . I would prefer to first check for existing files and then have a look at LFS to have the same behavior as in require and to enable overriding LFS with *.lua or *.lc files.
Also loadfile would be helpful to make it complete.

Would be a great enabler to quickly get existing projects to work with LFS.

TerryE, 2018-09-23:
> replacing that `dofile` with `LFS.a()` would work properly, i.e. it would print "hello world"? I doubt it as it would have to define the function automagically.

Sorry, I didn't answer the actual Q. Yes, this would work, and is the simplest and best way to do it. 😊

TerryE, 2018-09-23:
The scoping rules as just the standard Lua scoping rules. As far as the Lua VM is concerned any modules  LFS  are just normal compiled files in memory. It's just that they are in RO memory, and therefore not updateable or GCable. They are referenced via `node.flashindex()`, and when assigned to any Lua variable, global or local, the variable is of type function and is callable. `LFS.f2()` will always refer to the function in LFS, just like `string.sub()` refers to the `sub` function in `string`, and this doesn't stop you having a `fred.sub()` or a `_G.sub()`.

Incidentally IMO there are _very few_ times that you need to use the global environment _G. Locals are faster, take up less memory and are collected by the GC when no longer required. If you need persistent locals, then read up on how to use upvals and look at some of my coding examples.

If you are running out of RAM always enumerate `_G` and `debug.getregistry()`. Ask yourself what any variables are doing there and why. In the case of the latter, if you have any hanging functions that havent been node.compiled (and therefore had the line no info stripped out) then just call them interactively, e.g.   `debug.getregistry()[4]()`, and since most take a userdata as the first argument, this will throw an error, telling you what file / line number is in the function.

joysfera, 2018-09-23:
As for the scoping rules of `dofile("aaa')` vs `LFS.aaa()`, I'm going to try it out as I can't believe it.

As for running out of RAM: I have inserted tons of `print(node.heap())` everywhere in my app to I found out that  600 bytes of JSON can inflate into 5 kB of occupied memory after sjson.decode(). I am far from being able to judge if Lua indeed needs that much memory for the table with data decoded from a JSON file. An uneducated guy could think that if the JSON itself is 600 bytes that the decoded data should not take much more - definitely not 9 times, but that's just a feeling not based on anything. I definitely know nothing about memory organization of nested tables in Lua.
I have everything compiled, running uncompiled Lua files would render my application unusable right from the start, I am afraid. With the compiled one it starts with more than 20 kB of free memory but there are several async things invoked by various timers (like reading from DS18B20) and when they all happen at the same time by bad luck the memory is exhausted immediately. If they don't occur at the same time then there's more than 10 kB of free heap and it recovers quickly.

I feel like I can either give up on developing with Lua or try going the LFS route. NodeMCU for ESP32 having similar feature set does not seem to happen anytime soon so this way is closed right from the  start. But for LFS I'd have to develop a new OTA system, incompatible with the existing one, so that's another pain and obstacle in the way.

TerryE, 2018-09-24:
Petr, read my FAQ and some of the references.  Keyed arrays are quite expensive in terms of RAM. 24 bytes extra per entry IIRC, and that's as well as any variable costs.  And always explicitly set the optional `depth` parameter in any JSON encode / decide.

With LFS, everything is compiled, just on your host PC. `node. compile()` dumps line number info. Host compile and `string.dump()` doesn't. Having line numbers in your error tracebacks is pretty essential, IMO.  The code overhead is 1 byte per non-blank source line, and that's in flash using LFS. 

My LFS apps typically start with 44 kB free.  Using locals and upvals where ever possible and only using keyed tables where necessary makes a big difference.  My home automation front end controls 4 relays, 14 DS18B20s, 2 flow meters plus the command decode for these without any problems.

Doing the compiles on the host and downloading LC files was the main saving.

joysfera, 2018-09-24:
Thanks for the valuable information! Will start compiling on my PC then to get useful backtrace. Will also try specifying the JSON depth to see if it helps.
My system does similar things to yours, plus it reports measured values to "cloud" using `http.get()` regularly (every minute), runs an internal web server serving static files and JSON files generated on-the-fly (whenever a web client requests it), and also listens to MQTT commands and reports any changes to the "cloud" via MQTT as well. When any of these four-five features is executed alone it all works OK but when they happen to occur at the same time the whole thing tends to crash due to out of memory.
I also recently moved all my configuration to JSON files. So the application runs sjson.decode anytime it needs to know what to do because I try to not keep anything in memory so it needs to be loaded and decoded on-the-fly (like say names for the dozens of DS18B20 sensors). Maybe that resorting to JSON and reading it on-the-fly was not a good idea but I needed it for some other advanced features. Oh well...

vshymanskyy, 2018-10-15:
I think a good LFS Getting Started guide would be enough.
The existing docs (https://nodemcu.readthedocs.io/en/master/en/lfs/) are too detailed and confusing for beginners. There's no step by step guide how to use it out-of-the-box.

marcelstoer, 2018-10-16:
> There's no step by step guide how to use it out-of-the-box.

Not sure, I would argue there is. See https://nodemcu.readthedocs.io/en/master/en/getting-started/

vshymanskyy, 2018-10-16:
@marcelstoer I agree there is some info, but is it properly structured and usable out of the box?
I mean, an average newcomer would be overwhelmed with options, no simple "hello world" example, etc.
That's what's my comment about

vshymanskyy, 2019-10-11:
I'm not sure if anything was improved here... Need to re-check! ;)

joysfera, 2020-10-13:
I created the issue because I was unclear how to convert an advanced project consisting of many small lua files into the LFS world. You know, once you start missing free RAM you are suggested to split your large lua file into bunch of smaller files, so you go ahead and sooner or later end up with a pile of mini lua files that call each other using the `dofile()` or `require()`. Converting such mess into LFS without a complete rewrite back into a monolithic file was a mystery to me, so I asked.

I am explaining this so you know that simple LFS Hello world would not exactly help here.

But I no longer use NodeMCU so feel free to close this issue or do whatever you like with it, I don't need it anymore.

closed 2022-04-30 by stale[bot]