nodemcu/nodemcu-firmware #2496: ESP32 + OLED + CP2102 = WiFi not working

slepre, 2018-09-21:
Hi All,

I am experiencing two issues with a series of ESP32 with OLED modules and from different vendors (Heltec, TTGO) with or without LORA. And all have the same behavior

1. The USB is restricted to 74880 BPS.
2. WiFi does not work in any mode nor does it produce any error, the below code will always produce: Scan failed:	failure on scan done

Since these modules have a jst 1.25 connector, I plugged in a 18650 battery (some Arduino threads talk about power issues) also swapped the USB cable and PC...
I tired different firmware and loaded them in multiple ESP32 ; all the ones with OLED+CP2102 have the same problem. The exact same firmware on a LOLIN32/CPH340 works fine and the same code below does return all the APs in the vicinity, just as expected.

```Lua
wifi.mode(wifi.STATION,true)
wifi.start()

wifi.mode(wifi.STATION,true)
wifi.start()
wifi.sta.scan({ hidden = 0 }, function(err,arr)
  if err then
    print ("Scan failed:", err)
  else
    print(string.format("%-26s","SSID"),"Channel BSSID              RSSI Auth Bandwidth")
    for i,ap in ipairs(arr) do
      print(string.format("%-32s",ap.ssid),ap.channel,ap.bssid,ap.rssi,ap.auth,ap.bandwidth)
    end
    print("-- Total APs: ", #arr)
  end
end)
```
As far as I remember, the Heltec devices came with some arduino code scanning WiFi that did work ; I don't think it is a HW issue.

As I also had a bunch of modules enabled, I am rebuilding a FW from the latest release with the minimum modules but I have low hopes.

Thanks very much in advance,
./Steph

slepre, 2018-09-21:
Update: I do get an error when I compile the FW. The folder is the same as on Github.
Thanks,
./Steph
```bash
make[2]: Entering directory '/home/steph/nodemcu-firmware-esp32/build/modules'
/home/steph/nodemcu-firmware-esp32/components/modules/component.mk:16: warning: undefined variable 'CONFIG_LUA_MODULE_SPI_MASTER'
/home/steph/nodemcu-firmware-esp32/components/modules/component.mk:16: warning: undefined variable 'CONFIG_LUA_MODULE_I2C_HW_SLAVE'
/home/steph/nodemcu-firmware-esp32/components/modules/component.mk:16: warning: undefined variable 'CONFIG_LUA_MODULE_WIFI_AP'
/home/steph/nodemcu-firmware-esp32/components/modules/component.mk:16: warning: undefined variable 'CONFIG_LUA_MODULE_I2C_HW_MASTER'
/home/steph/nodemcu-firmware-esp32/components/modules/component.mk:16: warning: undefined variable 'CONFIG_LUA_MODULE_WIFI_STA'
/home/steph/nodemcu-firmware-esp32/components/modules/component.mk:16: warning: undefined variable 'CONFIG_LUA_MODULE_WIFI_COMMON'
CC build/modules/u8g2.o
```

slepre, 2018-09-22:
Update....
I tried a few other FWs and am still experiencing the same problem with OLED devices...
On the other hand, I've successfully loaded the FW on a Wemos ESP32 D1 which is also based on the CP210x and does operate at 115,200 Bps. As it was 'assigned' a different COM port than the OLED modules ; I suspect the CP210x is not configured in the same way on the two types of boards.
Thanks,
./Steph

marcelstoer, 2018-11-03:
One for https://nodemcu.readthedocs.io/en/latest/en/support/ -> closing

closed 2018-11-03 by marcelstoer