nodemcu/nodemcu-firmware #2502: System reset when initialising u8g2 (e-paper) display

FrankX0, 2018-10-02:
### Expected behavior
Initialised e-paper display.
### Actual behavior
Reset during initialisation of the dipslay, when using the `u8g2.ssd1607_200x200` constructor.
The reset seems to be caused by long delays in `u8x8_d_ssd1607_200x200.c`, causing the watchdog to expire. For example [here](https://github.com/olikraus/U8g2_Arduino/blob/7f2fc42af3d01fdfe2cc19320bdcfe693dd2b20d/src/clib/u8x8_d_ssd1607_200x200.c#L462).
@olikraus: we might need your help for this one.

### Test code
```Lua
local cs  = 8 -- GPIO15, pull-down 10k to GND
local dc  = 4 -- GPIO5
local res = 0 -- GPIO16

spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)
-- we won't be using the HSPI /CS line, so disable it again
gpio.mode(cs, gpio.INPUT, gpio.PULLUP)

disp = u8g2.ssd1607_200x200(1, cs, dc, res)
```
### NodeMCU version
Latest master or dev version.

### Hardware
Independent from ESP8266 platform. Not necessary to connect the actual display to reproduce.
Other (SPI) displays work correctly, or at least do not cause the reset.

devsaurus, 2018-10-02:
> The reset seems to be caused by long delays in `u8x8_d_ssd1607_200x200.c`, causing the watchdog to expire.

These `U8X8_DLY(250)` delays are translated to `os_delay_us()` in [our HAL layer](https://github.com/nodemcu/nodemcu-firmware/blob/62f48fe17079dab4dd4e12897cb128a9524caff8/app/platform/u8x8_nodemcu_hal.c#L51). That's nasty of course, but busy waiting is kind of inherent to u8g2's API.
A quick and dirty workaround would be to feed the WDT with `system_soft_wdt_feed()` after os_delay. This avoids the reset but stalls all other tasks like wifi and lwip networking.

FrankX0, 2018-10-03:
This workaround fixes the resetting during the long delays.

devsaurus, 2018-10-06:
Thanks for cross-checking, @FrankX0. I've added this change to #2500.

closed 2018-10-08 by FrankX0