nodemcu/nodemcu-firmware #2510: Text not visible on e-paper display

FrankX0, 2018-10-08:
With the reset issue resolved (#2502) the e-paper display based on ssd1607 can actually be used now. Unfortunately it does not behave as expected.
### Expected behavior
Text displayed on e-paper display.
### Actual behavior
Text is shown shortly, after which it is cleared again.
Also tested with the Arduino implementation, which shows correct behavior.
So it seems to be caused by the Nodemcu implementation (related to the used long delays?).
### Test code
```Lua
local cs  = 8 -- GPIO15, pull-down 10k to GND
local dc  = 4 -- GPIO5
local res = 0 -- GPIO16

spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)
-- we won't be using the HSPI /CS line, so disable it again
gpio.mode(cs, gpio.INPUT, gpio.PULLUP)

disp = u8g2.ssd1607_200x200(1, cs, dc, res)
disp:clearBuffer()
disp:setFont(u8g2.font_helvB18_tr)
disp:drawStr(0,20,"Hello World!")
disp:sendBuffer()
```
### NodeMCU version
Latest dev version.

### Hardware
Independent from ESP8266 platform.

FrankX0, 2018-10-09:
When comparing the SPI byte-stream of the Nodemcu implementation against the one produced by the Arduino implementation, I see that at the end of a `sendBuffer()` an extra byte (0x20) is being transmitted by Nodemcu.
This extra byte is generated by [U8X8_END_TRANSFER()](https://github.com/olikraus/U8g2_Arduino/blob/7f2fc42af3d01fdfe2cc19320bdcfe693dd2b20d/src/clib/u8x8_d_ssd1607_200x200.c#L354).
Don't know yet why....

FrankX0, 2018-10-09:
No, a single byte.
This byte is sent 1750ms after the correct byte-stream. So all the way at the end when the chipselect is disabled.

devsaurus, 2018-10-09:
Ok, thanks for confirmation. Then we're probably looking at a buffering issue.

For performance improvement reasons, the [hal layer for nodemcu](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/platform/u8x8_nodemcu_hal.c) buffers the bytes internally and postpones the spi transfer as long as possible.
The buffer is flushed when either happens:
- change of DC line
- transfer is ended

In your particular case, the last 0x20 remains buffered while the delays tick away. The U8X8_END_TRANSFER finally pushes the byte onto the bus. This is probably not according to the intention of this init sequence:
1. Run all the commands, completely
2. Wait
3. De-assert CS

I see two options to fix this:
- Buffering takes place only within [U8X8_MSG_BYTE_SEND](https://github.com/nodemcu/nodemcu-firmware/blob/509be837a839f86fc12514abdc568aa61e1805d7/app/platform/u8x8_nodemcu_hal.c#L254). Simple to implement but potentially deteriorates the overall performance.
- A [delay](https://github.com/nodemcu/nodemcu-firmware/blob/509be837a839f86fc12514abdc568aa61e1805d7/app/platform/u8x8_nodemcu_hal.c#L51) must trigger [`flush_buffer_spi()`](https://github.com/nodemcu/nodemcu-firmware/blob/509be837a839f86fc12514abdc568aa61e1805d7/app/platform/u8x8_nodemcu_hal.c#L207) first. This will be tricky to implement.

You could try the first option by inserting `flush_buffer_spi( hal );` right before the `break;` on [L272](https://github.com/nodemcu/nodemcu-firmware/blob/509be837a839f86fc12514abdc568aa61e1805d7/app/platform/u8x8_nodemcu_hal.c#L272).

devsaurus, 2018-10-11:
@FrankX0 I could observe the delayed sending like you described. The linked PR corrects this and I can see with LA that the 0x20 byte is sent *before* the delay phase.
However, without the hardware I can't verify that the display issue is actually fixed. Can you please check the patch on your setup and report the result?

FrankX0, 2018-10-13:
Hi Arnim. I tried the patch which resolves the delayed sending of the last byte, but unfortunately it does not resolve the issue I have with the display.
So I have to continue my search. It seems it might not be related to the Nodemcu implementation after all.

Code1110, 2018-11-06:
I can confirm, that the driver is working correctly with my Waveshare SSD1607 (1.54inch, 200x200). I built the firmware today with the cloud builder service from dev branch. But I am using a different setup (don't know, if this makes a difference):

```lua
cs  = 1
dc  = 4
bus = 1

spi.setup(bus, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)

disp = u8g2.ssd1607_200x200(bus, cs, dc)
```

Pin mapping (on WeMos D1 mini):
```
DIN -> D7 (MOSI)
CLK -> D5
CS  -> D1
DC  -> D4
RST -> RST
```

Sometimes its required to draw twice, to make the text appear sharp.

FrankX0, 2018-11-09:
Unfortunately I still have issues with my e-paper display.
As you can see from the attached [movie](https://github.com/nodemcu/nodemcu-firmware/files/2567294/IMG_5758.zip), the text is printed first with high contrast, after which a refresh takes place which results in a lower contrast end result.

@Code1110 do you see similar behaviour?

Code1110, 2018-11-10:
@FrankX0 yes, I see similar behavior. After the screen turns black and white again, the text appears in low contrast. To work around this I have to manually issue `disp:sendBuffer()` a second time. Besides this I have trouble writing text with good contrast in the lower left corner. According to https://github.com/olikraus/u8g2/wiki/gallery#5-feb-2017-ssd1607-200x200-eink, the extra refresh seems intended:

> There will be a black/white flicker for each refresh. This is intended and required for best contrast.

Sadly, I broke my display yesterday while writing way too long text. :(

FrankX0, 2018-11-20:
As the found behaviour is not caused by the Nodemcu implementation I will close this issue.

closed 2018-11-20 by FrankX0