nodemcu/nodemcu-firmware #2511: Optimising DNS memory usage

TerryE, 2018-10-09:
### Expected behavior

The DNS library should make minimal use of `.bss` RAM to avoid exhausting valuable RAM

### Actual behavior

The library uses an allocation system design for high-RAM general purpose and as a result `.bss.dns_table` takes up 1120 for no good reason.  (See `app/mapfile` of any build).

### NodeMCU version

All post LWiP

### Comments

Current build defaults to the following DNS configuration entries

```C
#define DNS_LOCAL_HOSTLIST            0
#define DNS_LOCAL_HOSTLIST_IS_DYNAMIC 0
#define DNS_TABLE_SIZE                4
#define DNS_MAX_NAME_LENGTH           256
#define DNS_MAX_RETRIES               4
#define DNS_MAX_TTL                   604800
#define DNS_MSG_SIZE                  512
#define DNS_SERVER_PORT               53
#define DNS_TMR_INTERVAL              1000
#define DNS_DEBUG                     LWIP_DBG_OFF
#define DNS_DOES_NAME_CHECK           1
```

We should define `DNS_LOCAL_HOSTLIST_STORAGE_PRE` to `static const` so that if a static host list is defined then it goes into flash rather than RAM.  We should also expose the following DNS defines to `user_config.h` to allow developers the option to overide these and to define their own static hosts enties:  
```C
/*
 * Use the following setting to include a static hosts list in your build.
 */ 
//#define DNS_LOCAL_HOSTLIST 1
//#define DNS_LOCAL_HOSTLIST_INIT = {{"entry1",0x123}, {"entry2",0x456}}
```
  
DNS_TABLE_SIZE is the size of the local DNS cache.  This cache allows you to resolve the same hosts multiple times without going back to the DNS server.  Resolving more hosts than this will result in some repeat resolutions going back to DNS.  So if you only refer to 2 external hosts or you cache your resolved IPs as Lua variables in your application, then set this to 2 as your application won't really make use of this cache.
```C
/*
 * By default 4 DNS entries are cached; this can be varied by setting the following
 */
//#define DNS_TABLE_SIZE                4
```

The current DNS use a fixed hostname length of `DNS_MAX_NAME_LENGTH` (256) so the cache allocates 4×256 bytes in  valuable `.bss` for host names.  Given that DNS resolution is very low frequency operation then it makes sense to replace the fixed `char name[DNS_MAX_NAME_LENGTH]` in `struct dns_table_entry` by `char *name;` and updating the assignment at `dns.c:930,933` 

```C
  pEntry->arg   = callback_arg;
  namelen = LWIP_MIN(os_strlen(name), DNS_MAX_NAME_LENGTH-1);
  MEMCPY(pEntry->name, name, namelen);
  pEntry->name[namelen] = 0;
```
by
```
  pEntry->arg   = NULL;
  namelen = os_strlen(name);
  if (namelen > DNS_MAX_NAME_LENGTH-1)
    return ERR_VAL;
  char *p = os_realloc(pEntry->name, namelen+1);
  if (!p)
    return ERR_MEM;
  else
    pEntry->name = p;
  os_strcpy(pEntry->name, name);
  pEntry->arg   = callback_arg;
```
and this will allocate string storage JiT, and use a lot less than the current preallocated 1K space.

If there is general support then I will raise the PR.

nwf, 2018-10-09:
SGTM.  Will returning ERR_MEM fail the lookup or just the caching operation?  The latter would be better, naturally.  Moreover, is it possible to hook into the Lua GC and dump the cache if it hasn't been hit in (say) more than two full collections?  (From your description, though I've not read the code, it sounds like the cache doesn't honor DNS TTLs?  That might be motivation to force it out after some full collections regardless of its hit recency....)

TerryE, 2018-10-10:
@nwf Nathaniel, I think that this is over-engineering this. A typical hostname will be what maybe 30 bytes long.  The code detects and reports malloc failures, but if your app can't find the odd 30 bytes then you are buggered anyway.  OK, so this algo will "leak" maybe 100 bytes in a typical usecase where your Lua app uses DNS resolution, but this is still 900+ of RAM saved overall.

As to TTL, etc. this is standard LwIP code so yes it does, but to KISS, I suggest that we leave the stale Cstr as is.  If this change gets too complicated, then I won't bother to do it and we're 900 bytes worse off.

nwf, 2018-10-10:
Fine.

TerryE, 2018-11-18:
This is a low risk patch so I will try to get this PR in before the next master cut.

PR #2763 (2019-05-20): Use dynamic memory for cached DNS names
Fixes #2511

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [ ] ~The code changes are reflected in the documentation at `docs/*`.~  N/A

Adds about 1K RAM to the free heap.
commit 036bff56 (2019-05-20): Use dynamic memory for cached DNS names (#2763)

Fixes #2511
closed 2019-05-20 by TerryE