nodemcu/nodemcu-firmware #2512: cron fails when setting minute and hour in mask

NicolSpies, 2018-10-09:
### Expected behavior
If the following cron mask ``` "33 16 * * *" ``` is specified it is expected that the cron scheduler will fire every day at 16h33.
### Actual behavior
Nothing happens. 
If the mask ``` "33 * * * *" ``` is specified it fires correctly every hour on 33 minutes past the hour.  If the mask ``` "* 16 * * *" ``` is specified it fires correctly at 16h00. 

### Test code
```Lua
cron.schedule("33 16 * * *", function(e) print("cron fired") end)
```
### NodeMCU version
branch: dev
commit: https://github.com/nodemcu/nodemcu-firmware/commit/3886d2c765b7ce0c7f22a2455fc673289a23cf4d

### Hardware
NodeMCU dev board

nwf, 2018-10-09:
Can you check the return value of cron.schedule?  The logic in `cron_handle_time` looks right from a quick skim, so I wonder if this is "the event didn't parse" or "the event parsed but didn't fire".

nwf, 2018-10-09:
@NicolSpies I think you have that backwards; if `cron.schedule` is returning `nil`, then the event didn't parse.  It might be helpful if you could report the second return value (which should be an error string).
I am unable to reproduce this on my local build using this test:

```
rtctime.set(1539118070)
h, he = cron.schedule("* 20 * * *", function() print("h fire") end)
m, me = cron.schedule("48 * * * *", function() print("m fire") end)
b, be = cron.schedule("48 20 * * *", function() print("b fire") end)
```

I see all three of the callbacks firing as they should.

NicolSpies, 2018-10-10:
@nwf I used your test and updated the time and masks for valid fire after few minutes. The schedule where minutes and hours are set, does not fire. 
Return value for b:  `userdata: 0x3fff0a90` , return value for be:  `nil`

nwf, 2018-10-10:
@NicolSpies I am mystified.  Are you equipped to build your own firmwares?  The `cron` module doesn't do a whole lot of debug reporting in the existing tree.

NicolSpies, 2018-10-11:
@nwf Sorry not equipped to build own firmwares.

closed 2018-10-11 by NicolSpies

marcelstoer, 2018-10-11:
-> ff44b2f015ae03fdbb86a6c214ac8e78d4edda27