nodemcu/nodemcu-firmware #2521: Object-based file API missing in dev-esp32 branch

tomsci, 2018-10-15:
The object-based file API appears to be missing in the dev-esp32 branch, is re-adding that blocked on #1613? Or was the esp32 code simply forked before the object-based API was added? Either way I have a lot of code written for the esp8266 which will need some major rewrites.

Happy to attempt bringing the esp32 file.c up to date but will only do so if it's not completely doomed or if there isn't a rewrite dropping imminently.

tomsci, 2018-10-15:
Merging master in seems to have worked ok. PR to follow.

devsaurus, 2018-10-16:
IIRC, the ESP32 branch forked before, and just the vfs layer was cherry-picked but not the object-based file API. Probably for reasons Johny @jmattsson described in #1613.
IMO the points there are still valid and maybe we tackle such restructuring at some point in the future. But for the time being: +1 from my side for aligning with the esp8266 API.

tomsci, 2018-10-16:
Definitely don't want to stand in the way of future cleanup refactors. Thanks for the background info, coming into the ESP32 port from fresh it can be quite tricky to work out what's what!

devsaurus, 2018-11-11:
Guess this one is resolved.
Note that I ported additional commits from dev branch to fully support the new API. SPIFFS integration and vfs layer were missing bits and pieces.

PR #2524 (2018-10-29): Add object-based file API to dev-esp32 branch
Merged in from master@c708828bbe853764b9de58fb8113a70f5a24002d minus
some VFS changes and similar. No actual new code added.

Fixes #2521.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.
_No docs changes made_
closed 2018-11-11 by devsaurus