nodemcu/nodemcu-firmware #2525: ESP32 Ethernet Support

Lir10, 2018-10-17:
Hello , 

I have tried to search for ethernet support in the documentation but couldn't find anything.

is their something i'm missing?

TerryE, 2018-10-17:
That because the ESP32 modules only have a WiFi NIC and not a wired one :smile:

closed 2018-10-17 by TerryE

Lir10, 2018-10-17:
@TerryE Ofcourse i have connected Ethernet to the chip.

It looks like IDF does support this feature , how come it's not available in nodemcu?

https://github.com/espressif/esp-idf/tree/master/examples/ethernet/ethernet

Lir10, 2018-10-21:
@marcelstoer @devsaurus @TerryE Someone can explain something about this?

marcelstoer, 2018-10-21:
In #1010 we stipulated that

> We will specifically close....Any enhancement requests where the originating poster is not proposing to resource the development him or herself

So, Terry acted in accordance with our policy even if his comment may not be 100% accurate 😜

Lir10, 2018-10-21:
@nwf @marcelstoer Thanks for the response! I will try to look if i can help to implement this into nodemcu

heythisisnate, 2019-05-03:
@Lir10 @TerryE @nwf @marcelstoer has any progress or further research been made on Ethernet capability for the ESP32?

I'm now very interested in this because I'm developing a new product using the ESP32 and I want to incorporate ethernet. My company would be willing to sponsor development of this capability into the firmware if we can find the right person to do it. Any thoughts?