nodemcu/nodemcu-firmware #2527: BME280 module only supports I2C?

JBFUK, 2018-10-20:
So far as I can tell it seems that the BME280 module only supports I2C whereas the BME280 itself supports both I2C and SPI.  It would be great to see SPI support added.  It would then be possible to use CS feature and run multiple devices on the same bus.

nwf, 2018-10-21:
Not to argue against SPI support, but in the interim... the BME280 already supports having two on the same bus, and you might be able to get away with an I2C mux (e.g. TCA9548A) if you need fewer than 17 of them (and don't mind having to steer the mux between samples, of course).  If #2465 lands, too, you'll be able to drive a lot more I2C busses, further multiplying the number of sensors. :)

JBFUK, 2018-10-21:
Sure, but the device supports SPI and that would be my preference as
opposed to I2C and messing around with mux etc.  I don’t see why the
library shouldn’t give me the choice to use either option.

nwf, 2018-10-21:
As I said, I didn't mean any of what I said to argue against SPI support, but unless you're willing to do the work, I suspect someone's going to close this as per #1010.

JBFUK, 2018-10-21:
I don’t disagree that this isn’t really an ‘issue’, I just couldn’t see
anywhere to post a feature/enhancement request.

marcelstoer, 2018-11-03:
The only likely candidate for enhancements to that module is @vsky279 (its creator). I'm closing this as suggested but PRs are of course more than welcome.

closed 2018-11-03 by marcelstoer