nodemcu/nodemcu-firmware #2529: ESP32: http module

tomsci, 2018-10-23:
**Rationale:**

The IDF has a full HTTP stack included in `esp_http_client.h` and it would be nice to expose this instead of the old http.c/httpclient.c version. Currently there's no `http` module and I assume everyone is using `net` directly.

At the same time, I'd like to reconcile the `http` vs `net/tls` dichotomy, that is to say that the APIs in `http` stop being usable if your body data is large, because only the `net` module offers a multiple-callback model, meaning you have to rewrite your code using `net` and parse the HTTP manually if you hit memory limits using the old `http` module. I'd like to keep the existing simple `http.get()` API but also have a more net-like `on("receive")` style API available as well to gain the benefits of HTTP header parsing etc even when the payload is large.

Does this sound sensible? It may sound like a laundry-list of requests but I'm planning to do the work myself and submit a PR for it (I really need HTTPS support!) so I'm keen to get feedback before I disappear too far down the rabbit hole of implementation.

(I'm aware of #1622 but I don't really want to have to keep doing HTTP header parsing myself given there are APIs in the SDK just waiting to be used for this.)

nwf, 2018-10-23:
How wedded to the ESP32 is the `esp_http_client.h` implementation?  It would be nice to replace the, ahem, lackluster ESP8266 http client with what you develop as well.

tomsci, 2018-10-23:
> How wedded to the ESP32 is the esp_http_client.h implementation? It would be nice to replace the, ahem, lackluster ESP8266 http client with what you develop as well

It's wedded to the new SDK, so it would depend on whether that was ever backported on to the ESP8266. As I understand it, that's a major question for the project because right now the ESP32 code (and the SDK being used underneath) is quite badly diverged from the ESP8266 code in `master`. I don't know if there's progress being made there that's not immediately obvious, but my guess would be it's unlikely any time soon. Also the `esp_http_client` code is necessarily more complex and put my ROM over the default ROM size limit, so it might not be suitable for the more modest 8266 (but I don't have solid figures for its ROM budget so don't quote me on that, I'm just thinking aloud).

> If there's a broader interest, I could share a branch for people to look at. There are also a couple of items to settle on the Lua API.

I'd definitely be interested. I was debating API issues too (which is why I didn't get very far prototyping before deciding I should check no-one else was working on it!)

tomsci, 2018-10-23:
+1 to all that! spi_compat.lua is exactly the sort of model to aim for, IMO.

Having the `net` module use `lwip` and `http` using ESP's `tcp_transport` makes me a bit nervous though. Does one backend into the other? Having two completely separate TCP stacks running does not seem desirable but I admit I don't know much about the lower levels of firmware fit together.

Right now I'm worrying about how blocking `esp_http_client_perform` is, since one of the first "temporary" hacks I had to make to my ESP32 project was to entirely disable the task watchdog as being impossible to satisfy, and doing HTTPS GETs is multiple seconds slower than HTTP on the ESP32. I think I would probably hit the task watchdog with a single zero-byte HTTPS GET. Perhaps I need to get my head around the dual core nature of the ESP32 and offload all this onto the second CPU. Any pointers on that would be much appreciated, eg what's currently running on CPU 0 vs 1, what threading primitives are available - it's quite a lot to wade through coming to the project from cold.

tomsci, 2018-10-24:
My current workingish API, supporting HTTP and HTTPS.

```lua
conn = http.createConnection(url, http.GET, headers)
conn:on("headers", function(status, headers)
    for k, v in pairs(headers) do
        print("Got header", k, v)
    end
end)
conn:on("data", function(status, data)
    print("Got some data of length", #data)
end)
conn:on("finish", function(status)
    print("All done! status=", status)
end)
conn:request()

-- For compat, headers and callback optional
http.get(url, headers, function(status, response, headers)
    -- As per master
end)
```

Threading of the `esp_http_client_perform()` call I haven't yet thought of (although the callbacks use task_post like net.c so should be safe if/when that's in a different thread). Also haven't worried about keep alive requests, setting individual parts of the URL separately (both of which the underlying API supports), or verifying HTTPS certificates.

devsaurus, 2018-10-24:
Looks good :+1:

I'm still struggling with my current use case where I need to cope with big amounts (several MByte) of data, while the data sink is much slower than the source (i.e. the http server). The Lua code has to throttle the incoming data stream in order to avoid overflows. The `net` module enables this with `socket:hold()` / `socket:unhold()`.

With synchronous `esp_http_client_perform()` there's no option to pause the incoming stream. There are just 2 options when the callback is triggered for new data: Either buffer the data or discard it. Buffering will quickly flood the heap (or spiffs), of course.
I expect that `esp_http_client_perform` in asynchronous mode could be useful but I hit espressif/esp-idf#2624 for the time being.

> Threading of the `esp_http_client_perform()` call

I would avoid threading unless it's required to enable a particular feature or use case.

> although the callbacks use task_post

Two aspects here:
- Do you post **copies** of header and data content to the task queue?
  I understood that the event payload provided to `event_handler()` is invalidated after `event_handler()` returns (memory is free'd or potentially gets overwritten)
- How do you avoid flooding the task post queue when the server sends a larger number of headers or data? The queue can't hold more than 8 (`TASK_DEFAULT_QUEUE_LEN`) events at a time.

nwf, 2018-10-24:
@devsaurus Assuming Espressif is doing something standard, TCP's receive buffers and window management should provide back-pressure on the server, so there's less of a need to buffer the whole stream on the client.

devsaurus, 2018-10-25:
> TCP's receive buffers and window management should provide back-pressure

Agreed 👍 I just didn't find a way yet to build up this pressure from within `esp_http_client_perform()`. I'll need to look into the other functions.

tomsci, 2018-10-25:
> I would avoid threading unless it's required to enable a particular feature or use case.

Happy to, but how does one avoid falling foul of the task watchdog, given TLS negotiation seems to take fully half of the default 5 seconds (in my very rough testing)? On the ESP8266 there was a very clear requirement to never block for any length of time, otherwise wifi would break (or something) because of the lack of task preemption. With the new SDK can we ignore this now and program a blocking task if it's convenient to do so? If so the defaults in `make menuconfig` might need to be updated to disable the task watchdog?

> Do you post copies of header and data content to the task queue

At the moment yes I'm copying. And feeling guilty about the heap space being used.

> avoid flooding the task post queue when the server sends a larger number of headers or data

I hadn't thought of that (still not very familiar with all the APIs), yeah currently it'll fall over and error the connection if that happens. How does the net module handle that, it seems to use a similar pattern?

Sounds like it'd be best to do as you suggest and skip the task posting and the copying and dispatch the callbacks directly from the `http_event_handle_cb`? It'll certainly simplify the code a bit. Assuming the prolonged blocking isn't an issue that is.

devsaurus, 2018-10-25:
> With the new SDK can we ignore this now and program a blocking task if it's convenient to do so?

Blocking NodeMCU's task is ok and will not do any harm to other tasks. However blocking the CPU still is harmful. Blocking the NodeMCU task on esp8266 non-os sdk is equivalent to busy-looping the CPU.
The idle task watchdog is supposed to "resolve" CPU blocking but not task blocking. It's actually watching the idle task, not NodeMCU's task. I.e. you won't find watchdog service calls in NodeMCU code at all. Instead, NodeMCU's task is expected to relinquish the CPU soon enough so that the idle tasks can resume and service their watchdogs in time.
When digging deep enough into SDK code (drivers, networking, etc) you will always end up at some instance of RTOS queues, mutexes or similar. These don't poll until the requested resource is available but re-schedule the CPU to the idle task in case no other tasks are pending.

Bottom line: NodeMCU's task might be blocked for extended periods of time, but that won't harm other tasks like wifi, networking, or the idle task per se.

> At the moment yes I'm copying.

IMO it's necessary to copy when posting http client events to our task queue. Posting them comes at this cost of heap and code complexity.

> How does the net module handle that

This is where [@nwf's reasoning](https://github.com/nodemcu/nodemcu-firmware/issues/2529#issuecomment-432857033) gets important. The `net` module would also see several events being posted, but soon the TCP window is exhausted and TCP's inherent flow control forces the sender to pause. Only when an event is finally being worked on by the net callback, its payload is acknowledged back to the lwip stack (the call to `netconn_recv()`). The TCP window increases accordingly and allows for further packets from the sender.

Contrary to this handshaking, the http client receives **and** acknowledges the payload simultaneously, then dispatches the `event_handler`. It will continue to receive&acknowledge assuming that `event_handler` consumed the payload before returning. The race is lost sooner than later if the handler doesn't consume but buffers and posts to subsequent queues.

Disclaimer: I might be missing important details with my (limited) view :wink:

tomsci, 2018-10-27:
Thanks @devsaurus, that helps a lot. Am I right in my understanding that therefore there's no reason not to reinstate the `tmr.delay()` function, with fewer warnings in the docs and possibly a new implementation, since it's no longer so inherently 'bad' to do so? When somebody gets to it, of course.

> IMO it's necessary to copy when posting http client events to our task queue. Posting them comes at this cost of heap and code complexity.

When using `esp_http_client_perform()` synchronously though, the `http_event_handle_cb` callbacks will still be from the same task right? Since it doesn't seem to offload any of its work into a different task it just loops and calls back in the same context? Meaning copies and task posts are not necessary in that situation?

devsaurus, 2018-10-27:
> therefore there's no reason not to reinstate the `tmr.delay()` function

At least not a technical reason as far as I understood https://github.com/nodemcu/nodemcu-firmware/issues/1619#issue-191839813.

> the `http_event_handle_cb` callbacks will still be from the same task right? Since it doesn't seem to offload any of its work into a different task it just loops and calls back in the same context? Meaning copies and task posts are not necessary in that situation?

Yes, that's my view as well (and I expect the same to be true for the asynchronous mode).

Though with `esp_http_client_perform()` in synchronous mode the Lua task won't do anything else. E.g. events waiting in the task queue are not served in parallel to the http transfer. That's not bad by default, but can be a restriction for a more complex application.

tomsci, 2018-10-28:
I've pushed a first attempt to https://github.com/tomsci/nodemcu-firmware/commits/tomsci-http. In particular, docs are here: https://github.com/tomsci/nodemcu-firmware/blob/tomsci-http/docs/en/modules/http.md.

There are some unit tests in examples/httptest.lua

All feedback welcome! The docs could probably do with some more work, a few bits are C&P from the old docs and aren't totally relevant any more. The more advanced connection configs (timeout, redirection, async mode) aren't exposed yet, but all the basic stuff is present and working.

tomsci, 2018-10-28:
> Though with esp_http_client_perform() in synchronous mode the Lua task won't do anything else. 

Yes not sure how best to handle that without creating more additional problems or adding a lot of extra complexity. Perhaps it's best not to, at least until someone really needs it?

For the first attempt linked above, I've left it synchronous and blocking, and documented it as such.

devsaurus, 2018-10-28:
> I've pushed a first attempt

Great :+1: I'll have a look at it.

> Yes not sure how best to handle that without creating more additional problems or adding a lot of extra complexity. Perhaps it's best not to, at least until someone really needs it?

Guess my current project needs it. I'll check the requirements against your linked implementation.

devsaurus, 2018-10-29:
@tomsci your `tomsci-http` branch works very well, good work!

Since my use case needs some kind of asynchronous behavior (other callbacks have to be served while the http request is ongoing), I experimented with threading and task posting, resulting in commit https://github.com/devsaurus/nodemcu-firmware/commit/e65b90cc8fc5296f7fe6cae1978835e06a9f44bb on top of `tomsci-http` branch. It's not a polished implementation, but is intended to highlight the deltas.

Key points are
- Asynchronous operation is enabled by a simple, one-shot RTOS task.
- The `event_handler` posts without copying. This is possible since RTOS and Lua tasks handshake the 'ownership' of the event payload.
- Callbacks handshake by default unless they return `true`. This is a bit clunky but provides compatibility with your test code. This could be solved with a better API mechanism in the end.
- `ack()` method for explicit handshaking of the last event once the application is ready for new data.
- I kept the core `http_event_cb` untouched and built wrappers for synchronous and asynchronous invocation.
- `context:request()` got an optional completion callback parameter. Its presence is used to enable asynchronous mode (again, to keep compatibility). Plus it unrefs the context object.

I used this code to receive data streams (several MBytes) from a server while other Lua callbacks are triggered periodically. Parallel http requests should also be possible, though I didn't try that yet.

tomsci, 2018-10-31:
Nice. I like the handshaking of tasks to avoid copying and also provide flow control.

> Callbacks handshake by default unless they return true.

I think that's a reasonable model to use. Perhaps we should define a constant `http.DELAYACK` (which can be implemented as `true`, or whatever) for the callback to return, to make the intent more obvious? Might we want to limit the delayed ack functionality to just the `data` callback? I can't immediately think of a reason to delay any of the others, although I suppose it doesn't cost us anything to support.

Overall I think this is a good approach to take, I made a couple of comments on your commit (hopefully github doesn't lose them), any objections if I take it onto this PR branch and run with it?

devsaurus, 2018-10-31:
> Perhaps we should define a constant `http.DELAYACK` (which can be implemented as true, or whatever) for the callback to return, to make the intent more obvious?

Yes, good point. I didn't care for such aspects since the commit was meant to just demonstrate the async implementation. I'm happy with whatever improvements you think are useful to get it polished.

> Might we want to limit the delayed ack functionality to just the `data` callback?

Personally I don't see use cases that would require delaying by other callbacks.
I briefly considered limiting when I built the code but decided against additional logic because I regarded this as out of scope for the demonstrator.
In the end, if limiting improves clarity on the user/API side then I'd vote for the latter.

> any objections if I take it onto this PR branch and run with it?

None at all! I'm glad that you took the initiative for developing the `http` module. It will be a major step forward for the esp32 branch. Thank you very much!

tomsci, 2018-11-03:
I've pushed some changes to the [tomsci-http](https://github.com/tomsci/nodemcu-firmware/commits/tomsci-http) branch. In addition to @devsaurus's asynchronous request support:

* `async` is now an option passed in to `createConnection()`, rather than being decided based on the arguments to `request()`. It made things easier to know up front and to not have any way of mixing sync and async behavior within a single connection.
* Added options to createConnection to support buffer size, max redirections, certificate validation, and timeout (although timeouts don't seem to work, in my testing).
* `get()` and `post()` functions now take an `options` parameter like `createConnection()`, rather than just a headers table, meaning all the above can be configured for the simple functions too.
* `get()` and `post()` can be used synchronously or asynchronously, depending on whether a callback is specified.
* Added `http.DELAYACK` as discussed above. I haven't thoroughly tested this so if @devsaurus you could rerun your multi-megabyte multi-callback tests that'd be appreciated!
* Renamed `finish` callback to `complete` as it no longer exactly matched the semantics of the underlying `HTTP_EVENT_ON_FINISH` callback.
* Increased the http task stack to 4KB to avoid stack overflows processing async HTTPS requests.
* Added checks for hopefully most ways people can abuse the async API.

I think I'm about done with features support so if people are happy with the API ([docs here](https://github.com/tomsci/nodemcu-firmware/blob/tomsci-http/docs/en/modules/http.md)) then I'll go ahead and open a PR?

devsaurus, 2018-11-03:
> Added checks for hopefully most ways people can abuse the async API.

:smile: excellent

> so if people are happy with the API then I'll go ahead and open a PR?

Looking forward to having this merged soon!

tomsci, 2018-11-09:
Hmm, isn't GitHub supposed to auto-resolved this? Closing now.

closed 2018-11-09 by tomsci