nodemcu/nodemcu-firmware #2535: i2c.setup() support multiple buses

sgjava, 2018-10-29:
I saw [commit](https://github.com/nodemcu/nodemcu-firmware/commit/0bdcb815263966d50639f59e4525ff99ae4a49aa)  that allows u8g2 to support I2C buses other than hard coded 0. I pulled the latest dev branch and built locally, however i2c.setup() does not support any other value except 0. Am I missing something using the current dev branch. I'm trying to get 2 ssd1306 displays working on different I2C buses.

sonaux, 2018-10-29:
That's because I2C driver in dev branch currently does support only bus 0. PR #2465 is not merged yet (I'm still polishing it), but you can pull it from my fork, branch `i2c_FAST`.

sgjava, 2018-10-29:
@sonaux OK, I'll try that, thanks!

closed 2018-10-29 by sgjava

sonaux, 2018-10-29:
I just realized that you need to apply that [u8g2 HAL patch](https://github.com/nodemcu/nodemcu-firmware/commit/0bdcb815263966d50639f59e4525ff99ae4a49aa) you mentioned over PR  #2465, because it was merged after my last rebase.

sgjava, 2018-10-29:
Ah, good catch. I'll do that and report back.

sgjava, 2018-10-30:
OK, I did the following:
```
git clone --recurse-submodules -b i2c_FAST https://github.com/sonaux/nodemcu-firmware.git

wget https://raw.githubusercontent.com/nodemcu/nodemcu-firmware/0bdcb815263966d50639f59e4525ff99ae4a49aa/app/platform/u8x8_nodemcu_hal.c
```
and it compiled fine. However when I try to use to separate displays it stays with the last i2c.setup I did:

```
function init_display(id, sda, scl, sla)
  i2c.setup(id, sda, scl, i2c.FAST)
  local disp = u8g2.ssd1306_i2c_128x64_noname(id, sla)
  -- Set your font here
  disp:setFont(u8g2.font_courR18_tr)
  disp:setFontRefHeightExtendedText()
  disp:setDrawColor(1)
  disp:setFontPosTop()
  disp:setFontDirection(0)
  disp:clearBuffer()
  return disp
end

function str_display(disp, x, y, str)
  if str then
    disp:drawStr(x, y, str)
  end
end

 disp0 = init_display(0, main_config.sda0_pin, main_config.scl0_pin, main_config.sla0_addr)
str_display(disp0, 0, 0, "disp0")
disp0:sendBuffer()

disp1 = init_display(1, main_config.sda1_pin, main_config.scl1_pin, main_config.sla1_addr)
str_display(disp1, 0, 0, "disp1")
disp1:sendBuffer()

--  This displays on disp1!
str_display(disp0, 0, 20, "disp0")
disp0:sendBuffer()
```
using disp0 still displays to disp1. I'm not sure if I coded it correctly.

sonaux, 2018-10-30:
I tested your code, it behaves as you described. It looks like both displays use the same framebuffer. If you add clearBuffer() before str_display() then it works, but it clears display before putting new string.
I did not test this case... 
I tried now this straightforward code:
```
GPIO= {[0]=3,10,4,9,2,1,14,10,13,11,12,9,6,7,5,8,0}
main_config = {
sda0_pin = GPIO[2],
scl0_pin = GPIO[0],
sla0_addr = 0x3C,
sda1_pin = GPIO[2],
scl1_pin = GPIO[16],
sla1_addr = 0x3C
}
i2c.setup(0, main_config.sda0_pin, main_config.scl0_pin, i2c.FAST)
i2c.setup(1, main_config.sda1_pin, main_config.scl1_pin, i2c.FAST)
disp0 = u8g2.ssd1306_i2c_128x32_univision(0, main_config.sla0_addr)
disp1 = u8g2.ssd1306_i2c_128x32_univision(1, main_config.sla1_addr)
disp0:setFont(u8g2.font_6x12_tf)
disp1:setFont(u8g2.font_6x12_tf)

disp0:clearBuffer()
disp0:drawStr(0, 0, "disp0")
disp0:sendBuffer()

disp1:clearBuffer()
disp1:drawStr(0, 5, "disp1")
disp1:sendBuffer()

--  This displays on disp1!
--disp0:clearBuffer()
disp0:drawStr(0, 10, "disp0")
disp0:sendBuffer()
```
And result is the same. 
I will look into u8g2 library, may be there is something we can do.

sgjava, 2018-10-30:
I found a yucky work around doing it synchronized: 
```
function init_display(id, sda, scl, sla)
  i2c.setup(id, sda, scl, i2c.FAST)
  disp = u8g2.ssd1306_i2c_128x64_noname(id, sla)
  disp:setFont(u8g2.font_courR18_tr)
  disp:setFontRefHeightExtendedText()
  disp:setDrawColor(1)
  disp:setFontPosTop()
  disp:setFontDirection(0)
end

init_display(1, 2, 3, 0x3c)
disp:drawStr(0, 0, "disp0")
disp:sendBuffer()

init_display(2, 5, 6, 0x3c)
disp:drawStr(0, 0, "disp1")
disp:sendBuffer()

init_display(1, 2, 3, 0x3c)
disp:drawStr(0, 20, "disp0")
disp:sendBuffer()

init_display(2, 5, 6, 0x3c)
disp:drawStr(0, 20, "disp1")
disp:sendBuffer()
```
Keep in mind that init_display clears the display without having to clear the buffer, so the use case is to update the entire display in one operation. I'm already doing this with my weather station, so it will work for me. Ultimately you want two separate displays without having to reinitialize.

sonaux, 2018-10-30:
With reinitialization of i2c bus and display every time you need to update particular display - even old i2c driver will work :) 
Just use clearBuffer() instead of init_display() and redraw whole display before sendBuffer()  - works well (my test shows ~30 fps with node.CPU160MHz and i2c.FASTPLUS on ssd1306_i2c_128x64_noname display)

sgjava, 2018-10-30:
So after i2c.setup, call clearBuffer, draw stuff, sendBuffer?

sonaux, 2018-10-30:
There is no need to call `init_display` and all that stuff inside it more than once.
Just keep in mind that u8g2 framebuffer in esp8266 RAM is shared between displays and must be cleared before drawing picture for next display.

sonaux, 2018-10-30:
I tried different display drivers and they use different framebuffers, so if you :
disp0 = u8g2.ssd1306_i2c_128x64_noname(0, main_config.sla0_addr)
disp1 = u8g2.ssd1306_i2c_128x32_univision(1, main_config.sla1_addr)
So, it looks like framebuffer is shared only for displays with identical driver, or, maybe, identical frambuffer size.

sgjava, 2018-10-30:
In a way this might not be a bad thing since memory is limited. I'll try this when I get home tonight.

sonaux, 2018-10-30:
It seems that 2 displays and buffer issue is a recurring topic in [olikraus/u8g2].(https://github.com/olikraus/u8g2/issues?utf8=%E2%9C%93&q=is%3Aissue+multiple+buffer)

sgjava, 2018-10-30:
Yes, I see that. I suppose as long as the use case works with one buffer then it's OK. I'll streamline my code tonight.

sgjava, 2018-10-31:
This code works as expected. i2c.setup is required to switch displays:
```
disp0 = u8g2.ssd1306_i2c_128x64_noname(0, 0x3c)
disp0:setFont(u8g2.font_courR18_tr)
disp0:setFontRefHeightExtendedText()
disp0:setDrawColor(1)
disp0:setFontPosTop()
disp0:setFontDirection(0)

disp1 = u8g2.ssd1306_i2c_128x64_noname(1, 0x3c)
disp1:setFont(u8g2.font_courR18_tr)
disp1:setFontRefHeightExtendedText()
disp1:setDrawColor(1)
disp1:setFontPosTop()
disp1:setFontDirection(0)

i2c.setup(0, 2, 3, i2c.SLOW)
disp0:clearBuffer()
disp0:drawStr(0, 0, "disp0")
disp0:sendBuffer()

i2c.setup(1, 5, 6, i2c.SLOW)
disp1:clearBuffer()
disp1:drawStr(0, 0, "disp1")
disp1:sendBuffer()

i2c.setup(0, 2, 3, i2c.SLOW)
disp0:clearBuffer()
disp0:drawStr(0, 20, "disp0")
disp0:sendBuffer()

i2c.setup(1, 5, 6, i2c.SLOW)
disp1:clearBuffer()
disp1:drawStr(0, 20, "disp1")
disp1:sendBuffer()
```

sgjava, 2018-10-31:
I finally have two displays working with the i2c.setup hack!
[image]