nodemcu/nodemcu-firmware #2536: Add support for vl53l0x

adrianomelo, 2018-10-29:
### Missing feature
Hi guys,

I would like to know if it makes sense to add support for the [vl53l0x](https://www.st.com/en/imaging-and-photonics-solutions/vl53l0x.html) Time-of-flight (ToF) ranging sensor.

I am considering writing the module myself and submit as a pull request, but I don't have experience with the platform. How complicated should be this task? I bought the ADXL345 to experiment with a sensor using I2C and maybe have some reference to start.

Also, should it be possible to use two I2C sensors with nodemcu? (ADXL345 and vl53l0x)

Any advice/help is appreciated.

### Justification
Currenlty, there are no modules supporting a ToF sensor. Adding support for it would open new possibilities.

### Workarounds
People usually use libraries for Arduino when they need to use this sensor with the esp8266.

sonaux, 2018-10-29:
[How to write a C module](https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-How-to-write-a-C-module) is a good example.
And there are examples for other firmwares/frameworks that can be useful to you:
[Pololu Arduino library for VL53L0X](https://github.com/pololu/vl53l0x-arduino)
[VL53L0X library for Mongoose OS](https://github.com/dcato/arduino-pololu-vl53l0x)

And for two sensors - many sensors allow to choose between 2 i2c addresses via pin or soldered 0 Ohm resistor, so you can put 2 sensors on one I2C bus. Different sensors use different addresses and therefore can be used on same bus.

vl53l0x has address 0x29 and cannot be altered, so one I2C bus - one device (but you can use pin 5 XSHUT to select particular sensor, though I don't know how fast it will be waking up).
ADXL345 address is defined by pin 12 SDO/ALT ADDRESS and can be 0x1D or 0x53, so you can use 2 ADXL345 sensors on the same I2C bus.

nwf, 2018-10-29:
If the sensor is just I2C and does not require sensitive timing concerns on the host, writing Lua rather than C might be vastly simpler.  From a quick skim of the datasheet, this chip is a perfect candidate, as it uses an IRQ solely to indicate the availability of data to be read on I2C, not for any actual timing work.  An example, if you like, can be found in my driver for the CAP1188 driver, in https://github.com/AcmeTensorToys/esp-lua_core/tree/master/cap1188

As to multiple sensors despite the fixed unit address, see the recent #2465 work.

marcelstoer, 2018-10-30:
So, if Nathaniel is right (I have no reason to doubt) then you could contribute a Lua module to https://github.com/nodemcu/nodemcu-firmware/tree/master/lua_modules

HHHartmann, 2018-10-30:
in the specs it even says "Programmable I2C address"
See [AN4846: Using multiple VL53L0X in a single design](https://www.st.com/resource/en/application_note/dm00280486.pdf) from the above link for reference.
Seems like the programming must be done every time on powerup though.

closed 2019-11-08 by stale[bot]