nodemcu/nodemcu-firmware #2537: ws2812_effects expects GRB led strip, so RGB led strip colors are off

NorthernMan54, 2018-10-31:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and **DELETE THE REST**.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

ws2812_effects like halloween, fire display correct color

### Actual behavior

ws2812_effects halloween displays as blue and green

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
I looked at the source and it appears to hard coded as GRB, rather than having an option to declare RGB or GRB.

https://github.com/nodemcu/nodemcu-firmware/blob/c708828bbe853764b9de58fb8113a70f5a24002d/app/modules/ws2812_effects.c#L587

### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

I presume master as of Sept 15, 2018

NodeMCU custom build by frightanic.com
	branch: master
	commit: c708828bbe853764b9de58fb8113a70f5a24002d
	SSL: false
	modules: bit,color_utils,crypto,file,gpio,mdns,net,node,pwm,sjson,tmr,uart,websocket,wifi,ws2812,ws2812_effects
 build created on 2018-10-15 23:42
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

Nodemcu connected to RGB LED Strip

https://github.com/NorthernMan54/homebridge-mculed/blob/master/nodemcu/README.md

8<------------------------ END BUG REPORT -------------------------------------

HHHartmann, 2018-10-31:
There are also other types of color coding like for the [APA102 module](https://nodemcu.readthedocs.io/en/master/en/modules/apa102/) it is ABGR. "A" being the Intensity Level.
Would be nice to be able to use the effects for all kind of LED strips.
Maybe also rename the Module then.

marcelstoer, 2018-10-31:
Maybe @skycoders is willing to pick this up.

skycoders, 2018-10-31:
@HHHartmann @NorthernMan54 can you tell me which stripes you use? I have thought about the APA102 stripes but that requires some more abstraction and refactoring. With respect to WS2812 stripes, I have not seen ones which use a different coloring scheme so far, so I'd be interested in knowing which one you are using. Providing a configurable color order for the WS2812 protocol should be rather easy to do.

NorthernMan54, 2018-10-31:
The strips I’m using are these

https://www.amazon.com/Tape-Light-Foot-Light-Lengths/dp/B01LXTUJ3M

The original controller supplied with these have failed, and am in the process of building an ESP8266 based replacement controller.

skycoders, 2018-11-02:
ok, I have never seen stripes with that order but there are other reports for e.g. FastLED where people faced the same issue. I will see what I can do, time is a bit limited at the moment... keep you updated.

NorthernMan54, 2018-11-02:
Tks Konrad

I have worked around these, so no rush.

Tks again

HHHartmann, 2019-07-16:
@NorthernMan54 May I ask how you worked around this issue? I'm facing the same problem now.

konradhuebner, 2019-07-16:
@HHHartmann sorry I just don't find the time to generalize this. The easy way is to simply set the colors different. The API assumes that the pixel order processed by the strip is GRB, what most WS2812 strips do. If your strip requires RGB order, then simply swap the colors when handing them over to the effects API. It should then result in the correct colors being shown

HHHartmann, 2019-07-16:
Thanx for the reply. Your suggestions works where a color can be given, But not for the fire effect. But I looked at the code and found that the fire mode is the same as the flicker mode with preset parameters for color and flicker amount.

NorthernMan54, 2019-07-16:
I just didn’t use those functions with the colour functions.

HHHartmann, 2020-07-11:
Will be fixed by #3122 and #3158

NorthernMan54, 2020-07-11:
Tks @HHHartmann

closed 2020-07-25 by stale[bot]