nodemcu/nodemcu-firmware #2541: node.input does not work with telnet inplementations

HHHartmann, 2018-11-04:
---------- BUG REPORT -----------------------------------------
### Expected behavior
calling node.input several times with parts of a command should execute after all chunks of the command are passed.

### Actual behavior
Only passing the whole command in one call works.

### Test code
called the following code from the console. The working example prints "1234" as expected after the delay.

This works
```Lua
tmr.create():alarm(1000, tmr.ALARM_SINGLE ,function() node.input("print(1234)") print('sent input')  end)
```

This does not work
```Lua
tmr.create():alarm(1000, tmr.ALARM_SINGLE ,function() node.input("print(12") node.input("34)") print('sent input')  end)
```

### NodeMCU version
Current dev

### Hardware
ESP8266 dev board

HHHartmann, 2018-11-07:
There seems to be some concatenation or buffering because after executing the failing command above, the next command I enter fails with an errormessage

```
tmr.create():alarm(1000, tmr.ALARM_SINGLE ,function() node.input("print(12") node.input("34)") print('sent input')  end)
> sent input
>> =node.heap()
stdin:2: ')' expected (to close '(' at line 1) near '='
> 
```

Just expiremented a bit further and found that indeed only one node.input per callback is possible,

```lua
tmr.create():alarm(1000, tmr.ALARM_SINGLE ,function() node.input("print(12") print('sent input')  end)  tmr.create():alarm(1500, tmr.ALARM_SINGLE ,function()  node.input(")") print('sent input')  end)
```

immediately (after 1.5 seconds) returns the above errormessage.

So another try with just the cloaisng bracket in the second call actually executed the command:

```lua
tmr.create():alarm(1000, tmr.ALARM_SINGLE ,function() node.input("print(12") print('sent input')  end)  tmr.create():alarm(1500, tmr.ALARM_SINGLE ,function()  node.input("34)") print('sent input')  end)
```
prints 
```
> sent input
>> sent input
12
```

So the problem seems to be that the lexer (assuming this works sort of like normal compiler/interpreter) passes on the token too early (before it is finished off by the next chunk).
So passing on the token would need to be delayed until it is certain that it is finished (the next token or whitespace has started)

BTW:
It would be a pitty if this API would be removed, cause there are lots of examples out there using it together with node.output().

nwf, 2018-11-07:
What does `node.input()` get you over `pcall`?

HHHartmann, 2018-11-07:
With Node.input if it was working as in many examples I do not have to care about where the lua command is complete. 
With pcall i have to parse the input myself trying to find command boundaries. 
I could use newlines here but then the new feature of being able to process multi line commands would not work anymore. 

Besides it seems to have worked at some stage. Else all serial forwarders would never have worked.

nwf, 2018-11-07:
Far be it from me to tell you how to design your application, but piping unparsed, unchecked input into the Lua interpreter is probably a bad plan.  I guess it is a use case, tho'; fair enough.

I think serial forwarders work because they called `node.input` once per packet.

HHHartmann, 2018-11-07:
Well it is not my application but the current sample in the lua_examples\telnet directory of nodemcu-firmware.
A serial forwarder or in this case telnet works by sending a packet per character if typed at my speed at least. 

I also only plan to have a telnet session online during testing and debugging. (Should make me a note to disable afterwards) Its quite handy to develop with the esp in place rather than connected to a computer.

@TerryE You introduced the telnet. How did your test scenario lock? did something break since then??

HHHartmann, 2018-12-30:
The problem is that the input is added to an empty command buffer only.
Could I change the ```lua_put_line``` function to add the input to the existing one?
Maybe we need a switch to only add further input if the already existing input also resulted from a call to ```lua_put_line``` which would be reset on serial line input by ```readline```.

I see that ```lua_put_line``` is also used in CoAP, would we need to have a separate function for that usecase then?

HHHartmann, 2019-01-03:
@TerryE Terry what do you think? I really would like to see if I can get this to work.

Should I implement another method which allows processing characters as read from uart0? So a telnet session would work nicely. Maybe add a parameter to ```node.input()```? It also might be ok for all use cases to process special characters.

So I would buffer the string given by ```node.input()``` and process it in ```lua.c: readline()```

TerryE, 2019-01-03:
@HHHartmann Gregor I am more than happy for you to have a go. This is an open collaboration. You've got my email and the chat forum of you need more 1on1.

HHHartmann, 2019-01-16:
@TerryE Terry Thank you for renewing your offer for 1on1. I will greatfully use it.
But for now I have another question which I feel is closely related to this bug:

Up to now sending node.input is blocked if the buffer already has content. When I want to allow chunks in node.input should the input from uart and node.input get mixed? should they mutually block each other?
blocking each other seems good but also could block uart input indefinitely, shutting down this way of interaction.
Another way would be to have two buffers.

HHHartmann, 2020-01-27:
I at least want to set this to fixed when #2912 hits master

HHHartmann, 2021-01-26:
Tests on a current dev build show that the fragmentation part is working now but the command now is not executed until a newline is added to the input.
That makes sense, so I will close this as fixed.

closed 2021-01-26 by HHHartmann