nodemcu/nodemcu-firmware #2546: LFS images bigger than 64kbyte give problems after flashreload

albert607, 2018-11-06:
### Expected behavior
After flashing image > 64k no error/reboot

### Actual behavior
After.during flashing of images bigger than 64kbyte the system esp (Wemos D1) reboots and outputs rubbish and following info:

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 28372, room 16 
tail 4
chksum 0x4d
load 0x3ffe8000, len 2252, room 4 
tail 8
chksum 0x68
load 0x3ffe88cc, len 176, room 0 
tail 0
chksum 0xd3
csum 0xd3

### Test code
Create an LFS image >64kbyte with luac.cross. Then upload to Wemos and flash with node.flashreload

I tried with cloudbuild settings 128Kbyte LFS, 1024K Base SPIFFS, all rest free
Also tried building firmware from Linux with various LFS and SPIFSS base setting without succes.

### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: c708828bbe853764b9de58fb8113a70f5a24002d
	SSL: false
	modules: bit,cron,file,gpio,http,net,node,rtctime,sjson,sntp,somfy,tmr,uart,wifi
 build created on 2018-10-23 18:13
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)

### Hardware
Wemos D1 mini with 4Mbyte flash

TerryE, 2018-11-11:
@albert607,  I've put this one down to me.  I'll take a look and see if I can reproduce this.

TerryE, 2018-11-17:
Yup, this one is my bad.  I amended the relocation algo and missed an offset check.  I can easily fix this in `dev`.  The workaround for `master` is to limit the LFS size to 64K for now, and to use the (still to be fixed) `dev` version,  if the application needs a larger LFS region.  I'll push the fix to dev in a day or so.  I am visiting family this weekend.

TerryE, 2018-11-18:
@albert607, no need to wait.  You can just use the new `dev` branch as this has already removed the bug, but remember that this includes a new `luac.cross` as well.

The issue with was a piece of code that has been replaced with new `dev` version.  Let me explain.  The LFS image can be loaded at a base address that isn't know at the `luac.cross` compile time so all of the addresses in the image are relocatable.  The first version of LFS used a 16bit word offset + 16bit link to next address chain, and there was a problem where I was doing a short << 2 without explicitly casting it to an int.  

The new `dev` version GZIPs the image for storing as a file and expands the file into the LFS region.  This drops the file-size to be uploaded and stored in SPIFFS by ~60%.  The address chain format compress poorly and it was easier to replace this by a relocation word bitmap which gives better compression performance.  As a side effect the new version incidentally removed the bug.