nodemcu/nodemcu-firmware #2553: Lua VM state corruption / Anonymous function arguments are sometimes nil?

nwf, 2018-11-11:
### Expected behavior

Code of the form `f(1,2,3,function() [...] end)` will always pass a non-nil fourth parameter to `f`.

### Actual behavior

Sometimes that doesn't happen; that is, if `f` prints out its argument, it is sometimes `nil`.  Charmingly,
if I replace the invocation with
```Lua
local cb = function() [...] end
f(1,2,3,cb)
```
I have not yet seen it reappear.

### Test code

I'm not sure how to minimize this, as I can only tickle it occasionally.  I'm using my telnet implementation.  When I write to a file using the `file pwrite` command, it ultimately invokes
```Lua
withfat(s,"r+",fn,off,function()
  if file.write(ddat) == nil
    then s("ERR: Write error")
    else s("OK")
   end
  end)
```
where `withfat` is a helper that knows how to `open` and `seek` before calling the specified callback. Here, `s` is a function that will queue its string argument onto a FIFO for delivery to the telnet client connection; I do not believe the details are relevant.  `fn` is the name of the file under manipulation, and `off` is the offset for the write.  If I instrument `withfat` to print out its arguments, I usually see lines of the form
```
withfat	function: 3fff1e60	r+	init3.lua	2944	function: 3fff26e8
```
which is as expected.  On occasion, and always? on the first `file pwrite` (i.e., the one at offset 0),
I see instead
```
withfat	function: 3fff1e60	r+	luac.out	0	nil
```
I have yet to see this bug with the `local cb = [...]` replacement, but that might just mean its more rare rather than impossible.

Occasionally, I have hit WTD timeouts in this code, wherein the module will spit out the first part of a line (e.g. "withfa") and then hang until the WDT hits, which may be part of the same bug or a different issue entirely.
Of course, there's a lot of code between input and getting to this point; I cannot rule out, for example,
state corruption from a bug in the `crypto.fromBase64` function or SPIFFS.

I have seen this arise with everything in LFS or with the `telnetd` `file` module coming from SPIFFS, so I do not think that LFS or `luac.cross` are involved, though there may be a common bug in the two Lua compilers (cross and on-module).

### NodeMCU version

I'm on f5e68157a136b5fe97df82c4db5018b548753a46 built using @marcelstoer's wonderful docker tooling.

### Hardware

A bog standard AI-THINKER ESP-12 module.

pjsg, 2018-11-11:
Have you tried building a debug build (with all the asserts) and seeing what happens? 

Also, I have a faint memory of (something like) this happening in the past....

TerryE, 2018-11-11:
Looks like I am going to have to have a Lua day tomorrow!

nwf, 2018-11-11:
Unfortunately, when building with `DEVELOPMENT_TOOLS` `#define`d, `luac.cross` asserts with
```ASSERT@../lapi.c(448): L->top < L->ci->top```
which appears to be a Lua stack overflow?  But I've no real idea what's going on there.

Anyway, using a LFS image from a non-`DEVELOPMENT_TOOLS` build of `luac.cross` and a debug firmware image, I'm afraid the situation is only so enlightening.  The `nil` arguments are gone, but I now always? get a WDT reboot after the `withfat` part of its printout (that is, after the first argument to print, a literal string) when using the `withfat([...], function() [...] end)` form, and have yet to see anything amiss with the `local cb =` form.

TerryE, 2018-11-11:
@nwf, Nathaniel I'll have a look tomorrow as a I said.  One potential issue is explained in [PiL 24.2.1](https://www.lua.org/pil/24.2.1.html). When the VM calls a C library function is makes sure that there are 20 free elements on the stack to give some headroom for  pushing content.  However, for runtime efficiency there is no bounds checking on this: it is up to the library routine to issue the appropriate `lua_checkstack()` calls though some Lua API calls do this internally.  If the stack is growing beyond the safe headromm than the RTS will `realloc()` the stack to make more headroom.  If the C code doesn't then it can overrun the end of the stack. If it does but uses `StkId` pointers then these will be invalidated.   Lots of bear-pits.

TerryE, 2018-11-19:
@nwf, Nathaniel, 
> Unfortunately, when building with `DEVELOPMENT_TOOLS` `#define`d, `luac.cross` asserts with
```ASSERT@../lapi.c(448): L->top < L->ci->top```
which appears to be a Lua stack overflow?  But I've no real idea what's going on there.

The best thing here is to enable development tools in the `lua` make only by uncommenting a couple of commented out compiler directives that make this happen -- though swap `-Og` for `-O0`.  You also need a remote gdb in your toolchain because with this enables, any assert firing will throw you into the remote debugger so you can walk up the stack to figure out why the assert was fired.

TerryE, 2019-11-28:
I am a little uneasy about letting this sort of issue from a committer lapse and close.

HHHartmann, 2019-11-28:
@TerryE so did I. Thanks for reopening.

nwf, 2020-11-22:
Still believed to be an issue.  Perhaps not in 5.3, but we never did have a very reliable test.

closed 2022-04-30 by stale[bot]