nodemcu/nodemcu-firmware #2554: A new Lua C API to reduce temporary memory usage and simplify buffer management

tomsci, 2018-11-12:
### Background

RAM is generally the most limited resource on the ESP8266 and ESP32, as evidenced by the RAM-saving changes made to the Lua core like ROtables and ROM string optimizations. One scenario where RAM is still spent unnecessarily (albeit temporarily) is when allocating temporary buffers for data that eventually will be exposed to Lua code as a `string` value. The common pattern is:

* Use `luaM_malloc()` to get a raw malloc'd memory region
* Pass this pointer to whatever native APIs will populate it
* Call `lua_pushlstring()` to make a Lua string object from the data
* Call `luaM_freemem()` on the pointer (note, this is regularly [done wrong](https://github.com/nodemcu/nodemcu-firmware/issues/2377#issuecomment-431107518))

This pattern is present throughout NodeMCU, and involves an extra temporary copy of the data due to there being no way to transfer ownership into `lua_pushlstring()`, which necessarily always copies the data. There are variations using `luaL_Buffer` but they still require an extra copy of the data in the case where the data size is known up front. While this is normally not a huge problem, with large allocations and fragmentation an unnecessary copy can make all the difference.

### Proposal

I propose an additional API in `lua.h` to eliminate the need for copying the data when converting it to a Lua string:

```c
// Get a new buffer of size len, same semantics as luaM_malloc
void *lua_newbuf(lua_State *L, size_t len);

// Realloc the buffer (if necessary)
void *lua_reallocbuf(lua_State *L, void *buf, size_t newlen);

// Convert it into a string value and push onto the stack, taking ownership
void lua_pushbuf(lua_State *L, void *buf);

// Or free it
#define lua_freebuf(L, b) lua_reallocbuf(L, (b), 0)
```

The above pattern then becomes:

* Use `lua_newbuf()` to obtain a malloc'd region
* Pass this around as before
* Call `lua_pushbuf(L, buf)` to _convert_ the buffer to a string, _or_, call `lua_freebuf(L, buf)` to dispose of the buffer without making a Lua value out of it.

### Implementation

The trick is that `lua_newbuf()` makes sure to allocate enough extra space for a `TString` structure prior to the returned pointer, which allows `lua_pushbuf()` to not have to copy the data and instead use that extra space to in-place construct a TString for it.

The changes already made to the `luaS_` APIs to support readonly strings means that the Lua core changes required to support this new API are quite small. It also has the benefit of eliminating a common misuse of the raw `luaM_*()` APIs, which hopefully we can discourage direct use of as a result. The buffers can track their own size internally, so there is no need for the caller to do so, unlike for correct use of `luaM_freemem()`, meaning the API is simpler and less error-prone than the current pattern.

I've already prototyped this because I had a project that was marginal on memory usage and became completely blocked, due the amount of temporary memory required to convert native byte arrays into Lua strings. Adding this API and updating a couple of key modules to use it saved enough memory to get the project working. If people are in favour of the change I'll raise a PR.

TerryE, 2018-11-12:
@tomsci, Tom I think that this approach is a mistake.  I will explain why.  The Lua API has evolved the way it has for sound reasons.  In order to ensure the integrity of the `G[L}->strt`, it make no assumptions about the long term value of the contents of a supplied string constant.  Also TStrings must to unique, 

-  so if you push a buffer where there is already a copy in the string table then it resolves this via lookup and the TValue on the stack refers to the existing TString.
-  As you discuss, if it is a new string then the TString comprises a single allocate region which is the string value _prefixed_ by its TString header.

So you either don't need to allocate space for string or you need to alloc `sizeof(TSting) + N` bytes.  Either way its is best to leave the calling C code get rid of its temporary string.

As you have mentioned elsewhere we have enough trouble getting library developers to use the Lua memory allocators correctly anyway.  My advice to library developers is to make sure that there have read and understood PiL before starting, but most seem to take another (often flawed) library example as a framework and tweak it to their needs. These aren't written a true Lua way.  

This approach takes us further away from standard Lua, and it opens up yet more bear-pits for unsuspecting developers.  Ultimately it at best saves an extra malloc/free pair, and in most cases some reduction in temporary RAM use.  I've been trying to thing of a usecase where this would genuinely make a material difference in terms of peak RAM and the one is large buffer assembly.  

The `luaL_Buffer` system is GC aware: the buffer is build up as Lua string chunks which are then concatenated in `luaV_concat()`.  The 5.1 version is this pretty dumb in the way it does this.  It adds up the length of the string pieces (totN), allocates a (totN) working buffer as a concatenation dest and then 
calls `lua_pushlstring()` which allocates another (`sizeof(TString)+totN`) buffer so you end up with 2 extra temporary copies of data not 3.  Note that the 5.3 version has fixed this, so any fix here isn't needed with 5.3.

If you want to do a fix, then can I suggest that you add an undocumented `luaL_pushTString()` which takes a `TString` buffer instead of a `char *` one and the application must use the Lua allocator to allocate the buffer and  the  treads the passed parameter as freeed by the call (e.g. it doesn't need to do GC).  It would be a small change to get  `luaV_concat()` to use this, and saving the unnecessary allocation.  Other core modules could also use this if the developers know how to.

I also realise that my suggestion is broadly the same as what you are proposing except one of exposure: the `lua_*` calls are public and documented is the Lua documentation and PiL.  There's a load of nice and complicated stuff (some with nasty side-effects or traps) in the other core lua headers that isn't exposed as a public interface.  The assumption is that any developers who use these calls know their way around the Lua source, and have worked out the constraints.  This extension definitely falls into this second category.  But as I say 5.3 removes what I see as the main usecase here anyway.

TerryE, 2018-11-12:
As a PS, one thing that you could take a look at is the EGC and the subset implemented in 5.3.  The current guiding parameters are a pain and don't work well.  At the moment any other collection strategy than ECG on every alloc can crater the system.  As result GC overhead is a big component of runtime.  If you are using LFS, then you will often have quite a lot of headroom, and so can afford to be a bit more lazy.  But the devil is in the detail.

tomsci, 2018-11-13:
I'm aware of the complexity of how strings are managed in Lua, in particular that strings have to be immutable because of how they are hashed, stored and potentially reused. Which is why these proposed changes don't go anywhere near `G[L}->strt` until `lua_pushbuf()` is called to convert the buffer into a fully-paid-up `TString`, after which point the caller must not use the `void*` buf pointer again. I may have muddied the waters by talking about using the `TString` struct prior to that, but until `lua_pushbuf()` is called, the extra space (which just happens to share a layout with `TString`) is just a convenient place to store a `len` field.

> So you either don't need to allocate space for string or you need to alloc sizeof(TSting) + N bytes. Either way its is best to leave the calling C code get rid of its temporary string.

lua_newbuffer() always allocates extra space for a TString, so if the caller ends up not calling lua_pushstring, or if the data already exists as a Lua string somewhere, then that extra space is unnecessary yes. But the focus of the API is for cases where in non-error cases, the data is destined to be turned into a string and it's a non-trivial amount of data that unlikely to already exist in the runtime, so you know that the `sizeof(TString)` overhead is worth paying up front.

> This approach takes us further away from standard Lua, and it opens up yet more bear-pits for unsuspecting developers. Ultimately it at best saves an extra malloc/free pair, and in most cases some reduction in temporary RAM use.

That's fair, but I'm hoping it's an API with fewer bear traps than say `luaM_free()` - hence why I didn't want it to explicitly expose `TString`s or require any pointer arithmetic on the caller's side.  In terms of deviations from the core, the only changes (as opposed to additions) are ~20 LOC in lstring.c and only to code that's already been modified for rom strings support, so it's not much of an increased maintenance burden. It could even be backed out simply by providing replacement fns that just used malloc and lua_pushlstring (unlike say the rotables changes which are inextricably required by the module build system) so it doesn't even commit us to having to support it in the future if we really didn't want to (or if a better Lua API came along instead).

The particular scenario I was hitting involved a 50KB blob of data I have to download and then transform (via an API that doesn't really lend itself to streaming or chunking) before writing to a file. A couple of extra 50 KB temporaries makes a significant difference in that scenario.

>  lua_* calls are public and documented is the Lua documentation and PiL

Well, except for `lua_pushrolstring`, `lua_pushlightfunction`, `lua_pushrotable`, etc. I'm not proposing anything that hasn't already been done? I have no objections to writing some documentation describing how these APIs are a deviation from standard Lua and explaining how to use them. If not for that precedent I'd be reluctant to call them `lua_*` and put them in `lua.h`, but there didn't seem to be any pre-existing pattern for segregation of eLua/NodeMCU additions vs what's actually core Lua.

I intend to float this past the Lua devs as well on `lua-l` since I don't think it's particularly nodemcu-specific. Although I'd expect it'd never make it in as-is and best case would end up as extensions to the userdata APIs because I doubt they'd want APIs that returned non-runtime-owned buffers. I didn't attempt that here as converting a userdata to a string would be complicated and I highly doubt I would get it right. And the ability to free immediately via `lua_freebuf` is a desirable property even if manually-managed memory doesn't really fit with the Lua GC philosophy.

> The assumption is that any developers who use these calls know their way around the Lua source, and have worked out the constraints

Hmm I'd hope nobody writing mere modules is using _any_ of the non-public Lua functions! Allocation functions notwithstanding of course. I suppose the `node` module is an exception too, although I was pleased to see that the addition of a strip arg to `lua_dump` means that on 5.3 `node.compile()` won't need to call the internal `luaU_dump` any more.

> current guiding parameters are a pain and don't work well. At the moment any other collection strategy than ECG on every alloc can crater the system

Can you elaborate? That is news to me and I admit not something I noticed in my own tinkerings with NodeMCU on 5.3. (I am getting off topic here, sorry).

TerryE, 2018-11-13:
Tom, I don't think that we're having as disagreement because our conclusions are largely the same.  We both acknowledge the utility of an internal means of creating a buffer with an inbuilt `TString` header and passing this to the Lua API would be a useful improvement.  The main difference between us is that I would prefer not to make it publicly available to the general C module developer.

I also realise that are (probably the only other potential contributor) familiar with the Lua kernel which is why I am taking the time out to give you detailed and considered responses. :smile: 

AS I have said elsewhere, I am less than satisfied with the coding style for the eLua and is non-conformance to Lua style.  Nonetheless, it's greatest merit is that it does the job it seeks to achieve.  However, I have been backing out this patch code or converting it to Lua style where practical.  I would really prefer not to extend its paradigm.

As to the roadmap but-list such as ECG improvements maybe I need to start documenting a proposed roadmap.

tomsci, 2018-11-13:
Thanks for clarifying - the first sentence of your response gave me somewhat the wrong impression :) Keeping functions such as these separate from the main API makes sense.

> 5.3 removes what I see as the main usecase here anyway.

Sounds like it improves matters, with the optimisations to concat, but I don't think it's fair to say it does away with them entirely. I would have liked a way to zero-copy convert a userdata / raw alloc into a Lua string in projects other than NodeMCU after all, they just haven't been quite so memory constrained as here. But maybe the usecase of really requiring a large single contiguous allocation is not so common as I thought.

> such as ECG improvements maybe I need to start documenting a proposed roadmap.

I think that would be very helpful. Then I won't be tempted to copy the existing style for lack of better examples :D

How would you suggest I move forward with this proposal, in terms of renaming/moving/rework?

closed 2019-11-22 by stale[bot]