nodemcu/nodemcu-firmware #2556: Building with Bloom module fails due to undefined `SHA256_xxx` functions

marcelstoer, 2018-11-15:
Recently one of the cloud builds failed with this error:

```
modules/.output/eagle/debug/lib/libmodules.a(bloom.o): In function `bloom_filter_reset':
/home/travis/build/marcelstoer/nodemcu-custom-build/nodemcu-firmware/app/modules/bloom.c:125: undefined reference to `SHA256_Init'
/home/travis/build/marcelstoer/nodemcu-custom-build/nodemcu-firmware/app/modules/bloom.c:125: undefined reference to `SHA256_Update'
/home/travis/build/marcelstoer/nodemcu-custom-build/nodemcu-firmware/app/modules/bloom.c:87: undefined reference to `SHA256_Final'
modules/.output/eagle/debug/lib/libmodules.a(bloom.o): In function `add_or_check':
/home/travis/build/marcelstoer/nodemcu-custom-build/nodemcu-firmware/app/modules/bloom.c:23: undefined reference to `SHA256_Init'
/home/travis/build/marcelstoer/nodemcu-custom-build/nodemcu-firmware/app/modules/bloom.c:26: undefined reference to `SHA256_Update'
/home/travis/build/marcelstoer/nodemcu-custom-build/nodemcu-firmware/app/modules/bloom.c:29: undefined reference to `SHA256_Final'
collect2: error: ld returned 1 exit status
```

I see that [`bloom.c` requires `SHA2_ENABLE` to be defined](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/bloom.c). That is fine; it's on by default and I'm not getting that 

> Must have SHA2_ENABLE set for BLOOM module

message. Yet, it appears the functions defined in `app/crypto/sha2.h` are not found.

What is going wrong here?

pjsg, 2018-11-15:
How many times has it happened? It seems very odd....

marcelstoer, 2018-11-15:
It is...I can't produce exact numbers but it happens every now and then. If I kick Travis for a rebuild it sometimes helps IIRC.

sonaux, 2018-11-15:
I tried now and got this error if `crypto` module is not selected.

marcelstoer, 2018-11-16:
That's what I initially suspected to be the root cause. However, I haven't found any code or Makefile logic yet to back this up. Have you?

pjsg, 2018-11-16:
I think that this is a problem in the way that we only build directories whose modules are being included.... I suspect the problem is in the line:

```
SEL_MKTARGETS        := $(shell $(CC) -E -dM include/user_modules.h | sed -n '/^\#define LUA_USE_MODULES_/{s/.\{24\}\(.*\)/\L\1/; p}')
```

marcelstoer, 2018-11-16:
Philip, what do you propose? Shall we just add a note to the Bloom docs saying that crypto be required (just like `SHA2_ENABLE`)?

pjsg, 2018-11-16:
Would compiling the crypto directory each time actually make the images any bigger? I.e. is the optional crypto directory just a build time performance optimization?

devsaurus, 2018-11-16:
> I.e. is the optional crypto directory just a build time performance optimization?

If I understood correctly, yes. #2301 introduced this optimization and I remember [polishing the u8g->u8g2 update](https://github.com/nodemcu/nodemcu-firmware/pull/2184#issuecomment-406867582) to align with it.

pjsg, 2018-11-16:
If so, then we can always build the crypto directory and it will just work (and cause no increase in code size).

marcelstoer, 2018-11-26:
@pjsg, @devsaurus would one of you have capacity to fix this before the next master drop (planned for next week)?

devsaurus, 2018-11-26:
If it was up to me I'd just re-move crypto from the `OPT_MKTARGETS` list back to the `SUBDIRS` and `COMPONENTS_eagle.app.v6` lists as @pjsg proposed.
Just as a workaround - not enough make-fu round here to teach `OPT_SEL_MKTARGETS` the `bloom` dependency for `crypto`.

sonaux, 2018-11-27:
> re-move crypto from the OPT_MKTARGETS list back to the SUBDIRS and COMPONENTS_eagle.app.v6 lists

Compared to including full `crypto` module it also saves some 4-6 kb from firmware size.

I can raise PR with this changes if it's appropriate: https://github.com/sonaux/nodemcu-firmware/commit/7a6a16950771d0b2f282330a50aadc98fa37caf8

pjsg, 2018-11-27:
That is exactly the change that I would have made. Can you raise a PR to do that please?

PR #2573 (2018-11-28): Always compile `app/crypto` folder (BLOOM modude depends on it)
Fixes #2556.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.
closed 2018-11-28 by marcelstoer