nodemcu/nodemcu-firmware #2564: ESP32: WS2812 breaks then using on system with psram

x-logLT, 2018-11-20:
### Issue
ws2812 module uses interrupts that are placed in IRAM, this requires that all the data that ISR routine accesses be in DRAM. Because there is no functionality in LUA to specify from which memory region  buffers should be allocated, this requirement cant be satisfied on system with PSRAM.

### Possible solutions
As I see there are several ways to solve this:
1. Copy data to correct memory block on each write - memory and cpu overhead.
2. Remove ESP_INTR_FLAG_IRAM flag from ISR hookup - might slow down things, add random delays.
3. Somehow allocate memory in correct memory block and hook that memory to LUA or extend LUA api so that the memory block can be specified.

I have no idea if No. 3 is at all viable, would be interesting to hear from folks who are more familiar with LUA implementation.

So far have implemented No. 2 , it can be found at  https://github.com/x-logLT/nodemcu-firmware/commit/390e8a0082331064ebf1abb0b6ce96f4b02d74c9 
Need feedback / better ideas for this.

devsaurus, 2018-11-20:
Seems your analysis is correct, thanks for highlighting this!
The PSRAM feature is probably not yet explored for all details in the context of NodeMCU firmware

1. Overhead could be reduced if memory is only copied in case of `esp_ptr_internal(data)` returns `false`. Pro: Can be handled in `platform/ws2812.c`.
2. A quick win. Don't know whether the potential delays are significant at all.
3. I'd stay away from the Lua API, and turn `ws2812_buffer.values` into a pointer. You'd get just the descriptor memory from `lua_newuserdata()` and have another call to `heap_caps_malloc()` with respective settings for internal memory. Con: Platform memory characteristics are exposed in the module code.

My tendency is towards 3. (and have the allocation encapsulated in a macro).

x-logLT, 2018-11-21:
Wouldn't that cause a memory leak if Lua GC freed the descriptor?

pjsg, 2018-11-21:
The PSRAM restrictions that I found do not mention not being able to use it
from interrupt level. The *reason* that interrupt handlers on the ESP8266
needed to be in RAM was that the cpu could get confused when it was in the
middle of performing an SPI operation on the flash chip when an interrupt
happened (as it would then start a new SPI transaction to fetch the code).
Maybe this has been resolved in the ESP 32.

devsaurus, 2018-11-21:
> Wouldn't that cause a memory leak if Lua GC freed the descriptor?

Yes, that would cause a leak if the descriptor object (based on `ws2812.buffer` metatable) is left untouched. To avoid the leak, a `__gc` method would need to be added to `ws2812_buffer_map` which frees the allocated chunk once the descriptor is collected.

devsaurus, 2018-11-21:
@pjsg I understand https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/intr_alloc.html#iram-safe-interrupt-handlers and https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spi_flash.html#iram-safe-interrupt-handlers in the way that it's forbidden to access anything else than internal RAM from within an ISR with ESP_INTR_FLAG_IRAM (code and data accesses). Reading data from PSRAM violates this requirement IMO.

It's even mentioned explicitly for [`rmt_driver_install()`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/rmt.html?highlight=PSRAM#_CPPv218rmt_driver_install13rmt_channel_t6size_ti).

pjsg, 2018-11-21:
I stand corrected.

devsaurus, 2019-05-31:
Pushed e30904b60760f8c290ada6deb2a44480ab884ba2 to fix the issue as proposed.

commit e30904b6 (2019-05-31): force common intr alloc flags for all platform components that use rmt
remove ESP_INTR_FLAG_IRAM to fix #2564
closed 2019-05-31 by devsaurus