nodemcu/nodemcu-firmware #2570: ESP32: WS2812FX module (work in progress)

x-logLT, 2018-11-25:
So far this module is in a very early stage of development and can be found at https://github.com/x-logLT/nodemcu-firmware/tree/esp32-working
Needs testing  & feedback on overall api/ implementation.

### Justification
This module, opposite from from ws2812-effects module (in master branch), utilizes unmodified WS2812FX/Neopixel libraries. Thus it should be a lot easier to maintain and keep up-to-date to the lib's functionality in the long run. 

### Current progress
* Neopixel lib is only used for low level pixel data manipulation, data stream is handled by RMT peripheral.
* Pretty effects are working
* Supports only one instance, not really "hooked" into LUA
* Service routine must be called from the LUA
* Only rudimentary control functions are exposed to LUA
* Pixel color order is set by passing a string

### Testing

At the moment, its best to call service routine from a timer:
```
ws2812fx.init(pin, pixelCount, "GRB")
wstmr = tmr.create()
wstmr:register(50, tmr.ALARM_AUTO, function (t) ws2812fx.service() end)
wstmr:start()
ws2812fx.start()
```
All available effects can be printed by executing:
```
i = 0
while(i < ws2812fx.getmodecount()) do
    print(i..": "..ws2812fx.getmodename(i))
    i = i+1
end
```

### Whats next

Going forward I'm thinking about making whole wrapper layer/ RMT interface thread-safe and offloading service routing to its of thread thus allowing LUA thread to do other things.

devsaurus, 2019-05-31:
@x-logLT any updates on this?

x-logLT, 2019-07-01:
Not yet, seeing as there was no feedback and for my purposes it was "good enough", it sorta went to the back of the "things to do" queue, though i'm planning on getting back to it by the end of a year.

nwf, 2020-06-26:
As per #3122, I think we should not be advancing ws2812 effects in C, and so I am content to close this out.

closed 2020-06-26 by nwf