nodemcu/nodemcu-firmware #2574: add a diy Endpoint in enduser_setup

hanfengcan, 2018-11-27:
### Missing feature
add a diy callback(Endpoint)
 
### Justification
I design a clock by nodemcu, I hope that I can set time pass by browser of phone.

if the enduser_setup have a diy callback that I can set the time by my html
### Workarounds

I implement Captive Portal by lua, but I work not well.

I found that it has dns redirection when I read code of enduser_setup.

this is the reason that I create the issuse

HHHartmann, 2018-11-27:
Maybe #2493 will help you

marcelstoer, 2018-11-27:
I was gonna post the same reference but I really don't understand what the OP's requirement is.

hanfengcan, 2018-11-28:
> Maybe #2493 will help you

maybe, how can I try

hanfengcan, 2018-11-28:
> I was gonna post the same reference but I really don't understand what the OP's requirement is.

such as POST /diy?datatime=timestamp  and rtctime.set(timestamp) to update the time

Gui13, 2018-11-28:
Just setup a web server once your device is loaded, and implement it like this (source: [this blog post](https://randomnerdtutorials.com/esp8266-web-server/)):

```lua
srv=net.createServer(net.TCP)
srv:listen(80,function(conn)
    conn:on("receive", function(client,request)
        local buf = "";
        local _, _, method, path, vars = string.find(request, "([A-Z]+) (.+)?(.+) HTTP");
        if(method == nil)then
            _, _, method, path = string.find(request, "([A-Z]+) (.+) HTTP");
        end
        local _GET = {}
        if (vars ~= nil)then
            for k, v in string.gmatch(vars, "(%w+)=(%w+)&*") do
                _GET[k] = v
            end
        end
        buf = buf.."<h1> ESP8266 Web Server</h1>";
        buf = buf.."<p>Time is " .. rtctime.get() .. "</p>";
        if(_GET.datatime)then
              rtctime.set(_GET.datatime);
        end
        client:send(buf);
        client:close();
        collectgarbage();
    end)
end)
```

hanfengcan, 2018-11-29:
@Gui13 
can I use `POST  /update?datetime=timestamp ` and read eus_params.lua through enduser setup?

hanfengcan, 2018-12-06:
I have the answer and close this issuse

closed 2018-12-06 by hanfengcan