nodemcu/nodemcu-firmware #2575: wifi clients disconnected during script execution for more than 2 sec

nedoskiv, 2018-11-28:
Hello, i am using NodeMCU 2.2.0.0  powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
i face a problem with wireless client disconnection (esp is in AP mode)  if s lua script is executed for 2 or more seconds.
Is there a way to avoid that, something like wifi.beacon.sent function or something that can be called from inside executed script to keep wifi working during execution and avoid client disconnection?

nwf, 2018-11-28:
In general, the scripts should never run for any substantial length of time; nodemcu is an event-driven programming system.  The Espressif SDK on which we build is non-preemptive, so Lua must regularly yield the core.  See https://github.com/nodemcu/nodemcu-firmware/blob/master/docs/en/lua-developer-faq.md for details.

marcelstoer, 2018-11-28:
One for #1010

closed 2018-11-28 by marcelstoer