nodemcu/nodemcu-firmware #2576: MQTT Timeout needs to be increased

dtran123, 2018-11-29:
### Expected behavior

Expect my mqtt client to successfully connect to mqtt broker without timeout when using
client certificates + CA certifcates.

My code is running in LFS to save memory. The issue was resolved when I
increased the mqtt connect timeout from 5 to 10 seconds

mqtt.c
#define MQTT_CONNECT_TIMEOUT  10

5 secs seems to strict for some scenarios.

### Actual behavior

I get the following traceback with event timeout. The mqtt connection fails with mosquitto broker.

require("send_to_my_mqtt"):connect_mqtt()
Handle Input: first=1, pos=41, len=256, actual=41, line=require("send_to_my_mqtt"):connect_mqtt()
40288
enter mqtt_socket_client.
NodeMCU_f8a186
length username: 5
length password: 5
MQTT: Init info: basicPubSub, admin, admin
leave mqtt_socket_client.
enter mqtt_socket_connect.
TCP ip is set: 108.61.159.165
TCP port is set: 8883.
enter socket_connect.
leave socket_connect, heap = 37472.
leave mqtt_socket_connect.
37472
> client handshake start.
espconn_mbedtls.c 664, type[certificate],length[846]
espconn_mbedtls.c 664, type[certificate],length[846]
espconn_mbedtls.c 664, type[private_key],length[1192]
espconn_mbedtls.c 664, type[certificate],length[939]
E:M 528
client handshake failed!
Reason:[-0x4310]
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 5.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 4.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 3.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 2.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 1.
event timeout. 
Can not connect to broker.
Connect failed
Reason=
-5
keep_alive_tick: 0
leave mqtt_socket_timer.
enter mqtt_socket_reconnected.
Connect failed
Reason=
-5
enter mqtt_socket_disconnected.
leave mqtt_socket_disconnected.
leave mqtt_socket_reconnected.

Mosquitto logs are:

ov 29 08:17:23 iotcloud mosquitto[5975]: 1543479443: New connection from 104.163.170.160 on port 8883.
Nov 29 08:17:30 iotcloud mosquitto[5975]: 1543479450: OpenSSL Error: error:140940E5:SSL routines:SSL3_READ_BYTES:ssl handshake failure
Nov 29 08:17:30 iotcloud mosquitto[5975]: 1543479450: Socket error on client <unknown>, disconnecting.

### Test code

The following code is run in LFS.

local MYMQTT = {}

function MYMQTT.connect_mqtt()

print(node.heap())
local mqttclientid="myclient"
local mqttkeepalive=1500
mqttserver='x.x.x.x'
mqttserverport=8883
mqttsecure=1
mqtttopic='test'
mqttsubscribeqos=1
themessage='{"state":{"desired":{"color":"YELLOW"},"reported":{"color":"GREEN"}}}'
mqttpublishqos=1
mqttpublishretain=0

tls.cert.auth(true)
tls.cert.verify(true)

  m = mqtt.Client('basicPubSub', 1500, "admin", "admin", 1)

  local connectresult = m:connect('x.y.z.z', 8883, 1, 0, function(conn)
  print("MQTT connect success")

  local subresult = m:subscribe(mqtttopic, mqttsubscribeqos, function(conn)

  local pubresult = m:publish(mqtttopic, themessage, mqttpublishqos, mqttpublishretain, function(conn)
  
	 m:close() conn = nil m = nil
   print(node.heap())
   print("Publish success")

   end)  -- function
      if pubresult == false then
        m:close() conn = nil m = nil
        print("Failed publish")
      end
  end)  -- function
    if subresult == false then
      m:close() conn = nil m = nil
      print("Failed subscribe")
    end
  end,
    function(conn,reason)
       print("Connect failed")
       print("Reason=")
       print(reason)
    end
  ) -- function
  
  print(node.heap()) 
 end
 
 return MYMQTT

### NodeMCU version

NodeMCU 2.2.0.0 built with Docker provided by frightanic.com
	branch: master
	commit: 4095c408e6a8cc9718cb06007b408d0aad15d9cd
	SSL: true
	Build type: integer
	LFS: Size: 0x10000
	modules: file,gpio,mqtt,net,node,tls,tmr,uart,wifi
 build created on 2018-11-29 05:02
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)

### Hardware

ESP-12

dtran123, 2018-11-29:
When MQTT_CONNECT_TIMEOUT =10, we have a successful publish as seen below:

> require("send_to_my_mqtt"):connect_mqtt()
Handle Input: first=1, pos=41, len=256, actual=41, line=require("send_to_my_mqtt"):connect_mqtt()
40736
enter mqtt_socket_client.
NodeMCU_f8a186
length username: 5
length password: 5
MQTT: Init info: basicPubSub, xxx, yyy
leave mqtt_socket_client.
enter mqtt_socket_connect.
TCP ip is set: x.61.159.y
TCP port is set: 8883.
enter socket_connect.
leave socket_connect, heap = 37520.
leave mqtt_socket_connect.
37520
> client handshake start.
espconn_mbedtls.c 664, type[certificate],length[846]
espconn_mbedtls.c 664, type[certificate],length[846]
espconn_mbedtls.c 664, type[private_key],length[1192]
espconn_mbedtls.c 664, type[certificate],length[939]
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 10.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 9.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 8.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 7.
enter mqtt_socket_timer.
timer, queue size: 0
event_timeout: 6.
client handshake ok!
enter mqtt_socket_connected.
Send MQTT connection infomation, data len: 39, d[0]=16 
leave mqtt_socket_connected.
enter mqtt_socket_sent.
enter mqtt_socket_received.
MQTT: Connected
MQTT connect success
enter mqtt_socket_subscribe.
subscribe string
topic: test - id: 1 - qos: 1, length: 11
msg_size: 1, event_timeout: 0
Sent: 11
send_if_poss, queue size: 1
subscribe, queue size: 1
leave mqtt_socket_subscribe.
send_if_poss, queue size: 1
leave mqtt_socket_received.
enter mqtt_socket_sent.
sent1, queue size: 1
sent2, queue size: 1
leave mqtt_socket_sent.
enter mqtt_socket_received.
MQTT_DATA: type: 9, qos: 0, msg_id: 1, pending_id: 1
MQTT: Subscribe successful
enter mqtt_socket_publish.
Sent: 79
send_if_poss, queue size: 1
publish, queue size: 1
leave mqtt_socket_publish.
send_if_poss, queue size: 1
leave mqtt_socket_received.
enter mqtt_socket_sent.
sent1, queue size: 1
sent2, queue size: 1
leave mqtt_socket_sent.
enter mqtt_socket_received.
MQTT_DATA: type: 4, qos: 0, msg_id: 2, pending_id: 2
MQTT: Publish with QoS = 1 successful
enter mqtt_socket_close.
Send MQTT disconnect infomation, data len: 2, d[0]=224 
leave mqtt_socket_close.
21448
Publish success
send_if_poss, queue size: 0
leave mqtt_socket_received.
client's data invalid protocol
Reason:[-0x45]
enter mqtt_socket_reconnected.
enter mqtt_socket_disconnected.
leave mqtt_socket_disconnected.
leave mqtt_socket_reconnected.

nwf, 2018-11-29:
I'm down.  We might want to raise it even higher; I think Linux, for example, defaults to 20 seconds before timing out a connect(), and I don't see why we wouldn't wait alt least that long.

If you're poking about in the mqtt code, be aware that #2571 is also ongoing, but it'd be great to not have the mqtt ticker wake up every second even when there's nothing to do.  We should be able to get a "tickless" approach pretty easily, just having the timer poised for the next event in queue?

dtran123, 2018-11-29:
I am not am expert in MQTT protocol so I am not sure what side effects there are for increasing the timeout to 20 secs.  I agree that ticker wake up every second is not efficient. I personally don't code in C (am a casual coder) so someone could help out with this (improve ticker mgmt & increase MQTT timeout).

closed 2019-02-19 by marcelstoer