nodemcu/nodemcu-firmware #2578: tls.cert.auth should be re-exposed in the documentation

dtran123, 2018-11-30:
### Expected behavior

Documentation to include tls.cert.auth as an available function.
As indicated by issue #2576, I was able to apply both auth & ca certs to tls
in my connection to mqtt broker (which requires both) using:

tls.cert.auth(true)
tls.cert.verify(true)

### Actual behavior

Documentation excludes mention of tls.cert.auth

### Test code

The following code will work:

tls.cert.auth(true)
tls.cert.verify(true)

### NodeMCU version

Both dev and master branches don't mention  tls.cert.auth.

### Hardware

N/A

marcelstoer, 2018-12-09:
Could you support NodeMCU by providing a PR with the necessary additions to the documentation?

dtran123, 2018-12-11:
More characterization needs to be done before we expose tls.cert.auth because there are many scenarios beyond mqtt (my usecase) such as https, etc. As such, more testing needs to be done first to determine the usecases where this function is reliably applicable.

HHHartmann, 2020-01-14:
We should either document this properly or remove the functionality.
@pjsg  could you as the original author (#1125) shed some light on this so it can be documented?
If there is a consensus I can also remove this functionality.

dtran123, 2020-01-15:
Please don't remove the functionality as it works fine. It just needs proper documentation. This feature is absolutely needed for a complete TLS secured connection.

dtran123, 2020-01-15:
To use this feature, you configure the cert and RSA key:

tls.cert.auth(
    [[
    -----BEGIN CERTIFICATE-----
    CLIENT CERTIFICATE String (PEM file)
    -----END CERTIFICATE-----
    ]]
    ,
    [[
    -----BEGIN RSA PRIVATE KEY-----
    CLIENT PRIVATE KEY String (PEM file)
    -----END RSA PRIVATE KEY-----
    ]]
    )

Then, enable verification:
tls.cert.auth(true)

PR #3010 (2020-02-01): document tls.cert.auth
Fixes #2578.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

\<Description of and rationale behind this PR\>
Trying to add documentation for tls.cert.auth
Please review and tell me if it is nonsens as I don't really know much about tls.
closed 2020-02-01 by marcelstoer