nodemcu/nodemcu-firmware #2584: HTTPS TLS requests don't work

dusblinov, 2018-12-06:
### Expected behavior
Connection via https works

### Actual behavior
I read all the issues tickets for https via the `http` module and the `tls` module. But nothing helped me. `http` module gave me this errors `HTTP client: Disconnected with error: -127` and `HTTP client: Connection timeout`. `tls` module just do nothing when I tried to run it. (I tried with my server, but also with public ones, such as `nodemcu-build.com/jumbotron-narrow.css`) 

As far as I see in this thread (https://github.com/nodemcu/nodemcu-firmware/issues/1707
), there is also no working solution.

Finally, are there any solutions that work? thank you

### Test code
```Lua
function test(host, path)
  local url = "https://" .. host .. path;
  local srv = tls.createConnection(net.TCP, 0)
  srv:on("receive", function(sck, c) print("net/TLS to " .. url .. " succeeded") end)
  srv:on("connection", function(sck, c)
    sck:send("GET " .. path .. " HTTP/1.1\r\nHost: " .. host .. "\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n")
  end)
  srv:connect(443, host)
end
```
### NodeMCU version
```
NodeMCU custom build by frightanic.com
        branch: master
        commit: f99f295d97ac3a4e147f601802d5869d9d2d3012
        SSL: true
        modules: crypto,file,gpio,http,mqtt,net,node,rtcmem,rtctime,tmr,uart,wifi,tls
 build created on 2018-08-28 20:17
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)

```
### Hardware
LoLin ESP8226 4MB

marcelstoer, 2019-01-05:
Maybe #2587 will fix this.

galjonsfigur, 2019-01-06:
I was testing HTTPS requests and they work just fine with `http` module - it supports HTTPS requests and even server certificate validation if `tls.cert.verify()` is used before request (that requires ~40kb free heap, but that is possible now with LFS). My test code:

```lua
local function test(host, path)
  local url = "https://" .. host .. path;
  http.get(url, nil, function(code, data)
    if (code < 0) then
      print("HTTPS request failed")
    else
      print("Success")
    end
  end)
end
```

But the content of most sites is just too long and that's why your code example fails - you will have to change `SSL_BUFFER_SIZE`. With `http` module if response is too long callback will still run and with example above it will just print `HTTP request failed`, unlike code using`tls` module. If you just want to make HTTPS requests then you should be fine. #2587 should make things better but if you can, build your version of firmware with  turned on `MBEDTLS_DEBUG_C` (in `app/include/user_mbedtls.h` in line `193` change `#undef` to `#define`) and use `tls.setDebug(2)` for example - it should make things easier to debug.

closed 2019-02-04 by marcelstoer