nodemcu/nodemcu-firmware #2591: https on ESP32 not working?

joysfera, 2018-12-13:
Sorry to bother, I just wanted to ask kindly if https connection in http module of dev-esp32 is supposed to work or not (yet).

### Expected behavior
https GET request works OK

### Actual behavior
E (65046292) esp-tls: mbedtls_ssl_handshake returned -0x4c
E (65046292) esp-tls: Failed to open new connection
E (65046292) TRANS_SSL: Failed to open a new connection
E (65046292) HTTP_CLIENT: Connection failed, sock < 0
W (65046302) http: esp_http_client_perform returned error 28674

### Test code
```Lua
http.get("https://dilbert.com",function(c, d) if c < 0 then print("fail") else print(c,d) end end)
```
### NodeMCU version
current dev-esp32 (i.e. including #2529, I believe)

### Hardware
ESP32_PICO_Core_Board_V2

devsaurus, 2018-12-16:
I've seen it working against my local server at least.

@tomsci are there specific tweaks that need to be set to boost https? Particularly for the mbedtls settings in menuconfig?

joysfera, 2018-12-16:
I have tested many different servers, the dilbert.com is just one example. None of the servers worked (I mean connection to any of the tested servers worked). Including google.com, seznam.cz, and my own servers that use Let's Encrypt's certificates.

As for menuconfig, I (intentionally) didn't touch there anything (well, apart from adding the NodeMCU modules I needed, of course).

devsaurus, 2018-12-16:
Hm, it still does work for me.

@joysfera's test code
```lua
http.get("https://dilbert.com",function(c, d) if c < 0 then print("fail") else print(c,d) end end)
```
yields
```
(38511) http: Error returned from callback for HTTP event -1: string length overflow
```
which suggests that handshake succeeded and just the payload is too big.

And the example in the doc is working as well:
```lua
headers = {
  Connection = "close",
  ["If-Modified-Since"] = "Sat, 27 Oct 2018 00:00:00 GMT",
}
connection = http.createConnection("http://httpbin.org/", http.GET, { headers=headers } )
connection:on("complete", function(status)
  print("Request completed with status code =", status)
end)
connection:request()
```
```
Request completed with status code =	200
```

tomsci, 2018-12-16:
Have retested with a ROM built cleanly from f338d31f054080ff18445faddf5846297b5b56cc and a default sdkconfig (modulo the couple of unrelated changes specific to my project) and everything is working ok:

```lua
> http.get("https://www.xkcd.com/",function(c, d) if c < 0 then print("fail") else print(c,#d) end end)
> 200   5887
```

One thing worth noting is some test sites like dilbert.com run out of memory for me, so it's possible that's the problem you're hitting too just with slightly different errors being manifest (OOM can cause all kinds of misleading errors sometimes)

tomsci, 2018-12-16:
Thanks for testing @devsaurus. I agree it does seem likely to be a free memory issue. I'm glad the updated ESP-IDF wasn't to blame :)

It's worth noting that `http.get` requires more temporary memory than using `createConnection` directly, because the `http` module has to buffer the response, so if it's possible to refactor your code to use `createConnection` and multiple callbacks, you may have more luck. Unfortunately HTTPS requests are always more memory-hungry than simple HTTP, and the ESP32 still doesn't have all that much RAM. I have some memory-saving improvements to `http.get` sitting on a branch but there is some debate how best to integrate them.

Also make sure (if you're not already) to set `node.egc.setmode(node.egc.ON_ALLOC_FAILURE)` on every boot otherwise no `http.get` call with a response > 4096 bytes is likely to work.

@joysfera can you retry with the following example? `m.xkcd.com` is my go-to for a really simple (but not completely trivial) test site.

```lua
http.get("https://m.xkcd.com/",function(c, d) if c < 0 then print("fail") else print(c,#d) end end)
```

devsaurus, 2018-12-16:
> I'm glad the updated ESP-IDF wasn't to blame :)

Yes, gladly. I took the opportunity to move esp-idf to the current head of the release/v3.2 branch.
All the testcases mentioned here work well on my build with esp-idf on [1c4d92f](https://github.com/espressif/esp-idf/commits/release/v3.2?before=1c4d9f2acacc666a4eb02f1450abd400e7df58c3+35).

devsaurus, 2018-12-16:
@joysfera if you're still not successful then there might be a subtle precondition lurking somewhere else.

AFAIK my local changes to the default sdk settings are just an increased main task stack size (24k instead of 8k) and enabling PSRAM (due to unrelated requirements of my projects).

joysfera, 2018-12-16:
I'll test it on Monday morning. BTW, if the http.get will fail with most pages due to too large response maybe it could get some kind of a simple regex filter as another input parameter that would filter out most of the unneeded junk in the response stream before storing it to memory and passing it back to Lua, so the memory would not overflow.

joysfera, 2018-12-17:
I am reporting gladly that HTTPS is working OK for me today. I have re-flashed the PICO ESP with Friday's build (because I tested Mongoose OS in the meantime) and run the same tests but it worked today, even without the suggested egc settings change:
```Lua
Mounting flash filesystem...
Formatting file system. Please wait...
...
NodeMCU ESP32 build unspecified powered by Lua 5.1.4
lua: cannot open init.lua
> wifi.mode(wifi.STATION)
> wifi.start()
> wifi.sta.config({ssid="dlink",pwd="dlink"})
> http.get("http://m.xkcd.com/",function(c, d) if c < 0 then print("fail") else print(c,#d) end end)
> I (136615) HTTP_CLIENT: Redirect to https://m.xkcd.com/
200	2459
> http.get("https://dilbert.com/",function(c, d) if c < 0 then print("fail") else print(c,#d) end end)
> W (192725) http: Error returned from callback for HTTP event -1: string length overflow
```
So all is great, your mind power got it fixed on my end. Thanks! I apologize for the github issue, it indeed didn't work on Friday for me and I was trying to interleave http with https requests to ensure it was not a network connectivity or similar issue. No idea what is was, glad it's working now.
Looks like the last missing bit in dev-esp32 for me is the rtcmem module (that I use as a memory that survives reboot), otherwise it seems to be ready for esp8266 replacement? Great! And thank you all for that.

closed 2018-12-20 by marcelstoer