nodemcu/nodemcu-firmware #2593: Keep rebooting on ESP32 and formatting the flash

Kashouryo, 2018-12-15:
### Expected behavior
After flashing the firmware into ESP32, I can access it from ESPlorer and operate it. 

### Actual behavior
ESP32 keep reboot and prints
`Mounting flash filesystem...
Formatting file system. Please wait...`

### NodeMCU version
The newest git pull from dev-esp32 at Dec, 15 2018, 18:35 GMT+8

### Hardware
ESP-WROOM-32

devsaurus, 2018-12-16:
Please post the full log of a boot starting with power on or a reset.

Kashouryo, 2018-12-16:
Now it works... When I tried to flash the bin back to my ESP32 and have the error log display.
But here're some detail: 
1. Before it, I installed MicroPython.
2. I forgot to erase the flash and started to flash the NodeMCU from 0x10000 offset.
3. Reasonable explanation: Some data in MicroPython from 0x00000 to 0x10000 helped nodemcu.

flashing command:
`esptool.py --port /dev/ttyUSB1 write_flash 0x10000 build/NodeMCU.bin`

closed 2018-12-16 by Kashouryo