nodemcu/nodemcu-firmware #2601: Resolve deprecation status of static timers

marcelstoer, 2018-12-30:
Ever since @djphoenix added an OO API to the timer module with f1afe7b6ab2ab2b385f28fda80a16a7c97ac8994 over two years ago our documentation has been saying that 

> Static timers are deprecated and will be removed later.

That's a bit problematic for two reasons:

1. The note is just in the documentation and not reflected in the code. One should use `platform_print_deprecation_note` in `tmr.c` to print a deprecation note to the console (like we do elsewhere) whenever a static timer function is used.
1. I couldn't find a follow-up issue here that would ask for the actual removal of the static timers. Who would/could/should have done that? Things get forgotten so easily.

It's all history now but I'd like to resolve that deprecation note in the docs. IMO we should either removed it or follow-up with code changes.

Tagging @djphoenix, @dnc40085 who is listed as the module maintainer in the docs and @devsaurus who ported the module to the ESP32 and thereby removed the static timers if I'm not mistaken

jmd13391, 2018-12-30:
IMO, Removing the static timers is a bad idea. Legacy compatibility aside, using static timers consumes less memory than using the OOP timers.

My vote is to remove the deprecation statement from the docs.

marcelstoer, 2018-12-30:
> Legacy compatibility aside

That's why we have been using `platform_print_deprecation_note`. We keep that in the code for a couple of releases (i.e. maybe half a year) until functionality is removed. That gives developers ample time to adjust their code where necessary.

> using static timers consumes less memory than using the OOP timers.

How much?

jmd13391, 2018-12-30:
"How much?"

That's a question best answered by @TerryE ;-)

TerryE, 2018-12-30:
Marcel, all I take by Joe's comment is that of the committers, I possably have the best understanding of the Lua VM internals and VM and so can answer the Q: what are the memory +/- of using the Object form of timers vs the static form?  That is providing a data point which aid any decision that we make. 😊

jmd13391, 2018-12-31:
"Terry didn't make any such claim"

Actually, he did... I can't recall at the moment the specific context of his statement but I do recall it was to justify his use of static timers in a snip of Lua code he wrote (if pressed, I will set off on a quest to find and re-post the statement). Also, I suggested Terry supply the answer as IMO, he is a very knowledgeable resource on the subject.

marcelstoer, 2019-01-08:
> I think, given the very low additional footprint of dynamic timers and their superior programming model ... static timers should be completely retired

👍 Any volunteers?

PR #2603 (2019-01-22): Expunge integer timers
Fixes #2601.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.
closed 2019-01-22 by marcelstoer