nodemcu/nodemcu-firmware #2602: Change u8g2 to work correctly with all screens on ESP32

StasBronchukov, 2019-01-07:
### Expected behavior
Change SPI frequency for u8g2 to work correctly on ESP32

### Actual behavior
Incorrect work when using the u8g2 library on lcd st7920 128x64 (The screen is defective, but working)
[image]
### Test code

```Lua
--Initialize lcd and output a filled rectangle ESP32
sclk = 19
mosi = 23
cs   = 22
dc   = 16
res  = 17
bus = spi.master(spi.HSPI, {sclk=sclk, mosi=mosi})
disp = u8g2.st7920_s_128x64(bus, cs, dc, res)
disp:clearBuffer()
disp:drawBox(5,5,30,30)
disp:sendBuffer()

```
### Hardware
ESP32

### Justification
This is because the SPI speed is higher than the speed at which the ST7920 can work normally. So for example when using ESP8266 with a standard example:

```Lua
--Initialize lcd and output a filled rectangle ESP8266
-- Hardware SPI CLK  = GPIO14
-- Hardware SPI MOSI = GPIO13
-- Hardware SPI MISO = GPIO12 (not used)
-- Hardware SPI /CS  = GPIO15 (not used)
cs  = 8 -- GPIO15, pull-down 10k to GND
dc  = 4 -- GPIO2
res = 0 -- GPIO16
bus = 1
spi.setup(bus, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 80)
-- we won't be using the HSPI /CS line, so disable it again
gpio.mode(8, gpio.INPUT, gpio.PULLUP)
disp = u8g2.st7920_s_128x64(bus, cs, dc, res)
disp:clearBuffer()
disp:drawBox(5,5,30,30)
disp:sendBuffer()
```
The same thing happens. However, with ESP8266, this problem is easily solved by changing the clock_div parameter:
 ```Lua
spi.setup(bus, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 80)
```
[image]
In ESP32, you can tune the SPI frequency with:
```Lua
device_config = {mode = 0, freq = 1000000}
device_config.cs = 22
dev1 = busmaster:device(device_config)
```
But
```Lua
u8g2.ssd1306_128x64_noname(bus, cs, dc, res)
```
Bus takes as a parameter the SPI bus master object, not the SPI device object, and we get the error:

```Lua
1.lua:10: bad argument #1 to 'st7920_s_128x64' (spi.master expected, got userdata)
```

### Workarounds
I still have little understanding, but as I understand it, the u8g2.st7920_s_128x64 function (bus, cs, dc, res) creates an SPI device object inside of itself and initializes it with standard values. If this is the case, as it seems to me, you need to send an SPI device object to u8g2.st7920_s_128x64 (bus, cs, dc, res) that we would create earlier, and not an SPI bus master object.

Note: Sorry if there are errors in the text, I write through the translator

devsaurus, 2019-01-07:
In general, the situation is as you described: The display constructors like `u8g2.st7920_s_128x64()` generate an SPI device object internally. This is done in [`u8x8_nodemcu_hal.c` line 271](https://github.com/nodemcu/nodemcu-firmware/blob/eb8d8b588ac518d6e611aece05366ca5800abc5b/components/platform/u8x8_nodemcu_hal.c#L271). u8g2 provides information about the SPI clock within the display info, and the HAL code just uses this value to configure the device object.
For your particular display this comes from [`u8x8_d_st7920.c` line 188](https://github.com/olikraus/U8g2_Arduino/blob/d4da8254220adf39db44faa52a0842967095d230/src/clib/u8x8_d_st7920.c#L188). I expect that the ESP32 code actually sets SPI frequency to 1&nbsp;MHz like you do for ESP8266 with clock divider 80.

I don't have a quick answer for what's going wrong with your display on the ESP32. You could try to modify either location, decrease the SPI frequency and see when the display starts to work. But the current situation is not quite consistent:
- u8g2 library expects the display to work at 1&nbsp;MHz
- You observe that it works at 1&nbsp;MHz on ESP8266
- The ESP32 code *should* set the frequency to 1&nbsp;MHz
- But the display doesn't work on ESP32

StasBronchukov, 2019-01-07:
> `u8x8_d_st7920.c` line 188

/* sck_clock_hz = */ 1000000UL  
Removed one zero, compiled and everything worked.Thank!
Maybe I got a defective lcd, but I think it would be good to allow the user to set the frequency as an optional parameter.
[image]

devsaurus, 2019-05-31:
Closing for now. u8g2 would need to provide respective functions for setting the clock frequency.

closed 2019-05-31 by devsaurus