nodemcu/nodemcu-firmware #2604: bitwise `bit` module producing incorrect results for large integers

jpeletier, 2019-01-09:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
`bit.bnot(0x80000000) == -2147483648` (`0x7FFFFFFF`)
### Actual behavior
`bit.bnot(0x80000000) == 0x80000000`
### Test code
Run this on a ESP8266 `master` branch
```lua
print(string.format("%x", bit.bnot(0x80000000)))
```
You'll get the correct result: `7fffffff`.
Now run the same in an ESP32 using the `dev-esp32` branch.
You'll get this: `80000000`, which is incorrect.

### NodeMCU version
`dev-esp32`

### Hardware
ESP32, ESP-WROOM-32
8<------------------------ END BUG REPORT -------------------------------------

TerryE, 2019-01-09:
@jpeletier Javier, are you doing an FP or integer build?

jpeletier, 2019-01-13:
ok, I got to the bottom of this.

**This issue affects both ESP8266 and ESP32**. This only happens in floating point builds. The reason this happens is that casting a double to an integer when the double would overflow is "undefined behavior" according to C99.

In particular the following line in `luaconf.h` is the culprit:
```c
#define lua_number2integer(i, d) ((i) = (lua_Integer)(d))
```
In integer builds, `d` is an integer (in fact, it is the same type), so this is really not even a conversion. The compiler will just emit code to copy the values.

In floating point builds, however, `d` is a double, and the behavior above is what we get. I have edited the bug description with the new findings.

jpeletier, 2019-01-13:
@TerryE, @devsaurus 
#2609 for ESP8266.
#2605 for ESP32.
actual fix is the same for both.

PR #2609 (2019-01-15): fix number2integer conversion in ESP8266 for floating-point builds
Fixes #2604 for ESP8266 .

ESP32 PR: #2605 

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `doc [... cut]
PR #2605 (2019-01-15): fix number2integer conversion in ESP32
Fixes #2604 for ESP32.

ESP8266 PR: #2609.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This PR fixes a qui [... cut]