nodemcu/nodemcu-firmware #2611: ESP32: Long Range Wi-Fi Protocol Mode

LDmitryL, 2019-01-14:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
We really need support for switching Wi-Fi Protocol mode **802.11 LR** in NodeMCU. 

### Justification
According to the manufacturer communication distance in **one-kilometre** is very much helped in the attraction device on the basis of ESP32
8<------------------------ END FEATURE REQUEST --------------------------------

nwf, 2019-01-14:
Please see #1010 and note that
- the attainable bandwidth for this 802.11 LR is likely to be quite small and
- it requires that both endpoints of the communication be ESP32 devices.

marcelstoer, 2019-01-14:
@LDmitryL are you proposing to resource the development yourself?

LDmitryL, 2019-01-16:
@marcelstoer No. I can't. I ask for help from outside to implement this feature! I think this functionality is very important in the area where we use NodeMCU.

marcelstoer, 2019-01-16:
> in the area where **we** use NodeMCU

That's the point that #1010 is trying to make. If you need a feature, how ever useful it might be (like this one), that the volunteers here haven't implemented yet then you should be prepared to fund it yourself - time- or otherwise.

LDmitryL, 2019-01-17:
@marcelstoer The existance of the switching protocols function is needed not only for me. But the existance of such function in LUA for NodeMCU makes it more attractive and useful for other developers of automation at home. Maybe that will push other developers to the nodemcu-firmware issue

closed 2019-01-21 by LDmitryL

x-logLT, 2019-03-16:
For now I have implemented this functionality in my "[work in progress](https://github.com/x-logLT/nodemcu-firmware/tree/esp32-working)" branch, if it works as intended I'll make a PR some time later.
mode can be set then calling config for ap or sta and setting phymode in cfg table, available values are:
```
wifi.PHYMODE_B
wifi.PHYMODE_BG
wifi.PHYMODE_BGN
wifi.PHYMODE_BGNLR
wifi.PHYMODE_LR
```