nodemcu/nodemcu-firmware #2613: ESP32: Support wakeup from deep sleep from external source

migerh, 2019-01-14:
### Missing feature

#### ESP32: Support wakeup from deep sleep from external source

### Justification
Using `node.dsleep()` there does not seem to be any way to wake up the ESP32 with an external trigger (e.g. a button). There is `gpio.wakeup()` but that can't be used to wake up from **deep** sleep because it uses the `gpio_wakeup_enable()` API which [only works with light sleep, not deep sleep](https://github.com/espressif/esp-idf/blob/master/docs/en/api-reference/system/sleep_modes.rst#gpio-wakeup-light-sleep-only). For deep sleep there seems to be another API, [esp_sleep_enable_ext0_wakeup()](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html#_CPPv228esp_sleep_enable_ext0_wakeup10gpio_num_ti).

### Workarounds
None. Or maybe I missed something and the `esp_sleep_enable_ext0_wakeup()` API is already mapped to some NodeMCU API - in that case I'd be glad to get a pointer in the right direction.

In case I didn't miss anything: I already patched the firmware to go on with my project and added a new function `gpio.dsleep_wakeup(pin, intr_type)` that uses the IDF sleep API instead of the gpio API. If you're interested I can improve the patch (extend docs, add tests if necessary, incorporate results from discussions in this issue...) and open a PR. Of course that's a quick hack and considering there are more options to wake up an esp32 in deep sleep it might make sense to add a new `sleep` module for that.

Possibly related: #1612 #1614

marcelstoer, 2019-01-14:
> If you're interested I can improve the patch (extend docs, add tests if necessary, incorporate results from discussions in this issue...) and open a PR.

As per #1010 your are "encouraged" to propose to resource the development yourself if you raise feature requests. Thanks for offering just that.

migerh, 2019-01-20:
I really like your API proposal @devsaurus, it's very compact.

In my use case the only purpose of the pin was to wake up the ESP32, nothing else - so I didn't really have to think about `rtc_gpio_deinit()`. The only thing I can think of is storing the config in RTC memory and restore the pin on wakeup.

Also, there's a few more things to consider:

* IDF supports disabling a wakeup source. I guess that's not really necessary here since the the wakeup sources are configured right before the esp goes to sleep.
* touch pad wakeup. That could be just a bool field in the dsleep config parameter since the configuration is separate (if it is supported by nodemcu at this point, couldn't find anything in the docs about that)
* how to deal with `rtc_gpio_isolate`?
* get wakeup cause (`esp_sleep_get_wakeup_cause`) after a wakeup. According to the docs there is `node.bootreason` but that did not work when I tried it. Grep on the source did not find anything either.

Instead of adding all the features the IDF API has to offer all at once, would it be ok for you to start small and only add e.g. the external source wakeup stuff?

mjmcginty, 2019-01-28:
You should also consider supporting ext1 sources (multiple pins.)

migerh, 2019-01-31:
@mjmcginty Yes. Whenever I said ext0 please read as "ext0 *and ext1*".

So let's start with the timeout trigger and the pin triggers, both ext0 and ext1. How about an API similar to `gpio.config()`? `node.dsleep()` would be a variadic function where the parameters can either be

* a number (time based wakeup)
* a table with a pin property that is a number (ext0) and a level entry which defines whether HI or LO triggers a wakeup
* a table with a pin property that is a list of pin numbers (ext1) and a wakeup_mode enum (ALL_LOW & ANY_HIGH)

The latter two will use `esp_sleep_enable_gpio_wakeup()` on non-RTC pins and `ext0/1` on RTC pins.

`gpio.wakeup()` will be marked as deprecated.

In `gpio.config()` `rtc_gpio_deinit()` should be called to make sure the pin is set up correctly.

Does this sound reasonable?

devsaurus, 2019-01-31:
> `node.dsleep()` would be a variadic function where the parameters can either be

Does the SDK allow to sleep and wake up upon timeout or a pin event - whatever happens first?

> `gpio.wakeup()` will be marked as deprecated.

This can even be removed right away without deprecation period IMO. It doesn't work atm, so nobody would miss it :smiley: 

> In `gpio.config()` `rtc_gpio_deinit()` should be called to make sure the pin is set up correctly.

Agree

> Does this sound reasonable?

Yes :+1:

mjmcginty, 2019-02-01:
>> Does the SDK allow to sleep and wake up upon timeout or a pin event - whatever happens first? <<

Yes, they are configured separately but can be active at the same time.  It also supports wake-up on multiple pins, with two options: wake when all are low, or wake when any is high.  It's extremely useful, imho.

One other thing, I can't remember if you have any support for pcnt (pulse counters) but if you do, it must be disabled before sleeping if any of the wake pins are also used with pcnt (otherwise it will wake immediately.)  To do that call pcnt_unit_config with both pulse_gpio_num and ctrl_gpio_num set to PCNT_PIN_NOT_USED.  (perhaps not the most intuitive thing in the world.)

tomsci, 2020-04-25:
Can we also address #1944 here too, ie to allow wakeups greater than 35 mins on integer builds?

> a table with a pin property that is a number (ext0) and a level entry which defines whether HI or LO triggers a wakeup

Is there any real reason to ever use ext0 wakeup? It seems like ext1 offers everything it can do and more? I wasn't able to measure any difference in power consumption doing an wake-on-ext0 deep sleep versus a wake-on-ext1 deep sleep. Although it was on the edge of what my multimeter could measure so I'm not 100% sure of that.

Is anyone actively working on this? I have code for 90% of this (ie a working patch supporting ext1 wakeup which I've been using for a while, but not quite in the exact syntax proposed here) which I will tidy up and PR if it's not going to conflict with ongoing work.

tomsci, 2020-05-17:
Huh, didn't realise github would ping this issue every time I amended a commit even before opening a PR... sorry about the noise.

Anyway, since no-one objected I've opened a PR #3115 with my version of node.dsleep() adapted to the format discussed here. Let me know what you think.