nodemcu/nodemcu-firmware #2619: Automated doc build broken?

devsaurus, 2019-01-18:
### Expected behavior
Latest commits reflected on https://nodemcu.readthedocs.io/en/dev/ and https://nodemcu.readthedocs.io/en/dev-esp32

### Actual behavior
There haven't been updates on https://nodemcu.readthedocs.io/ since ~1.5d.
I'm missing the changes introduced by #2607, #2608.

### Test code
Read the docs.
E.g. https://nodemcu.readthedocs.io/en/dev/sdcard/ still claims that fatfs is on version 0.12a.

### NodeMCU version
`dev` and `dev-esp32`

marcelstoer, 2019-01-21:
https://github.com/rtfd/readthedocs.org/issues/5142

marcelstoer, 2019-01-22:
-> we need access to the repository settings to configure the new webhook OR
-> we need @nodemcu / @funshine to do this for us

galjonsfigur, 2019-02-19:
I saw commit https://github.com/nodemcu/nodemcu-firmware/commit/5032483a174b8b55886ec16337c790bf78b50530 and @nodemcu activity - Has a new webhook been configured?

marcelstoer, 2019-02-19:
New webhook is in place.

closed 2019-02-19 by marcelstoer