nodemcu/nodemcu-firmware #2622: SNTP success callback function runs when sync not successful

alistairwitt, 2019-01-22:
### Expected behavior
Success callback function gets called only after successful SNTP sync. Failure callback function gets called after SNTP syncfailure.

### Actual behavior
The success callback function seems to get called when (or possibly before) the NTP sync fails. This happens for example if the wifi is not connected, or sometimes on the first sync attempt after boot (with wifi connected). Also `rtctime.get()` within the callback function returns zero which seems to further indicate that the NTP sync hasn't been successful (yet the success callback function ran).

Apologies if I've missed something, and it's not a bug after all. Also, unfortunately the C source code for the SNTP module is a bit over my head, so I can't offer any further insight or suggest a fix (though I wish I could).

### Test code
```Lua
-- Define callback function for ntp sync success
function ntpSyncSuccess (sec, usec, server, info)
    print('SNTP time sync successful!')
    print("rtctime.get() returns: ", rtctime.get())
end

-- Configure and start NTP time sync with auto repeat enabled
sntp.sync("0.au.pool.ntp.org",
    ntpSyncSuccess(sec, usec, server, info),   --success callback
    function()          -- error callback
        print('SNTP time sync failed!')
    end,
    1 -- enable autorepeat (SNTP sync every 1000 seconds (~17 min))
)
```
### NodeMCU version
```
NodeMCU custom build by frightanic.com
    branch: master
    commit: 11592951b90707cdcb6d751876170bf4da82850d
    SSL: false
    modules: cron,file,gpio,i2c,net,node,rotary,rtctime,sntp,struct,tmr,uart,wifi
 build created on 2019-01-16 03:11
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
```

### Hardware
Wemos D1 Mini (or clone), connected via USB, no other wiring or connections.

galjonsfigur, 2019-01-22:
Can you try following change in your code? I think the SNTP module is fine - only callback function is passed incorrectly.
```lua
sntp.sync("0.au.pool.ntp.org",
    ntpSyncSuccess,
    function()          -- error callback
```
instead of
```lua
sntp.sync("0.au.pool.ntp.org",
    ntpSyncSuccess(sec, usec, server, info),   --success callback
    function()          -- error callback
```

alistairwitt, 2019-01-22:
I've just tested this code change and it does indeed work. Apologies for the wild goose chase!

marcelstoer, 2019-01-22:
Sorry I didn't spot this when I saw the question on SO.

closed 2019-01-22 by marcelstoer

NiklausS, 2019-03-18:
Sorry to bring this up again, but somehow this does not work for me either.

When i block the ntp address on my dsl, the esp crashes after 1000sec, after the 2nd try for ntp.

this is my code:

```
    print('start SNTP time sync')   
    local function sntpG(sec,usec,server,info)
        print(' good sntp')
    end

    local function sntpB(e)
        print(' bad sntp')
    end

    local function set_sntp1(tmsrv)
                sntp.sync( tmsrv, 
                    function(sec,usec,server,info) print("good sntp") end,
                    function(e) print ("bad sntp") end,
                    1 )
          end

    local function set_sntp2(tmsrv)
                sntp.sync( tmsrv, 
                    sntpG,
                    sntpB,
                    1 )
          end
            
    set_sntp2('168.1.23.122')

```
and this is the output:

```
NodeMCU 2.2.0.0 build unspecified powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
start SNTP time sync
>  bad sntp
....
here it takes 1000sec...
....
PANIC: unprotected error in call to Lua API (bad argument #-1 (string expected, got nil))

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 28360, room 16 
tail 8
chksum 0x5e
load 0x3ffe8000, len 2280, room 0 
tail 8
chksum 0x73
load 0x3ffe88e8, len 136, room 0 
tail 8
chksum 0xb3
csum 0xb3

NodeMCU 2.2.0.0 build unspecified powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
start SNTP time sync
>  bad sntp
... and it repeates after 1000 sec
```

is this only me having this problem, or doesn't this code work for others either? 
My last snapshot is from about october 2018.

Thanks  & Regards
 Nik